espressif/esp-idf #1475: Toolchain HTTPS Cert invalid (https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz)

Curclamas, 2018-01-07:
Hello, the certificate for the toolchain-URL (https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz) seams to be invalid as of today.
This breaks our CI-Workflow and raises security concerns. See attached picture:

[image]

Curclamas, 2018-01-07:
Issue seams to be fixed!

closed 2018-01-07 by Curclamas

dbagnara, 2019-03-29:
Issue seems to be back again. Invalid certificate.
I am not an expert but could the *.espressif.com certificate be used instead of the Lets Encrypt certificate.

[image]

dbagnara, 2019-03-30:
Fixed