espressif/esp-idf #1478: Wi-Fi example error

Cooler1989, 2018-01-08:
Hi I encountered problems with 'console' example. I have tried with origin/release/v3.0 (6c69d5e) and with master ( f9ad17e ). Same results. Could you help me identify whether it is an issue?
```
esp32> join ********* **********
I (353576) connect: Connecting to '***********'
W (353576) wifi: wifi nvs_open fail ret=4353

E (353576) wifi: wifi_init 1408 ret=4353
ESP_ERROR_CHECK failed: esp_err_t 0x1101 at 0x4010781c
0x4010781c: initialise_wifi at /home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./cmd_wifi.c:52 (discriminator 1)

file: "/home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./cmd_wifi.c" line 52
func: initialise_wifi
expression: esp_wifi_init(&cfg)

Backtrace: 0x400886e8:0x3ffbb520 0x40088a55:0x3ffbb540 0x4010781c:0x3ffbb560 0x40107873:0x3ffbb5d0 0x40107967:0x3ffbb670 0x4011eac9:0x3ffbb690 0x4010739d:0x3ffbb6b0 0x400d11fa:0x3ffbb6e0
0x400886e8: invoke_abort at /home/cooler1989/embedded/home_automation/esp-idf/components/esp32/./panic.c:572

0x40088a55: _esp_error_check_failed at /home/cooler1989/embedded/home_automation/esp-idf/components/esp32/./panic.c:584

0x4010781c: initialise_wifi at /home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./cmd_wifi.c:52 (discriminator 1)

0x40107873: wifi_join at /home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./cmd_wifi.c:61

0x40107967: connect at /home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./cmd_wifi.c:95 (discriminator 1)

0x4011eac9: esp_console_run at /home/cooler1989/embedded/home_automation/esp-idf/components/console/./commands.c:195

0x4010739d: app_main at /home/cooler1989/embedded/home_automation/esp-idf/examples/system/console/main/./console_example_main.c:153

0x400d11fa: main_task at /home/cooler1989/embedded/home_automation/esp-idf/components/esp32/./cpu_start.c:449

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4540
load:0x40078000,len:0
load:0x40078000,len:12980
entry 0x40078d14
I (240) cpu_start: Pro cpu up.
I (240) cpu_start: Starting app cpu, entry point is 0x40081288
0x40081288: call_start_cpu1 at /home/cooler1989/embedded/home_automation/esp-idf/components/esp32/./cpu_start.c:215

I (226) cpu_start: App cpu up.
I (243) heap_init: Initializing. RAM available for dynamic allocation:
I (250) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (256) heap_init: At 3FFB8F78 len 00027088 (156 KiB): DRAM
I (262) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (269) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (275) heap_init: At 400927F0 len 0000D810 (54 KiB): IRAM
I (281) cpu_start: Pro cpu start user code
I (299) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

This is an example of ESP-IDF console component.
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
esp32> 
```

negativekelvin, 2018-01-09:
Possible hint https://github.com/espressif/esp-idf/pull/1279

igrr, 2018-01-09:
Yes, adding nvs_flash_init to app_main will make it work. We will have a fix shortly.

closed 2018-01-16 by igrr