espressif/esp-idf #1479: OTA updates failing randomly inside spi_flash_write_inner

MrSurly, 2018-01-08:
Having a lot of trouble with OTA updates.  

After doing a bit of digging, it appears that `spi_flash_write_inner` is not returning properly from where it's called from `flash_ops.c`:
https://github.com/espressif/esp-idf/blob/master/components/spi_flash/flash_ops.c#L375

Line 377 is never reached.

I've updated to the latest IDF (hash f9ad17eee45e6fb6c48ef72f8ff8aa1480bb2453), and the IDF submodules are up-to-date as well.  Toolchain is: `xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-75-gbaf03c2) 5.2.0`

Doesn't seem to matter if `SPI_FLASH_VERIFY_WRITE` (and setting all suboptions) is set or not. Unsetting `CONFIG_SPI_FLASH_ROM_DRIVER_PATCH` seems to make it worse, when usin

When it freezes, the serial monitor shows:

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: ipc1
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: ipc1
```

Relevant code section is below.  `esp_ota_write` fails to return.

        bool downloadUpdate(const string& url, string updateHash) {
            const char * TAG = "update";

            const esp_partition_t *configured = esp_ota_get_boot_partition();
            const esp_partition_t *running = esp_ota_get_running_partition();
            const esp_partition_t *update_partition = esp_ota_get_next_update_partition(NULL);

            if (configured != NULL ) {
                L(TAG, "Configured for partition subtype %d at offset 0x%x", configured->subtype, update_partition->address);
            } else {
                L(TAG, "Configured partition is NULL");
            }

            if (running != NULL) {

                L(TAG, "Running from partition subtype %d at offset 0x%x", running->subtype, update_partition->address); 
            } else {
                L(TAG, "Running partition is NULL");
            }

            if(update_partition != NULL) {
                L(TAG, "Writing to partition subtype %d at offset 0x%x", update_partition->subtype, update_partition->address);

            } else {
                L(TAG, "Running partition is NULL");
            }
            esp_ota_handle_t update_handle;
            L(TAG, "update begin");
            if (esp_ota_begin(update_partition, OTA_SIZE_UNKNOWN, &update_handle) == ESP_OK) {
                L(TAG, "after update begin");

                Http http("<redacted>/" VERSION_STRING, ROOT_CERT);
                int result = http.get(url);
                if (result == 200) {

                    Sha256 sha256;
                    uint8_t buf [1024];
                    int read;
                    int ttl_read = 0;
                    int timeout = 5000;
                    elapsedMillis printTimer = 3000;

[... 607 lines cut ...]
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y

#
# Debug Configuration
#
CONFIG_SPIFFS_DBG=
CONFIG_SPIFFS_API_DBG=
CONFIG_SPIFFS_GC_DBG=
CONFIG_SPIFFS_CACHE_DBG=
CONFIG_SPIFFS_CHECK_DBG=
CONFIG_SPIFFS_TEST_VISUALISATION=

#
# tcpip adapter
#
CONFIG_IP_LOST_TIMER_INTERVAL=120

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096
```

MrSurly, 2018-01-10:
I have a minimalized reproduction of this issue.

negativekelvin, 2018-01-11:
>EDIT 3.5: Increasing WDT timeout to 10s (per #578) fixes this

Fixes all the issues or just with esp_ota_begin?

MrSurly, 2018-01-11:
> Fixes all the issues or just with esp_ota_begin?

Just the `esp_ota_begin`, the main issue is still there.

projectgus, 2018-02-06:
Hi @MrSurly,

Thanks for being patient while someone gets back to you about this. I have a couple of questions:

* If you add a vTaskDelay(1) somewhere in the main OTA while loop, does anything change?
* Which CPU is the update task running on?
* What other tasks (non-IDF-internal) are running in the system while the OTA update is running?
* If you turn on Comprehensive heap checking, does anything jump out at you?

The first question is because we've noticed, particularly on LANs where the TCP latency is minimal and bandwidth is much higher than the flash write bandwidth, the OTA loop can starve other tasks in the system. This may explain the the task WDT, although it doesn't explain the other part.

The high-priority IPC (inter-processor) task being involved in the crash dump is related to flash write operations. The CPU doing a flash write will call spi_flash_guard_start() which signals the IPC task on the other CPU to prevent it from reading flash (via code cached in flash) until the flash write is over. This doesn't explain why it's not returning from this properly, though... which is why I mention heap corruption as a possibility.

> I have a minimalized reproduction of this issue.

Is this something you're able to share?

MrSurly, 2018-02-06:
> If you add a vTaskDelay(1) somewhere in the main OTA while loop, does anything change?

Yes, this helps immensely. Seems much more rel

> Which CPU is the update task running on?

Honestly, not sure. Using `xTaskCreate(mainTask, "mainTask", 32768, NULL, tskIDLE_PRIORITY, NULL);` from `app_main`.  Code below.

> What other tasks (non-IDF-internal) are running in the system while the OTA update is running?

For my actual project, there are several; stuff like keeping Wifi active and what not.  For the example code below, there's only one user-space task.

> If you turn on Comprehensive heap checking, does anything jump out at you?

Not for the example code; haven't tried yet for production.

> The first question is because we've noticed, particularly on LANs where the TCP latency is minimal and bandwidth is much higher than the flash write bandwidth, the OTA loop can starve other tasks in the system. This may explain the the task WDT, although it doesn't explain the other part.

I am downloading from my local workstation; latency is likely < 1ms

> The high-priority IPC (inter-processor) task being involved in the crash dump is related to flash write operations. The CPU doing a flash write will call spi_flash_guard_start() which signals the [...]

Hrm.  Let me turn on the heap checking for my main code and we'll see

>> I have a minimalized reproduction of this issue.

>Is this something you're able to share?

Yes, below.  I can provide the supporting code if you actually wish to compile.

```c

// C
#include <stdio.h>
#include <sys/time.h>
#include <time.h>
#include <unistd.h>
#include <stddef.h>
#include <string.h>
#include <sys/param.h> // MIN/MAX

// ESP
#include <esp_task_wdt.h>
#include <esp_wifi.h>
#include <esp_wifi_types.h>
#include <esp_log.h>
#include <esp_event_loop.h>
#include <esp_ota_ops.h>
#include <nvs_flash.h>

// local
#include "Http.hpp"
#include "Sha256.hpp"
#include "elapsedMillis.h"
#include "log.h"
#include "config.h"

#define HEXDIGIT(x) (x) <= 9 ? '0' + (x): 'A' + ((x)-10)
#define ARRAY_LEN(x) (sizeof(x) / sizeof(x[0]))

using namespace std;

class Main {

    public:

        Main() : m_haveIp(false) {}

        static esp_err_t eventHandler(void *ctx, system_event_t *event) {
            const char * TAG = "Event";
[... 234 lines cut ...]
                downloadUpdate("http://172.16.23.157/test.bin","13fbead996e7fa70612467c3ea478fb8db80db47b0d5f97e0f7662027bbf9cd9");
                L(TAG, "End test #%d\n", i + 1);
            }
            L(TAG, "finished, idle");
            while(true) {
                sleep(3);
            }
        }

    private:
        volatile bool m_haveIp;
};

void mainTask (void* ctx) {
    (void)ctx;
    Main test;
    test.run();
}

extern "C" {
    void app_main() {
        xTaskCreate(mainTask, "mainTask", 32768, NULL, tskIDLE_PRIORITY, NULL);
    }
};
```

MrSurly, 2018-02-06:
Just got this ... a little different this time since CPU 0 says "wifi".  The ones after it say "IDLE" for CPU 0.

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: wifi
CPU 1: ipc1
```

MrSurly, 2018-02-06:
Not sure of the relevance, but upon _success_, this is seen when finishing the OTA

```
I (34910) esp_image: segment 1: paddr=0x0017a324 vaddr=0x3ffc0000 size=0x03320 ( 13088) 
I (34940) esp_image: segment 2: paddr=0x0017d64c vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _iram_start at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (34970) esp_image: segment 3: paddr=0x0017da54 vaddr=0x40080400 size=0x025bc (  9660) 
I (35000) esp_image: segment 4: paddr=0x00180018 vaddr=0x400d0018 size=0xbc07c (770172) map
0x400d0018: _stext at ??:?

I (36070) esp_image: segment 5: paddr=0x0023c09c vaddr=0x400829bc size=0x110c0 ( 69824) 
0x400829bc: _xt_coproc_exc at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/freertos/./xtensa_vectors.S:999

I (36190) esp_image: segment 6: paddr=0x0024d164 vaddr=0x400c0000 size=0x00064 (   100) 
I (36220) esp_image: segment 7: paddr=0x0024d1d0 vaddr=0x50000000 size=0x00008 (     8) 
[  update]: computed hash 13FBEAD996E7FA70612467C3EA478FB8DB80DB47B0D5F97E0F7662027BBF9CD9
[  update]: Updating bootable partition
I (36280) esp_image: segment 0: paddr=0x00140020 vaddr=0x3f400020 size=0x3a2fc (238332) map
I (36610) esp_image: segment 1: paddr=0x0017a324 vaddr=0x3ffc0000 size=0x03320 ( 13088) 
I (36630) esp_image: segment 2: paddr=0x0017d64c vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _iram_start at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (36660) esp_image: segment 3: paddr=0x0017da54 vaddr=0x40080400 size=0x025bc (  9660) 
I (36680) esp_image: segment 4: paddr=0x00180018 vaddr=0x400d0018 size=0xbc07c (770172) map
0x400d0018: _stext at ??:?

I (37740) esp_image: segment 5: paddr=0x0023c09c vaddr=0x400829bc size=0x110c0 ( 69824) 
0x400829bc: _xt_coproc_exc at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/freertos/./xtensa_vectors.S:999

I (37850) esp_image: segment 6: paddr=0x0024d164 vaddr=0x400c0000 size=0x00064 (   100) 
I (37870) esp_image: segment 7: paddr=0x0024d1d0 vaddr=0x50000000 size=0x00008 (     8) 
```

projectgus, 2018-02-07:
Thanks for all this info. There's a lot here, but one thing jumped out at me:

> >    Which CPU is the update task running on?
> Honestly, not sure. Using xTaskCreate(mainTask, "mainTask", 32768, NULL, tskIDLE_PRIORITY, NULL); from app_main. Code below.

This will allow the task to migrate between cores depending on which is idle. Given the crash seems to be related to inter-core communciation in a busy system, this may be relevant.

Can you please try changing this to xTaskCreatePinnedToCore(). You'll need another argument at the end of the args list, either 0 or 1 (for which core to pin to). If you're not sure, try 1.

(This isn't a fix as much as trying to narrow down whatever the underlying cause of the bug is.)

MrSurly, 2018-02-07:
Using `xTaskCreatePinnedToCore` it's better.  It no longer goes into a WDT loop w/o resetting, but now it resets (maybe 10% of the time):

```ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5632
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13716
entry 0x40078fb4
W (30) boot: PRO CPU has been reset by WDT.
W (30) boot: WDT reset info: PRO CPU PC=0x40083a99
0x40083a99: spi_flash_op_block_func at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/spi_flash/./cache_utils.c:82 (discriminator 1)

W (31) boot: WDT reset info: APP CPU PC=0x400d1bde (waiti mode)
0x400d1bde: esp_vApplicationIdleHook at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/esp32/./freertos_hooks.c:85

I (38) boot: ESP-IDF v3.1-dev-171-gf9ad17e-dirty 2nd stage bootloader
```

MrSurly, 2018-02-08:
So, this just happened, inside of esp_wifi_init().  Wifi is managed by a different task. Should all tasks I create be pinned to a core to prevent this issue?

```
[initWifi]: esp_wifi_init()
I (8855) wifi: wifi firmware version: 403db1d
I (8855) wifi: config NVS flash: enabled
I (8855) wifi: config nano formating: disabled
I (8855) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (8865) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 1)
Tasks currently running:
CPU 0: IDLE
CPU 1: ipc1
< loops forever >
```

rahul-bedarkar, 2018-02-09:
I am facing exactly same with OTA. I have external SPI RAM. There are many tasks running while OTA is running. If I run only wifi and OTA tasks, OTA completes successfully without watchdog trigger issue.

MrSurly, 2018-02-15:
Now I'm seeing WDT resets during _normal_ operation, which appear flash related:

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5832
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:15376
entry 0x400790d4
W (30) boot: PRO CPU has been reset by WDT.
W (30) boot: WDT reset info: PRO CPU PC=0x40083c37
0x40083c37: spi_flash_op_block_func at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/spi_flash/./cache_utils.c:82 (discriminator 1)

W (31) boot: WDT reset info: APP CPU PC=0x400d1e92 (waiti mode)
0x400d1e92: esp_vApplicationIdleHook at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/esp32/./freertos_hooks.c:85

I (38) boot: ESP-IDF v3.1-dev-185-gdcd7a1e 2nd stage bootloader
I (44) boot: compile time 09:18:35
I (49) boot: Enabling RNG early entropy source...
I (54) boot: SPI Speed      : 40MHz
I (58) boot: SPI Mode       : DIO
I (62) boot: SPI Flash Size : 4MB
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00003000
I (84) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 00130000
I (99) boot:  3 otadata          OTA data         01 00 00140000 00004000
I (107) boot:  4 XXX           WiFi data        01 02 00144000 00006000
I (114) boot:  5 ota_0            OTA app          00 10 00150000 00130000
I (122) boot:  6 ota_1            OTA app          00 11 00280000 00130000
I (130) boot: End of partition table
I (134) boot: Defaulting to factory image
I (138) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3c988 (248200) map
I (234) esp_image: segment 1: paddr=0x0004c9b0 vaddr=0x3ffc0000 size=0x034e0 ( 13536) load
I (239) esp_image: segment 2: paddr=0x0004fe98 vaddr=0x40080000 size=0x00178 (   376) load
0x40080000: _WindowOverflow4 at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (241) esp_image: segment 3: paddr=0x00050018 vaddr=0x400d0018 size=0xcf5c4 (849348) map
0x400d0018: _flash_cache_start at ??:?

I (544) esp_image: segment 4: paddr=0x0011f5e4 vaddr=0x40080178 size=0x00288 (   648) load
0x40080178: _WindowUnderflow12 at ??:?

I (544) esp_image: segment 5: paddr=0x0011f874 vaddr=0x40080400 size=0x14e9c ( 85660) load
I (586) esp_image: segment 6: paddr=0x00134718 vaddr=0x400c0000 size=0x00064 (   100) load
I (586) esp_image: segment 7: paddr=0x00134784 vaddr=0x50000000 size=0x00008 (     8) load
I (605) boot: Loaded app from partition at offset 0x10000
I (605) boot: Disabling RNG early entropy source...
I (607) cpu_start: Pro cpu up.
I (610) cpu_start: Starting app cpu, entry point is 0x40081014
0x40081014: call_start_cpu1 at /home/epoulsen/workspaces/XXX/ext_deps/esp-idf/components/esp32/./cpu_start.c:215

I (0) cpu_start: App cpu up.
I (621) heap_init: Initializing. RAM available for dynamic allocation:
I (628) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (634) heap_init: At 3FFD01E8 len 0000FE18 (63 KiB): DRAM
I (640) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (646) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (653) heap_init: At 4009529C len 0000AD64 (43 KiB): IRAM
I (659) cpu_start: Pro cpu start user code
I (9) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) gpio: GPIO[13]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 0| Pulldown: 0| Intr:0 
I (10) gpio: GPIO[14]| InputEn: 1| Out
```

projectgus, 2018-02-16:
Hi @MrSurly ,

Sorry, I've been out sick the past week.

Can I please take you up on the offer of a full project I can reproduce this in?

Angus

MrSurly, 2018-02-16:
> Sorry, I've been out sick the past week.

=(  Sorry to hear that.  I hope you're feeling better.

> Can I please take you up on the offer of a full project I can reproduce this in?

Absolutely.  It's presently a branch from a client project.  I'm out for the next 4 days -- I'll share a repo when I can get it properly sanitized and anonymized.

MrSurly, 2018-02-21:
@projectgus 
Sorry for the delay: https://github.com/MrSurly/ESPIDF-issue-1479

I use an image > 1MB.  Usually fails around attempt # 10, with a WDT.

MrSurly, 2018-03-20:
@projectgus Any word?

chegewara, 2018-03-20:
I have one question, maybe its relevant or maybe not. Why you create main task with so low priority?
`xTaskCreate(mainTask, "mainTask", 32768, NULL, tskIDLE_PRIORITY, NULL); `

Could you try something like:
`xTaskCreate(mainTask, "mainTask", 32768, NULL, 5, NULL); `

projectgus, 2018-03-21:
Hi @MrSurly ,

Sorry for the extended silence I've been busy with some other tasks and this slipped.

@chegewara makes a very good point about the task priority. I totally missed this when you posted it earlier. This means that the main task is always competing with the idle task for CPU time, and this may have unexpected consequences (at minimum, degraded performance as the idle task generally assumes that any "real" task will pre-empty it). Try raising it to a value like the 5 suggested, or even `(tskIDLE_PRIORITY + 1)`.

Angus

MrSurly, 2018-03-22:
@chegewara @projectgus 

Increasing the priority seems to have entirely fixed the issue, though I don't understand the mechanism of _why_.

chegewara, 2018-03-22:
https://www.freertos.org/Documentation/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf

This is good position to start. Sure, you dont need to know freeRTOS, but if you will learn just a basic stuff then it will help you a lot in work with esp-idf.

Task has few parameters, one of them is priority. In your code it was set to tskIDLE_PRIORITY which is lowest possible. This priority is for idle freeRTOS task to make some cleanings etc. When you got your task set to tskIDLE_PRIORITY freeRTOS has been switching between your task and idle task even if your main task required more time to finish job. Basicaly for normal applications its good to set task priority between 5 and 10. But like i said, its good to read about it from this document to save time and health and avoid strange app behaviour.

projectgus, 2018-03-22:
@MrSurly ,

To add to what @chegewara says, the "idle task" is special because it's assumed that it's the task that runs when there's nothing of higher priority to run. So it may do things like, for example, suspend the CPU clock  to put the system into a low power state.

The idle task can do this because it knows that if anything else needed to run, it would have a higher priority so it would already be running (in a hard real-time system like an RTOS, the highest priority "ready" task will always run before any lower priority task(s)).

There is also a small difference in the ESP-IDF implementation of the FreeRTOS scheduler where [we don't guarantee fair round-robin preemption of tasks at the same priority](https://esp-idf.readthedocs.io/en/latest/api-guides/freertos-smp.html#round-robin-scheduling). The idle task never suspends waiting on a mutex or a semaphore (by design, because the assumption is that any other task in the system will have higher priority so why should the idle task ever need to block). So there's no guarantee that preemption will fairly wake up another task at the same "idle" priority if it's blocked on something.

(Normally this scenario would trigger the ESP-IDF task watchdog to say "hey, task X is never blocking and might be starving something else out." But, again, the idle task is special as the assumption is that it's the lowest priority thing in the system so it can never starve anything out.)

TLDR: Every task created should be at least `tskIDLE_PRIORITY +1` priority, unless you're doing something very complex with FreeRTOS internals.

Sorry that I didn't spot this in your sample code the first time around, and well done @chegewara for spotting it.

chegewara, 2018-03-22:
Its my bad too, because i dont read most of issues ( i just dont feel i have knowledge to answer, this time i just got lucky), and im glad it solved your problem.

Have a fun with esp32.

MrSurly, 2018-03-23:
@chegewara @projectgus 

Thank you both for this. That's what I get for copy-paste the task create code. Going to close this.

closed 2018-03-23 by MrSurly