espressif/esp-idf #1481: BLE Security on addressing

hopkinskong, 2018-01-09:
Does ESP32 offer BLE Security on the device addressing aspect to prevent identity tracking?

I am interested in the following addressing scheme:

1. Random Static- These addresses are either burned into the device’s silicon during manufacture or generated when the device power cycles. If the device generates a new address every power cycle and the user power cycles the device on a regular basis, then this address type offers some protection from identity tracking. Otherwise, the protection this address type offers is limited.

2. Random Private Resolvable- This addressing method can only be used if the Identity Resolving Key (IRK) is exchanged between the two devices during the bonding process. With this method, the device will use the IRK to translate its device address in to a random address that appears in the advertisement packet. A second device that also possesses the IRK is then able to convert the random address back into the real address and identify the first device. In this method, the device will periodically generate a new random address based off the IRK, providing significant protection against identity tracking.

3. Random Private Non-Resolvable- In this addressing method, the device address is simply a random number and a new device address can be generated at any time. If new addresses are generated fairly often, then this method offers significant protection against identity tracking. 

Source: https://eewiki.net/display/Wireless/A+Basic+Introduction+to+BLE+Security#ABasicIntroductiontoBLESecurity-Addressing:

chegewara, 2018-01-10:
Some of your questions can be answered with this secure server example. Here you can see how to setup random address:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L90
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L372
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L338

I cant help you more because i didnt play with random addresses yet, but @Yulong-espressif can help for sure.

Weijian-Espressif, 2018-03-14:
@chegewara thanks for your help. @hopkinskong are there any questions?

closed 2018-03-14 by hopkinskong