espressif/esp-idf #1482: How to resolve ipc1 task

sjoteppagol, 2018-01-10:
I have three task, 
1)Main task (Higher priority level- **10**)
2) Wifi ( Priority level - **15**)
3) Data save ( priority level - **15**)
I am not getting why ipc2 task is getting trigeered !? How to resolve it?
[image]

chegewara, 2018-01-10:
Can you paste just `xTaskCreate()` commands for all tasks?

sjoteppagol, 2018-01-10:
```
  xTaskCreate(main_renesas, "main_renesas", 2048+1024, NULL, 10, NULL);
  xTaskCreatePinnedToCore(aws_iot_task, "aws_iot_task", 16384+1024, NULL, 15, NULL, 1);
  xTaskCreate(power_task, "power_task", 2048, NULL, 15, NULL);
```

igrr, 2018-01-10:
Is your task doing some flash, filesystem, or NVS operation when this happens?

sjoteppagol, 2018-01-10:
Yes...
In case of Power loss we are doing NVS operation just to have backup of important data, in this power_task is dedicated for NVS flash operation.

igrr, 2018-01-10:
ipc1,ipc0 tasks are auxiliary tasks used during flash operations. If code on CPU0 is doing flash operation, then ipc1 task runs for the duration of the operation; and vice versa. In the log you have posted, it says that IDLE task on CPU0 has not been running for a while. It means that the task on CPU0 did not yield to IDLE task for that duration of time. Ipc1 task is running on CPU1, so it is not itself responsible for the problem. It just happens to be running on CPU1 at the moment when the task watchdog is printing this message.

sjoteppagol, 2018-01-10:
Hello igrr,
Thank you for your quick response.
Do you have any suggestion on how to resolve it?

negativekelvin, 2018-01-10:
> Main task (Higher priority level- 10)

Low priority numbers denote low priority tasks.

How to resolve it would be break up long flash operations to smaller operations with vtaskdelay or increase task watchdog timeout.

sjoteppagol, 2018-01-22:
` vTaskDelay(10 / portTICK_RATE_MS); `
Just introducing small delay in application, solved the issue.

closed 2018-01-23 by FayeY