espressif/esp-idf #1483: Country code in scan result

lollizze, 2018-01-10:
Hi! Is it possible to add the country code in the scan result, we tried to be in promiscuous mode while scanning and parsing there but we need to get that information while connected.

liuzfesp, 2018-01-22:
@lollizze sure, we can add the country code into the scan result for each scanned AP, but I'm afraid we don't have plan to support it in short term. We will support this feature in v3.1 or v4.0.

XinDeng11, 2018-05-11:
Hi,@lollizze, we have added the country information in the wifi_ap_record_t, we will found a parameter named country after 3.1 released, if it earge, you may pull down the commit f3df30f9 after it merged, but it may need few days for review and test

closed 2018-05-19 by igrr