espressif/esp-idf #1484: gatts_add_char_descr_evt_param is missing characteristic handle?

chegewara, 2018-01-10:
Im wondering now, since descriptor is strictly connected to characteristic and not to service, why in `gatts_add_char_descr_evt_param `structure we have `uint16_t service_handle` but we dont have characteristic handle. Instead we have `char_uuid`, but what if we have few characteristics with the same UUID in one serviceand each characteristic have CCC descriptor? This is very real situation with ie HID devices. We can have many raport characteristics (0x2a4d).

negativekelvin, 2018-01-10:
You're not wrong but if you are creating characteristics and descriptors one at a time then you already know the handle from the last ESP_GATTS_ADD_CHAR_EVT

chegewara, 2018-01-10:
I see how i can get characteristic handle, but i see also we have here 2 fields that are pointless. Let say i have stored characteristic handle with descriptor then i dont need:
- characteristic uuid, because i can retrieve it from characteristic if/when it is needed; i know what characteristic it is so i can find it
- service handle, because descriptor does not belong to service, it belongs to characteristic and if for some reason i would need service handle i can retrieve it from characteristic which have it already stored in structure; this way we have 2 field that takes space in heap

More important is i cant search for characteristic that is owner of descriptor with uuid. I know its been changes introduced in v3.0 and now every attribut is accessed with handle, thats why im asking why we dont have characteristic handle in descriptor structure? Is it ommited intencionaly or just need to be fixed.

negativekelvin, 2018-01-10:
Well I suppose you could have 2 tasks each creating service so you could tell events apart by service handle.  But yes I don't know what you would use char uuid for.  The way it is currently working you shouldn't need to search for char at all because you should know which char you are expecting event for.

Yulong-espressif, 2018-01-11:
@chegewara Good, I think we should also added a char handle in the gatts_add_char_descr_evt_param structure is better. Because like the HID service have so many char uuid are the same. We will change it soon, thanks again.

chegewara, 2018-01-11:
Hi @Yulong-espressif 
its not something i need or require. Im using Neil's Kolban BLE library and we maneged this. We also have working HID library already. Just i found it and i think it may be useful to someone else. Especialy if hid component will be part of esp-idf it can be used. Thank you againg for you response.

chegewara, 2018-01-12:
Hi @Yulong-espressif 
I did some more reaserch and looks like this:
https://github.com/espressif/esp-idf/blob/abacf8d2a0347a4bf95315cdb227e9b8b1c93794/components/bt/bluedroid/api/include/esp_gatts_api.h#L165
is not exactly what it seems to be. As we can see from description it is characteristic uuid, but in real its attribute uuid (in our case its descriptor uuid).

Yulong-espressif, 2018-01-12:
@chegewara Yes, yo are right, I have seem it.  So in this case, I think change the char_uuid to descr_uuid is better.

chegewara, 2018-01-12:
Yes, but still char_handle would be nice addition. Have you seen this?
https://esp32.com/viewtopic.php?f=2&t=4193

Yulong-espressif, 2018-01-12:
@chegewara OK, we will change the char_uuid to descr_uuid and added the char handle also. Thank you very much.

vonnieda, 2018-01-12:
Just want to say thank you! I posted the thread above and this solution sounds great!

chegewara, 2018-03-22:
@Yulong-espressif As promissed here
https://github.com/espressif/esp-idf/issues/1484#issuecomment-357162497
name has been changed, thanks. If there is no plan to add char handle then this issue can be closed.

Yulong-espressif, 2018-04-03:
@chegewara We have fix this bug in the new version of SDK, thanks.

closed 2018-04-03 by chegewara