espressif/esp-idf #1485: Problem: 'ESP_OK' could not be resolved

mkvn, 2018-01-10:
Hi,

I am going to use the ble_adv exapmle from esp_idf. I didn't change any part of the program and have already done all the mentioned settings in eclipse propeerty. 
I see the problem "ESP_OK" could not be resolved", both when I didn't do the "make" and after doing that from inside the eclipse .
Can any body help me with a solution for that?

Vima

negativekelvin, 2018-01-10:
https://github.com/espressif/esp-idf/issues/356

mkvn, 2018-01-10:
Thank you negativekelvin. I had already read and done the points mentioned in your suggested link, bit it didn't solve.

For those who might have the same problem:
I solved it now by adding "#include "esp_system.h" inside my program.

closed 2018-01-17 by FayeY