espressif/esp-idf #1486: BLE disconnect error 0x003e and mtu: 0

chegewara, 2018-01-10:
I believe its not bug but my client or server disconnected with this error code:
```
E (2782) BT: bta_gattc_conn_cback() - cif=4 connected=0 conn_id=4 reason=0x003e
E (2783) BT: p_conn is NULL in bta_gattc_conn_cback
```
and then when tried to connect one more time i get this error:
[D][BLEUtils.cpp:1423] dumpGattClientEvent(): [reason: ESP_GATT_CONN_FAIL_ESTABLISH, conn_id: 0, remote_bda: 24:0a:c4:03:b1:c2]
```
[D][BLEDevice.cpp:140] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1377] dumpGattClientEvent(): GATT Event: ESP_GATTC_OPEN_EVT
[D][BLEUtils.cpp:1503] dumpGattClientEvent(): [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: 24:0a:c4:03:b1:c2, mtu: 0]
```
Most intriguing is `mtu: 0` part.

Ericvf, 2018-01-14:
I'm also getting this error. I'm using the latest version of everything (fresh install today). I'm using the GATT_client example to connect to a BLE keyboard. The keyboard works fine with apps such as LightBlue.

This is the error:
```
I (8062) GATTC_DEMO: f9 15 cc 6d e8 9c
I (8062) GATTTC_DEMO: searched Adv Data Len 19, Scan Response Len 16
I (8062) GATTC_DEMO: searched Device Name Len 6
I (8072) GATTC__DEMO: VB87MS
I (8072) GATTC_DEMO:

I (8082) GATTC_DEMO: searched device VB87MS

I (8082) GATTC_DEMO: connect to the remote device.
I (8092) GATTC_DEMO: stop scan successfully
E (88212) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x003e

E (8212) BT: p_conn is NULL in bta_gattc_conn_cback
```

Initially this error rose using PlatformIO on ESP-IDF v2.2.1 but I can assure you I am using the latest version at the moment. I followed the get started guide to be sure. 

If anyone can shed some light on this issue that would be great. Thanks in advance.

chegewara, 2018-02-11:
Now, after a month i can only tell it most likely was master commit when this error occured. I didnt do much tests with gatt_server and gatt_client examples but i think ive had no issue, in other hand those demo examples are very simple and i think it shouldnt be taken as benchmark or test examples.

FayeY, 2018-03-14:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-03-14 by FayeY