espressif/esp-idf #1488: Make on Mac OS X error : "cc1: internal compiler error: Illegal instruction: 4"

gerardwr, 2018-01-10:
Hi,

I installed ESP-IDF on my iMac acc. the instruction on the website. That worked fine.

But, when I "make" the hello-world example i get this error:
```
imac-van-gerard:hello_world gerard$ make
CC build/bootloader/bootloader_support/src/bootloader_clock.o
cc1: internal compiler error: Illegal instruction: 4
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
make[2]: *** [src/bootloader_clock.o] Error 1
make[1]: *** [component-bootloader_support-build] Error 2
make: *** [/Users/gerard/esp/hello_world/build/bootloader/bootloader.bin] Error 2
imac-van-gerard:hello_world gerard$ 
```

I have a  similar error ("cc1plus: internal compiler error: Illegal instruction: 4") when I use ESP32-Arduino, and reported a Github Issue here:
https://github.com/espressif/arduino-esp32/issues/764

In this issue @igrr Ivan Grokhotkov commented on 30 Oct 2017:
>Thanks for reporting this issue. We need to set macos_verison_min option when compiling on new hardware. It was done when the toolchain was built manually in the past, but the new CI build script doesn't set this option. So the toolchain ends up being incompatible with Macs which have older processors.

I'm using an iMac (half 2007) with Mac OS 10.11.6, and I would like to keep using it.

Could you please supply an updated version of the toolchain with the "macos_version_min" option set for older Mac processors as Ivan suggested, please!

I would be happy to test an updated version for you on my trusty old iMac 2007 ;-)

tlerebours, 2018-01-22:
same issue my old iMac 2007

gerardwr, 2018-01-23:
@tlerebours 

Thanks for your message, at least I'm not the only one anymore ;-)

Let's hope they can solve it soon.

gerardwr, 2018-02-04:
@FayeY 

Could you please give us an update on what is happening re. this issue?

Will the issue be adressed?

Thx.

muekno, 2018-02-04:
For me it seam the problem is know, an option is missing in the toolchain. So it should not be too hard to fix ist. Do not like to throw my fine working 2008 24" iMAC away.

gerardwr, 2018-02-04:
@muekno 
Hi, You're #3 with the same problem.

Acc. @igrr it's a simple solution. Hoe they can get it fixed real soon.

Thx.

wulph111, 2018-02-11:
@muekno @gerardwr @tlerebours 
I have received the same error message on my Mac Pro 1,1 running Yosemite (MacOS 10.10.1)
I was able to get around it by following the "Setup Toolchain for Mac OS from Scratch" procedure.

http://esp-idf.readthedocs.io/en/latest/get-started/macos-setup-scratch.html

Be sure to set aside a couple of hours.

I used the Homebrew procedure because (1) I'm not familiar with MacPorts, and (2) I don't have XCode and as far as I can tell, you can't get the old versions anymore.

I ran into a snag at the `git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git` step because it complained that the directory was not empty, due to the symbolic link created in the previous step. I worked around this by cloning into a subdirectory of the mounted image. Just be sure to adjust the subsequent steps accordingly.

I hope this helps get you up and running!

gerardwr, 2018-02-11:
@wulph111 
Seems like a horrofic lengthy procedure, but I’m gonna try it anyway :-)

Thanks for sharing this solution. Too bad Espressif does not make the time for a quick fix for all of us.

Code on Espressif, help us out here!

muekno, 2018-02-15:
not so horrofic. I did it once, but something a the end was wrong. So I deleted the created files and started from scratch with Creating the esp directory and then hdiutil create ~/esp/crosstool.dmg -volname "ctng" -size 10g -fs "Case-sensitive HFS+" but get hdiutil: create failed - Ungültiges Argument (ungültig = invald). I remember I had the same the first time but could fix it but do not remember how. I amnow googeling and searching for hours, no chanche to go around that. Running OS X El Capitan 10.11.6 4 GB RAM Intel Core Duo 24* iMac mid 2007
btw. had the same problem "complained that the directory was not empty," the first time.

I hope somebody can give me a hint
thanks Rainer

gerardwr, 2018-02-15:
@muekno @wulph111 

Did not have the time yet to build the toolchain, but found this link describing a similar procedure @wulph111 described. Looks nice!

Also describes why using a .dmg is necessary.

Maybe this helps dealing with the “directory not empty” issue.

https://v3.pubpub.org/pub/esp32-osx-setup

muekno, 2018-02-15:
The main difference ist the physical partition to the virtual partition for the build. In my research I found out the version of the ready ESPsoftware from espressif is the same download in the link from your first post. Espressif did update git hub seven days ago. but I will try your new link tomorrow thanks.

gerardwr, 2018-02-15:
@muekno 
I executed the commands in the linked description, but used hdiutil to create the virtual partition.

All commands worked fine, it's now working on the final "build" command:
```
imac-van-gerard:crosstool-NG gerard$ ./ct-ng build # this take a while
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180215.222532
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 110.00s (at 01:54)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[03:39] \ 
```
Looks promising. I assume that after a while the build will finish OK.

FYI the full log on the steps i took is here:
[ESP32 toolchain build on Mac OS 10.11.txt](https://github.com/espressif/esp-idf/files/1729469/ESP32.toolchain.build.on.Mac.OS.10.11.txt)

gerardwr, 2018-02-15:
@muekno 
The toolchain build was completed in 58 minutes.
```imac-van-gerard:crosstool-NG gerard$ ./ct-ng build # this take a while
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180215.222532
<snip>
[INFO ]  Build completed at 20180215.232318
[INFO ]  (elapsed: 57:46.00)
[INFO ]  Finishing installation (may take a few seconds)...
[57:46] / imac-van-gerard:crosstool-NG gerard$ 
```

NOTE : Juggling commands I probably did not build the latest xtensa version.

Instead of the command on the Espressif site:
```
git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git
```
I used the command on the v3.pubpub.org
```
git clone -b esp108-1.21.0 git://github.com/jcmvbkbc/crosstool-NG.git
```

gerardwr, 2018-02-15:
@muekno 

I tested the built toolchain in my Arduino environment, to see if the "CC1 error" I had before was gone. Normally the compile would stop immediately wiith the CC1 eroor, but now it started compiling. Nice.

But I got this error at the end of the compile :-(
```
/Volumes/ctng/crosstool-NG/.build/src/gcc-5.1.0/libstdc++-v3/libsupc++/del_op.cc:46:(.text._ZdlPv+0x5): dangerous relocation: call8: call target out of range: free
collect2: error: ld returned 1 exit status
exit status 1
```

So, not here yet, but quite some progress. Will also try tomorrow to built the latest version.

Could you keep us posted?

muekno, 2018-02-16:
Did you try to run the program, to see the compilation ist OK?
Thanks again for your work.
For the moment I hang still at the point creating the virtual volume, what definitly worked 3 days before.

gerardwr, 2018-02-16:
@muekno 
YES, it compiles OK :-)

I repeated the procedure but this time used the latest XTENSA as described on the Espressif site
```
git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git
```
Then copied the created xtensa-esp32-elf directory from the virtual volume into my Arduino hardware directory. The Arduino compile finished succesfully, no more "CC1 error". Probably works in IDF too, but did not try that. 

Added the log of this latest run below, starting with the creating of the link to the virtual volume. Maybe it helps you.
[untitled text 11.txt](https://github.com/espressif/esp-idf/files/1730730/untitled.text.11.txt)

gerardwr, 2018-02-16:
@muekno 

Final update:
- As mentioned before also Arduino compiles OK, and now also checked that sketches run without problem
- Installed ESP-IDF and with the local built toolchain it compiles OK. Output from Hello-World example:
```
imac-van-gerard:hello_world gerard$ make
CC build/bootloader/bootloader_support/src/bootloader_clock.o
CC build/bootloader/bootloader_support/src/bootloader_flash.o
<many similar lines deleted>
LD build/hello-world.elf
esptool.py v2.2.1
To flash all build output, run 'make flash' or:
python /Users/gerard/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /Users/gerard/esp/hello_world/build/bootloader/bootloader.bin 0x10000 /Users/gerard/esp/hello_world/build/hello-world.bin 0x8000 /Users/gerard/esp/hello_world/build/partitions_singleapp.bin
imac-van-gerard:hello_world gerard$ 
```

dikkehenk, 2019-04-29:
Hi Wulph111,
I do not have enough knowledge to build a Setup Toolchain for Mac OS from Scratch. Is it possible to get the builded code from Wulph111 and instructions how to implement this in the Arduino IDE. I have almost the same computer set-up Mac Pro 1,1 running Yosemite (MacOS 10.10.5).

Hope you can answer
Henk

gerardwr, 2019-04-29:
Hallo Henk,

I built the toolchain to run on my 2007 IMac, maybe it works for you too. Was running fine on ElCapitan.

I'm happy to sent you the toolchain folder, but once zipped it's nearly 52MB. Tried to share via Dropbox, but failed to create a working link. Wetransfer.com maybe, but this needs a mailadres.

Any ideas?

Groetjes,
Gerard.

dikkehenk, 2019-04-29:
Thanks for your offer.
Sure I would like to try the toolchain you have built. My email address is dikkehenk@zonnet.nl <mailto:dikkehenk@zonnet.nl>

If you can please give instructions to implement it in the Arduino IDE 1.8.9

After a long time I upgraded my old Mac Pro 1.1 2006 from OSX 10.6.6 to Yosemite 10.10.5 especially to write and upload sketches into the ESP32. Until now I am using a Windows 7 laptop aside but I do not like that. Sadly I am still at square one. Hope there will be a solution.

thanks
Henk

kan ook in het Nederlands misschien.

wulph111, 2019-04-29:
Unfortunately no, I have needed to replace that old system for other
reasons and I have no idea how to incorporate anything into the Arduino
toolkit.
Best wishes.

igrr, 2019-06-20:
Sorry, forgot to update this issue. Master branch now uses an updated toolchain (https://github.com/espressif/esp-idf/commit/ca323fbaad540cf66cb267c0f749ad24f3a9b211), which is built with support for OSX 10.8 (MACOS_DEPLOYMENT_TARGET=10.8).

Since arduino-esp32 tracks IDF release branches, it might need some time before the new toolchain appears in the Arduino distribution.

closed 2019-06-20 by igrr