espressif/esp-idf #1489: ESP32-OTA partition table issue

dwitamghosh, 2018-01-11:
"Error "ota_begin error err=0x104"
If you see this error then check that the configured (and actual) flash size is large enough for the partitions in the partition table. The default "two OTA slots" partition table only works with 4MB flash size. To use OTA with smaller flash sizes, create a custom partition table CSV (look in components/partition_table) and configure it in menuconfig."
can anyone explain what I have to do step by step. I found the same problem while running the ota example and can't make out any solution.
Thanx in advance.

chegewara, 2018-01-12:
Maybe this will help:
http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html

Just for percuation, are you working with esp-idf or with arduino-ide?

dwitamghosh, 2018-01-12:
I am working with esp-idf in ubuntu. I already followed that tutorial and now that issue had been solved. I just copy the partitions_two_ota.csv is ota folder and then it builds successfully. But,still the main error is coming..
I (467) ota: Starting OTA example...
I (467) ota: Running partition type 0 subtype 0 (offset 0x00010000)
I (707) wifi: n:2 0, o:1 0, ap:255 255, sta:2 0, prof:1
I (1687) wifi: state: init -> auth (b0)
I (1697) wifi: state: auth -> assoc (0)
I (1707) wifi: state: assoc -> run (10)
I (1727) wifi: connected with testing, channel 2
I (2657) event: sta ip: 192.168.43.158, mask: 255.255.255.0, gw: 192.168.43.1
I (2657) ota: Connect to Wifi ! Start to Connect to Server....
I (2657) ota: Server IP: 127.0.0.1 Server Port:8070
E (2667) ota: Connect to server failed! errno=104
E (2667) ota: Connect to http server failed!
E (2677) ota: Exiting task due to fatal error...
I (4707) wifi: pm start, type:0

chegewara, 2018-01-12:
You are trying to connect to itself. It wont work
`I (2657) ota: Server IP: 127.0.0.1 Server Port:8070`

dwitamghosh, 2018-01-12:
ok. I think I understand the problem now.
actually, i am using ubuntu in vmware in my windows pc. If a simple http server with a port number is running in ubuntu, how can i access to that server outside from my laptop. That will be my exact solution.

chegewara, 2018-01-12:
Just find IP on this VM by running command ifconfig in terminal.

dwitamghosh, 2018-01-12:
yes ..i do that. but, the problem arises that when i  try to access the vmware localhost by the vm ip from my mobile while both the windows and mobile are in same wifi, i am getting error..

dwitamghosh, 2018-01-12:
if and only the above way is possible,then only my esp32 can access the bin file in localhost

negativekelvin, 2018-01-12:
https://kb.vmware.com/s/article/2006955

dwitamghosh, 2018-01-12:
thank you so much...let me try this...

chegewara, 2018-01-12:
I am using vbox with bridge mode in ubuntu VM machine, so i dont have such issues.

dwitamghosh, 2018-01-12:
thank you so much...the link you provided, I cant follow that steps but as you told to use bridge network..it works fine...thank you so much..thank you. I can access the data outside from my phn.

closed 2018-01-12 by dwitamghosh

Abdul-Samad-Saleem, 2018-10-09:
> Maybe this will help:
> http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html
> 
> Just for percuation, are you working with esp-idf or with arduino-ide?

Any guidance on how to do in arduino-ide?. I have generated a new bin file suing the gen_esp32part.py but how do I actually update the device?