espressif/esp-idf #1490: sd_card interfacing error

cryarchy, 2018-01-11:
I am trying to run the example provided with **esp-idf** at *examples/storage/sd_card*. I have uncommented the `#define USE_SPI_MODE` line. After flashing, running `make monitor` results in the following output:
```
I (0) cpu_start: App cpu up.
I (209) heap_init: Initializing. RAM available for dynamic allocation:
I (216) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (222) heap_init: At 3FFB2A20 len 0002D5E0 (181 KiB): DRAM
I (228) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (234) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (241) heap_init: At 4008929C len 00016D64 (91 KiB): IRAM
I (247) cpu_start: Pro cpu start user code
I (265) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (267) example: Initializing SD card
I (267) example: Using SPI peripheral
I (267) gpio: GPIO[13]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Name: APPSD
Type: SDSC
Speed: default speed
Size: 1904MB
CSD: ver=0, sector_size=512, capacity=3899392 read_bl_len=10
SCR: sd_spec=2, bus_width=5
I (527) example: Opening file
E (527) sdspi_host: sdspi_host_start_command: cmd=24 error=0x108
E (537) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (537) diskio_sdmmc: sdmmc_write_blocks failed (264)
E (547) sdspi_host: sdspi_host_start_command: cmd=24 error=0x108
E (557) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x108
E (557) diskio_sdmmc: sdmmc_write_blocks failed (264)
I (567) example: File written
I (567) example: Renaming file
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
E (5577) sdspi_host: sdspi_host_start_command: cmd=24 error=0x107
E (5587) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (5587) diskio_sdmmc: sdmmc_write_blocks failed (263)
E (5597) example: Rename failed
```

I have tried adding the recommended external pull-up resistors for *SD* mode to no avail. The hardware connection is as recommended. i.e:
```
#define PIN_NUM_MISO 2
#define PIN_NUM_MOSI 15
#define PIN_NUM_CLK  14
#define PIN_NUM_CS   13
```
What may be the cause of the failure?

SebiTimeWaster, 2021-10-20:
i have the exact same problem with 1-wire SPI.

adokitkat, 2021-11-09:
Hello @cryarchy and @SebiTimeWaster,

could you please tell me what ESP32 model, board and SD card are you using?

@cryarchy 

> I have tried adding the recommended external pull-up resistors for _SD_ mode to no avail. 

SD mode (SDIO/SDMMC) uses different protocol and GPIO pins number than SPI mode.

adokitkat, 2021-11-28:
I am closing this issue due to no response. Feel free to reopen when needed.

closed 2021-11-28 by adokitkat