espressif/esp-idf #1491: loop reboot on register_chipv7_phy

xarrange, 2018-01-11:
What could be a possible problem?:

```
D (161) BTDM_INIT: .bss initialise [0x3ffbdb28] - [0x3ffc0000]
I  (161) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
D (411) phy_init: loading PHY init data from partition at offset 0xf000
D (411) phy_init: PHY data partition validated
D (411) nvs: nvs_open_from_partition phy 0
D (421) nvs: nvs_get cal_version 4
V (421) phy_init: phy_get_rf_cal_version: 366

D (421) nvs: nvs_get_str_or_blob cal_mac
D (431) nvs: nvs_get_str_or_blob cal_data
D (441) nvs: nvs_close 1
V (441) phy_init: register_chipv7_phy, init_data=0x3ffd8390, cal_data=0x3ffd7c14, mode=0

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
cconfigsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5792
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13804
entry 0x40079030

```

negativekelvin, 2018-01-12:
Try make erase_flash to remove old calibration data

xarrange, 2018-01-12:
Thanks! 
it works now

closed 2018-01-12 by xarrange