espressif/esp-idf #1492: SPI RAM abort on startup with esp_event_loop_init / vPortCPUInitializeMutex

markwj, 2018-01-12:
I tried setting the "SPI RAM access method" to "Make RAM allocatable using malloc() as well", and reducing "Maximum malloc() size, in bytes, to always put in internal memory" to 32 bytes, but get this crash on startup:

```
#0  0x4008dfbc in invoke_abort () at /Users/mark/esp3o/esp-idf/components/esp32/./panic.c:139
#1  0x4008e0b2 in abort () at /Users/mark/esp3o/esp-idf/components/esp32/./panic.c:148
#2  0x4008975e in vPortCPUInitializeMutex (mux=0x3f800238) at /Users/mark/esp3o/esp-idf/components/freertos/./port.c:303
#3  0x400899fc in xQueueGenericReset (xQueue=0x3f8001e8, xNewQueue=1) at /Users/mark/esp3o/esp-idf/components/freertos/./queue.c:270
#4  0x40089a78 in prvInitialiseNewQueue (uxQueueLength=32, uxItemSize=48, pucQueueStorage=0x3f800240 '\376' <repeats 200 times>..., ucQueueType=0 '\000', pxNewQueue=0x3f8001e8)
    at /Users/mark/esp3o/esp-idf/components/freertos/./queue.c:441
#5  0x40089ac9 in xQueueGenericCreate (uxQueueLength=32, uxItemSize=48, ucQueueType=0 '\000') at /Users/mark/esp3o/esp-idf/components/freertos/./queue.c:408
#6  0x40162490 in esp_event_loop_init (cb=<optimized out>, ctx=0x3ffb4980 <MyEvents>) at /Users/mark/esp3o/esp-idf/components/esp32/./event_loop.c:103

(gdb) frame 2
#2  0x4008975e in vPortCPUInitializeMutex (mux=0x3f800238) at /Users/mark/esp3o/esp-idf/components/freertos/./port.c:303
303	    configASSERT(esp_ptr_internal((const void *) mux));
```

Presumably that is esp_event_loop_init trying to create a FreeRTOS queue, and then objecting because it is not in internal RAM? It seems that the ESP IDF framework is not fully working with SPI RAM yet (due to inherent limitations). It would be better if the memory to be allocated must be internal, it is specifically allocated as internal (rather than just rely on a generic malloc).

OtherCrashOverride, 2018-01-12:
I encountered a similar issue.  A mutex can not reside in PSRAM.  However, there is currently no way to prevent it.

markwj, 2018-01-17:
Note:

https://www.esp32.com/viewtopic.php?f=13&t=4197

Re: Using SPI RAM from C++ programs
Post by ESP_Sprite » Fri Jan 12, 2018 2:53 pm

There's a bug we're working on where a structure containing a spinlock can sometimes be allocated in SPI RAM, and as such will not work correctly. We're at the moment working on a solution for that; a fix should appear in esp-idf master soon.

paulreimer, 2018-02-08:
Just to check, but is this (recent) commit related to the fix for this issue?
https://github.com/espressif/esp-idf/commit/16de6bff245dec5e63eee994f53a08252be720d4

mahavirj, 2018-02-09:
@paulreimer Yes.

Alvin1Zhang, 2019-06-28:
@markwj Thanks for reporting the issue, would you help share if any updates of this issue? Has your issue been resolved? Thanks.

milamber-ls, 2019-08-10:
is there an update on this?

suda-morris, 2019-12-16:
@milamber-ls @markwj  Does the issue still exists in your project? Or you just encounter another kind of PSRAM issue? We tried as your configuration and run a basic Ethernet DHCP example, no error occurred.

closed 2020-05-06 by suda-morris