espressif/esp-idf #1496: ota data partition invalid, falling back to factory

Rajkumar181, 2018-01-12:
Hi,
I am try to do increase the flash area then i am getting this error " E (94) boot: ota data partition invalid, falling back to factory"

any suggestion for this.

I (30) boot: ESP-IDF v3.1-dev-69-g8688f0ec0-dirty 2nd stage bootloader
I (30) boot: compile time 17:15:24
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (75) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  3 factory          factory app      00 00 00010000 00200000
I (90) boot: End of partition table
E (94) boot: ota data partition invalid, falling back to factory

negativekelvin, 2018-01-12:
There are no ota app partitions so you don't need otadata and booting from factory is correct

dwitamghosh, 2018-01-12:
you can use this document.
http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html

chegewara, 2018-01-12:
Probably you changed partition table and you still have old data in otadata. Just `make erase_flash` and flash it again

FayeY, 2018-01-23:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-23 by FayeY