espressif/esp-idf #1497: write aws certificate in flash

tdesmet, 2018-01-12:
I want to provision a whole set of devices each with their own aws certificate.
But I do not want to recompile the code for each device.

is it possible to use the flash tool to write the certificate file directly to flash?

Whats the best way to handle this scenario, this problem is a common I suppose, how else do manufacturers build iot devices with different ids, mac's etc? Some tips, documentation, blogs, or whatever are welcome.

chmorgan, 2018-01-19:
@tdesmet what about storing in nvs?

shirish47, 2018-02-05:
Yea I am also looking for such thing. we have over project done. but if we want to scale it's difficult as I would have to generate certs(which can be automated with scripts) and compile it with the code. But I think its possible to put it in the filesystem in ESP32. and then from code load it. This would also help do OTA securely from AWS as we can have a flag data of the firmware version. which if changed then the device would download new firmware from the certain topic or AWS services.

I am just not sure which library to use for the internal filesystem? looking at the docs.

@chmorgan NVS suits for small data. keys and values of small strings. but certificates are larger I think.

chegewara, 2018-02-05:
By default nvs partition is small, by its not the problem to increase its size to keep certificates and some other data. Certificates can be stored as blob.

tdesmet, 2018-02-05:
I don't want to use ble or wifi to write the certificate.
I want to flash it on the chip during production while we are flashing the application image. 

I guess we could create a spiffs image for each device and flash that using the esp tool, I have no idea on how to flash a file to nvs...

shirish47, 2018-02-06:
Yes, I looked into partition creation and understood everything hopefully. I want to  do this:

1) to have a Secure connection with AWS IoT and relay Data from BLE nodes. (done)
2) have OTA update facility with secure connection or maybe not secure. (I don't have any idea how to do this securely with AWS IoT certs as I don't think I should do it over MQTT)
3) and certs be will NOT be compiled with firmware and would separately reside in SPIFFS. So program after restarting can use Certs as it is.

chegewara, 2018-02-06:
You dont need factory partition or 2 OTA partitions.
https://esp32.com/viewtopic.php?f=18&t=3133&p=17597&hilit=esp_partition_find#p17569

https://esp32.com/viewtopic.php?f=2&t=3775&p=17333&hilit=esp_partition_find#p17229

http://esp-idf.readthedocs.io/en/latest/api-reference/storage/spi_flash.html#_CPPv218esp_partition_find20esp_partition_type_t23esp_partition_subtype_tPKc

shirish47, 2018-02-06:
well, thanks for info @chegewara. currently, I was working to provide SPIFFS to my esp-if(I am using an older version of IDF as my code is completely built on older 2.something). I just git spiffs to components and after some struggle I made it work.

my current partition is
```
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x4000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x148000,
ota_0,    app,  ota_0,   ,        0x148000,
ota_1,    app,  ota_1,   ,        0x148000,
spiffs,   data, spiffs,  ,        0x7FFF,

``` 
I need SPIFFS  to store AWS certs and some other data.
at one point my code was not working after boot (though nothing crashed) and would sit doing nothing.
then I added "app, factory" and it started working I suppose boot was looking directly at OTA_0 and showed some magic number error.

Now its able to boot show SPIFFS and read and write files.

Yes, I think I want a default app to just download the firmware from say Amazon S3(when devices will be deployed to customers). 

now I think I should put factory app as firmware updater(it should be executed if OTAs detect that they need an update) and OTA_0 or OTA_1 would have latest firmware which would also contain code to detect firmware needs to update, then how should I switch to factory app? or should I be able to do updates from any of the partition? 
I want to reduce factory app size so I can put more of necessary firmware and thus wants the OTA updating part to the factory. this would require to switch back to the factory every time a new firmware is available.  

I hope you understand my problem. :)
cheers

chegewara, 2018-02-06:
You can switch to factory partition if update is required (detected by app in OTA patition). You can do this with this command:
http://esp-idf.readthedocs.io/en/v2.1.1/api-reference/storage/spi_flash.html#_CPPv218esp_partition_find20esp_partition_type_t23esp_partition_subtype_tPKc

with those parameters:
http://esp-idf.readthedocs.io/en/v2.1.1/api-reference/storage/spi_flash.html#_CPPv222ESP_PARTITION_TYPE_APP
http://esp-idf.readthedocs.io/en/v2.1.1/api-reference/storage/spi_flash.html#_CPPv233ESP_PARTITION_SUBTYPE_APP_FACTORY

shirish47, 2018-02-06:
@chegewara I see the command you are showing is to find the specific partition.  how can I switch to factory firmware from any OTA currently running? I think I need to change the bootloader. such as update some flag in NVS and restart and bootloader detects that flag and starts factory firmware which downloads the latest firmware say OTA_1 partition.  I hope you understand now.

chegewara, 2018-02-06:
@shirish47 You dont need to change bootloader code. All you need is to start this code to restart esp32 from factory partition:
```
    const esp_partition_t *partition;
    esp_partition_iterator_t it = esp_partition_find(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_FACTORY, "factory");
	partition = (esp_partition_t *) esp_partition_get(it);
	if(partition != NULL){
		esp_ota_set_boot_partition(partition);
	}
    esp_partition_iterator_release(it);
    esp_restart();
```

shirish47, 2018-02-06:
@chegewara this is what I think should work.
```

const esp_partition_t *factory_partition = NULL;

//find factory partition
factory_partition = esp_partition_find(..., "factory");

//configure for booting in factory (any other way of setting boot partion??)
 err = esp_ota_set_boot_partition(factory_partition);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "esp_ota_set_boot_partition failed! err=0x%x", err);
        task_fatal_error();
    }
 ESP_LOGI(TAG, "Prepare to restart system!");
 esp_restart();

```
then factory code would do the OTA part.

shirish47, 2018-02-06:
@chegewara Thanks for all that support now its more clear how I should go ahead. It feels great to work with people like you always ready to help. 
Though I have a long way to go as I am working with backend developer. I also have to now integrate this to my current firmware and make another factory app for this. 

You simplified my work. 🥇 👍

mahavirj, 2018-10-23:
@tdesmet Storing to custom NVS partition is recommended way. NVS partition generator utility can help in this regard, please refer https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/storage/nvs_partition_gen.html?highlight=nvs#nvs-partition-generator-utility (Recently this also comes with additional security layer of encryption support).

mahavirj, 2018-10-25:
If you still run into any trouble with certificate storage, please feel free to raise new issue. Closing this one.

closed 2018-10-25 by mahavirj