espressif/esp-idf #1499: Strange memory leak in lwip stack

sulyokg, 2018-01-12:
I've been developing a central heating controller and monitoring system for my home. My new buddy is the ESP32, which is ideal solution for this. My system consists of several nodes, and one apache webserver on a PC for data storage and centralized communication.
I'm developing for ESP32 with ESP-IDF toolchain in C++ language. The test communication method with webserver is working properly, which is called in 10 seconds interval, but the free heap is lower and lower after each call. The communication method is based on the https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_request example.
I can't figure out, where is the source of memory leak, somewhere in the lwip stack. The test log (https://pastebin.com/raw/yJnGEFDf) shows, the memory leak is "stabilised" at 580-584 bytes per Wlan::data() method call. When the system runs continuously, after several hours the system runs out of memory, that cause a reboot. This unexpected reboot breaks the IIC communication with an OLED display (which is not have a reset pin), and the screen is shifted over horizontally and vertically until a power cycle.
While testing, and creating the linked log, no other tasks are running (eg. display, button, temp sensor tasks).

Here is the Wlan::data() method:
```
void Wlan::data(char *dat)
{
	uint32_t heapc = xPortGetFreeHeapSize();
	printf("\r\n\r\n\r\n");
    ESP_LOGI(TAG, "free heap at start: %d, ms = %d", heapc, NETDB_ELEM_SIZE);

    struct addrinfo hints;
    hints.ai_family = AF_INET;
    hints.ai_socktype = SOCK_STREAM;

    struct addrinfo *res;
    struct in_addr *addr;
    int s, r;
    char recv_buf[64];

    char *send_buf = (char*) malloc(2048);
    if (send_buf == NULL)
    {
    	ESP_LOGE(TAG, "send_buf malloc failed");
    	return;
    }
	char *send_data = (char*) malloc(1500);
    if (send_data == NULL)
    {
    	ESP_LOGE(TAG, "send_data malloc failed");
    	free(send_buf);
    	return;
    }
    ESP_LOGI(TAG, "free heap after malloc: %d", xPortGetFreeHeapSize());

	int err = getaddrinfo("192.168.2.2", "80", &hints, &res);

	        if(err != 0 || res == NULL) {
	            ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
	            vTaskDelay(1000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }

	        // Code to print the resolved IP.
	        //   Note: inet_ntoa is non-reentrant, look at ipaddr_ntoa_r for "real" code
	        addr = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
	        //ESP_LOGI(TAG, "DNS lookup succeeded. IP=%s", inet_ntoa(*addr));

	        s = socket(res->ai_family, res->ai_socktype, 0);
	        if(s < 0) {
	            ESP_LOGE(TAG, "... Failed to allocate socket.");
	            freeaddrinfo(res);
	            vTaskDelay(1000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }
	        //ESP_LOGI(TAG, "... allocated socket");
	        ESP_LOGI(TAG, "free heap after socket: %d", xPortGetFreeHeapSize());

	        if(connect(s, res->ai_addr, res->ai_addrlen) != 0) {
	            ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
	            close(s);
	            freeaddrinfo(res);
	            vTaskDelay(4000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }

	        ESP_LOGI(TAG, "free heap after connect: %d", xPortGetFreeHeapSize());
	        freeaddrinfo(res);

	        strcpy(send_buf, REQUEST);
	        strcpy(send_data, "auth=6hGThH634FF");
	        if (strlen(dat))
	        {
	        	strcat(send_data, "&");
	        	strcat(send_data, dat);
	        }

	        sprintf(recv_buf, "Content-Length: %d\r\n\r\n", strlen(send_data));
	        strcat(send_buf, recv_buf);
	        strcat(send_buf, send_data);
	        printf("Request:\r\n-----------------------------\r\n");
        	printf(send_buf);
        	printf("\r\n --- data sent: %d bytes\r\n", strlen(send_buf));
	        if (write(s, send_buf, strlen(send_buf)) < 0) {
	            ESP_LOGE(TAG, "... socket send failed");
	            close(s);
	            vTaskDelay(4000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }
	        ESP_LOGI(TAG, "... socket send success");
	        ESP_LOGI(TAG, "free heap after send: %d", xPortGetFreeHeapSize());

	        struct timeval receiving_timeout;
	        receiving_timeout.tv_sec = 1;
	        receiving_timeout.tv_usec = 0;
	        if (setsockopt(s, SOL_SOCKET, SO_RCVTIMEO, (const void*) &receiving_timeout,
	                sizeof(receiving_timeout)) < 0) {
	            ESP_LOGE(TAG, "... failed to set socket receiving timeout");
	            close(s);
	            vTaskDelay(4000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }
	        //ESP_LOGI(TAG, "... set socket receiving timeout success");

	        // Read HTTP response
	    	char *response = (char*) malloc(4096);
	        ESP_LOGI(TAG, "free heap after recv malloc: %d", xPortGetFreeHeapSize());
	    	uint8_t iscontent = 0;
	    	uint8_t prevlf = 0;
	    	uint16_t j = 0;

	        do {
	            bzero(recv_buf, sizeof(recv_buf));
	            r = read(s, recv_buf, sizeof(recv_buf)-1);
	            for(int i = 0; i < r; i++) {
	                //putchar(recv_buf[i]);
	                if (j < 4096) response[j++] = recv_buf[i];
	            }
	        } while(r > 0);
	        response[j] = 0;
	        ESP_LOGI(TAG, "free heap after recv: %d", xPortGetFreeHeapSize());
	        char *content = strstr(response, "\r\n\r\n");
	        ESP_LOGI(TAG, "free heap after strstr: %d", xPortGetFreeHeapSize());
	        printf("Response:\r\n-----------------------------\r\n");
	        printf(response);
        	printf("\r\n --- data recv: %d bytes\r\n", strlen(response));
	        if (content)
	        {
		        content += 4;
		        printf("\r\nContent:\r\n-----------------------------\r\n");
	        	printf(content);
	        	printf("\r\n");

		    	JsonObject json = JSON::parseObject(content);
		    	if (json.isValid())
		    	{
			    	temp.out = json.getDouble("out");
			    	temp.wanted = temp.setwanted = json.getDouble("wanted");
			    	temp.temperature = json.getDouble("room");
			    	flameicon = json.getInt("boiler");
			    	timestamp = json.getInt("time");
			    	thermomode = json.getInt("mode");
		    	} else ESP_LOGE(TAG, "Invalid json string returned: %s", content);
		        ESP_LOGI(TAG, "free heap after json: %d", xPortGetFreeHeapSize());
	        }
	        free(response);
	        ESP_LOGI(TAG, "... done reading from socket. Last read return=%d errno=%d", r, errno);
	close(s);
	free(send_buf);
	free(send_data);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    ESP_LOGI(TAG, "free heap at end: %d, memory leak: %d bytes", xPortGetFreeHeapSize(), heapc - xPortGetFreeHeapSize());
}
```

stickbreaker, 2018-01-13:
@sulyokg 
the only thing I found, is a potential buffer overwrite:
```c++

do {
  bzero(recv_buf, sizeof(recv_buf));
  r = read(s, recv_buf, sizeof(recv_buf)-1);
  for(int i = 0; i < r; i++) {
//putchar(recv_buf[i]);
    if (j < 4096) response[j++] = recv_buf[i];
    }
  } while(r > 0);
response[j] = 0;
```
if j==4096 bad things happen. 

I would paste it with `ESP_LOGI("free heap")`  and order your `close()`, `free()` and scoping such that you can anticipate the freeHeap values.

Like putting a `ESP_LOGI()` before the `JsonObject` declaration and a matching `ESP_LOGI()` after `JsonObject` falls out of scope. 

like this:
```c++
void Wlan::data(char *dat)
{
uint32_t heapc = xPortGetFreeHeapSize();
printf("\r\n\r\n\r\n");
ESP_LOGI(TAG, "free heap at start: %d, ms = %d", heapc, NETDB_ELEM_SIZE);

struct addrinfo hints;
hints.ai_family = AF_INET;
hints.ai_socktype = SOCK_STREAM;

struct addrinfo *res;
struct in_addr *addr;
int s, r;
char recv_buf[64];

char *send_buf = (char*) malloc(2048);
if (send_buf == NULL){
  ESP_LOGE(TAG, "send_buf malloc failed");
  return;
  }
char *send_data = (char*) malloc(1500);
if (send_data == NULL){
  ESP_LOGE(TAG, "send_data malloc failed");
  free(send_buf);
  return;
  }
ESP_LOGI(TAG, "free heap after malloc: %d, espected: %d", xPortGetFreeHeapSize(),(heapc-(2048+4+1500+4));

uint32_t getAddrInfo_size = xPortGetFreeHeapSize();

int err = getaddrinfo("192.168.2.2", "80", &hints, &res);

getAddrInfo_size -= xPortGetFreeHeapSize();

ESP_LOGI(TAG, "free heap after getAddr: %d", xPortGetFreeHeapSize());

if(err != 0 || res == NULL) {
  ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
  vTaskDelay(1000 / portTICK_PERIOD_MS);
//continue;
  free(send_buf);
  free(send_data);
  return;
  }

// Code to print the resolved IP.
//   Note: inet_ntoa is non-reentrant, look at ipaddr_ntoa_r for "real" code
addr = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
//ESP_LOGI(TAG, "DNS lookup succeeded. IP=%s", inet_ntoa(*addr));

uint32_t socket_size = xPortGetFreeHeapSize();
s = socket(res->ai_family, res->ai_socktype, 0);
socket_size  -= xPortGetFreeHeapSize();

if(s < 0) {
  ESP_LOGE(TAG, "... Failed to allocate socket.");
  freeaddrinfo(res);
  vTaskDelay(1000 / portTICK_PERIOD_MS);
//continue;
  free(send_buf);
  free(send_data);
  return;
  }
//ESP_LOGI(TAG, "... allocated socket");
ESP_LOGI(TAG, "free heap after socket: %d, size=%d", xPortGetFreeHeapSize(),socket_size);

if(connect(s, res->ai_addr, res->ai_addrlen) != 0) {
  ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
  close(s);
  freeaddrinfo(res);
  vTaskDelay(4000 / portTICK_PERIOD_MS);
//continue;
  free(send_buf);
  free(send_data);
  return;
  }

ESP_LOGI(TAG, "free heap after connect: %d", xPortGetFreeHeapSize());
uint32_t res_test = xPortGetFreeHeapSize();
freeaddrinfo(res);
res_test = xPortGetFreeHeapSize() - res_test;
if(res_test!=getAddrInfo_size){
  ESP_LOGI(TAG,"getaddrinfo free size mis-match %d != %d",res_test,getAddrInfo_size);
  }

--- cut

if (content){
  content += 4;
  printf("\r\nContent:\r\n-----------------------------\r\n");
  printf(content);
  printf("\r\n");

  ESP_LOGI(TAG, "free heap before json: %d", xPortGetFreeHeapSize());
  JsonObject json = JSON::parseObject(content);
  if (json.isValid()){
    temp.out = json.getDouble("out");
    temp.wanted = temp.setwanted = json.getDouble("wanted");
    temp.temperature = json.getDouble("room");
    flameicon = json.getInt("boiler");
    timestamp = json.getInt("time");
    thermomode = json.getInt("mode");
    } else ESP_LOGE(TAG, "Invalid json string returned: %s", content);
  }
ESP_LOGI(TAG, "free heap after json: %d", xPortGetFreeHeapSize());
free(response);
ESP_LOGI(TAG, "free heap after free(response): %d", xPortGetFreeHeapSize());

--- cut
```

chuck.

negativekelvin, 2018-01-13:
JSON::deleteObject ?

stickbreaker, 2018-01-13:
@negativekelvin Shouldn't the object destructor handle that? (going out of scope).   I agree `JsonObject` could be the culprit. That's why I wanted more granularity on the `xPortGetFreeHeapSize()`.

It is either `JsonObject` or `connect()` or `getaddrinfo()` or `socket()`  or `close()`.

I vote for `JsonObject` not cleaning up after itself.

Chuck.

igrr, 2018-01-13:
Would heap tracing be of help here? With it you know which call stack has allocated memory. https://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html#heap-tracing

negativekelvin, 2018-01-13:
@stickbreaker https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/JSON.cpp

sulyokg, 2018-01-13:
@negativekelvin JSON::DeleteObject() recovered about 300 bytes, thanks!

@igrr heap tracing is on.

@stickbreaker
I've modified the code with your logging suggestions, new log is here: https://pastebin.com/raw/mLfWKwdn

New test code:
```c
void Wlan::data(char *dat)
{
	uint32_t heapc = xPortGetFreeHeapSize();
	printf("\r\n\r\n\r\n");
    ESP_LOGI(TAG, "free heap at start: %d", heapc);
    ESP_ERROR_CHECK( heap_trace_start(HEAP_TRACE_LEAKS) );

    struct addrinfo hints;
    hints.ai_family = AF_INET;
    hints.ai_socktype = SOCK_STREAM;

    struct addrinfo *res;
    struct in_addr *addr;
    int s, r;
    char recv_buf[64];

    char *send_buf = (char*) malloc(2048);
    if (send_buf == NULL)
    {
    	ESP_LOGE(TAG, "send_buf malloc failed");
    	return;
    }
	char *send_data = (char*) malloc(1500);
    if (send_data == NULL)
    {
    	ESP_LOGE(TAG, "send_data malloc failed");
    	free(send_buf);
    	return;
    }
    ESP_LOGI(TAG, "free heap after malloc: %d, expected: %d", xPortGetFreeHeapSize(),(heapc-(2048+4+1500+4)));

    uint32_t getAddrInfo_size = xPortGetFreeHeapSize();
	int err = getaddrinfo("192.168.2.2", "80", &hints, &res);
	getAddrInfo_size -= xPortGetFreeHeapSize();
	ESP_LOGI(TAG, "free heap after getAddr: %d, size = %d", xPortGetFreeHeapSize(), getAddrInfo_size);

	        if(err != 0 || res == NULL) {
	            ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
	            vTaskDelay(1000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }

	        // Code to print the resolved IP.
	        //   Note: inet_ntoa is non-reentrant, look at ipaddr_ntoa_r for "real" code
	        addr = &((struct sockaddr_in *)res->ai_addr)->sin_addr;
	        //ESP_LOGI(TAG, "DNS lookup succeeded. IP=%s", inet_ntoa(*addr));

	        uint32_t socket_size = xPortGetFreeHeapSize();
	        s = socket(res->ai_family, res->ai_socktype, 0);
	        socket_size  -= xPortGetFreeHeapSize();

	        if(s < 0) {
	            ESP_LOGE(TAG, "... Failed to allocate socket.");
	            freeaddrinfo(res);
	            vTaskDelay(1000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }
	        ESP_LOGI(TAG, "free heap after socket: %d, size=%d", xPortGetFreeHeapSize(),socket_size);

	        uint32_t connect_size = xPortGetFreeHeapSize();
	        if(connect(s, res->ai_addr, res->ai_addrlen) != 0) {
	            ESP_LOGE(TAG, "... socket connect failed errno=%d", errno);
	            close(s);
	            freeaddrinfo(res);
	            vTaskDelay(4000 / portTICK_PERIOD_MS);
	            //continue;
	        	free(send_buf);
	        	free(send_data);
	            return;
	        }
	        connect_size  -= xPortGetFreeHeapSize();

	        ESP_LOGI(TAG, "free heap after connect: %d, size = %d", xPortGetFreeHeapSize(), connect_size);
	        uint32_t res_test = xPortGetFreeHeapSize();
	        freeaddrinfo(res);
	        res_test = xPortGetFreeHeapSize() - res_test;
	        if(res_test!=getAddrInfo_size)
	        {
	        	ESP_LOGI(TAG,"getaddrinfo free size mis-match %d != %d",res_test,getAddrInfo_size);
	        }
	uint32_t close_test = xPortGetFreeHeapSize();
	close(s);
	close_test = xPortGetFreeHeapSize() - close_test;
	if(close_test != socket_size+connect_size)
	{
		ESP_LOGI(TAG,"socket free size mis-match %d != %d",close_test,socket_size+connect_size);
	}
	free(send_buf);
	free(send_data);
    vTaskDelay(500 / portTICK_PERIOD_MS);
    ESP_LOGI(TAG, "free heap at end: %d, memory leak: %d bytes", xPortGetFreeHeapSize(), heapc - xPortGetFreeHeapSize());
    ESP_ERROR_CHECK(heap_trace_stop());
    heap_trace_dump();

}

```

negativekelvin, 2018-01-13:
Are you sure that it still goes to zero?  There are other tasks using heap outside your code.

stickbreaker, 2018-01-13:
@sulyokg in the "readtheDocs" for [heap memory Debugging](http://esp-idf.readthedocs.io/en/latest/api-reference/system/heap_debug.html) It list a couple of false positives:
* The Bluetooth, WiFi, and TCP/IP libraries will allocate heap memory buffers to handle incoming or outgoing data. These memory buffers are usually short lived, but some may be shown in the heap leak trace if the data was received/transmitted by the lower levels of the network while the leak trace was running.
* TCP connections will continue to use some memory after they are closed, because of the TIME_WAIT state. After the TIME_WAIT period has completed, this memory will be freed.

One way to differentiate between “real” and “false positive” memory leaks is to call the suspect code multiple times while tracing is running, and look for patterns (multiple matching allocations) in the heap trace output.

Maybe the only actual leak is in `JsonObject`

Chuck.

stickbreaker, 2018-01-13:
@negativekelvin I just looked at the `JsonObject` code, do you know why there is not a defined destructor? I would have expected:
```c++
class JsonObject{
  public:
    ~JsonObject{JSON::DeleteObject(this);}
}
```

Chuck.

negativekelvin, 2018-01-13:
@stickbreaker you would have to ask @nkolban about that

nkolban, 2018-01-13:
We can't implicitly delete an object ... and here is why.  Consider the following JSON:

```
{
  "a": 123,
  "b": "Xyz",
  "c": {
     "p": 456,
     "q": true
   }
}
```

Now if we imagine parsing this, what we get back is a JsonObject.  All good ... and it *would* be ok to dispose of the underlying data if that object goes out of scope.

However, now imagine:

```
JsonObject jo2 = myJsonObject.getObject("c");
```

Now the `jo2` C++ class instance is a JsonObject that references the "c" member of the original (`myJsonObject`).   This isn't a copy ... this is a reference to the actual data structure itself.  If we deleted the data when `jo2` goes out of scope, what would that mean to the existing `myJsonObject`?

My thinking thus is that underlying data for a JSON object is created through either:

`JSON::createObject()`

or 

`JSON::parseObject()`

Using either of these, one must eventually delete the resulting data ... but other than that, there should be no need.  Happy to discuss further if desired or be persuaded that my thinking is wrong and we can make changes as needed.

stickbreaker, 2018-01-13:
@nkolban ok, I can see that your are using a memory conservation model, instead of duplicating structures you are using references. 

The OP's origional code
```c++
char* content = strstr(response,"/r/n/r/n");
if (content) {
  content += 4;
  printf("\r\nContent:\r\n-----------------------------\r\n");
  printf(content);
  printf("\r\n");
  
  JsonObject json = JSON::parseObject(content);
  if (json.isValid()){
    temp.out = json.getDouble("out");
    temp.wanted = temp.setwanted = json.getDouble("wanted");
    temp.temperature = json.getDouble("room");
    flameicon = json.getInt("boiler");
    timestamp = json.getInt("time");
    thermomode = json.getInt("mode");
    }
  else ESP_LOGE(TAG, "Invalid json string returned: %s", content);
   }
```
When the `parseObject()` function is called shouldn't it reuse the prior allocated memory.
I mean, a JSON object cannot have duplicate leaves?

Does this usage create a memory leak?  

Must the OP have a `JSON::deleteObject(json);` at the end of the `if(json.isValid){}` to plug the memory leak?

Chuck.

sulyokg, 2018-01-13:
`JSON::deleteObject()` is solved my problem, I did not see the wood from the tree :). No more memory leak, no crash, no reboot. After several hours of run, free heap is fluctuating around 127kb, not decreasing to zero.

Json routine at this moment:
```c
 JsonObject json = JSON::parseObject(content);
 if (json.isValid())
 {
	temp.out = json.getDouble("out");
	temp.wanted = json.getDouble("wanted");
	temp.temperature = json.getDouble("room");
	displayVars0.flame.show = json.getInt("boiler");
	timestamp = json.getInt("time");
	thermomode = json.getInt("mode");
	JSON::deleteObject(json);
 } else ESP_LOGE(TAG, "Invalid json string returned: %s", content);
```

Thanks for help! :)

negativekelvin, 2018-01-14:
@stickbreaker what prior allocated memory?  JSON::parseObject parses a string into a JSON tree. https://github.com/DaveGamble/cJSON#parsing-json

stickbreaker, 2018-01-14:
@negativekelvin The base JSON object is a static Object, JsonObject is just a pointer to a node in the JSON tree held by the JSON static object.  When `parseObject()` is called, it add the parsed data to the static JSON object, then returns a pointer to the node that represents this new object. When the JsonObject goes out of scope, only the pointer is freed the json tree still exists in the JSON static object. The node is only pruned when you call `deleteObject()`. I am referring to the memory that was allocated when `parseObject()` was called. I seems to me that consecutive calls to `parseObject()` should result in overwriting the prior equivalent node.  I don't see why JSON should allow duplicate nodes.

My understanding of JSON is shallow.  I may be wrong in my assumptions.

Chuck.

nkolban, 2018-01-14:
The JSON C++ classs found [here](https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/JSON.h) was intended to be a loose wrapper around the underling `cJSON` C library in an attempt to hide some of the details of `cJSON`.    Within the C++ classes there is a class called `JSON` which contains nothing but static methods.  This means that one will never instantiate `JSON` and it will hold no state.  As such, it really isn't able to hold any data so there must be something wrong in the notion "... a pointer to a node in the JSON tree held by the JSON static object.".

Keeping it simple and ignoring arrays for just now, there are two occasions where data is allocated and needs to be freed.  The first is calling `JSON::parseObject` which parses a string and returns a tree of data in memory.    The return from this is a `JsonObject` class instance.  To prevent memory leaks, the returned JsonObject **must** be explicitly deleted by calling `JSON::deleteObject` when the JsonObject is no longer needed.

The second function that allocates storage is `JSON::createObject`.  This also creates and returns a JsonObject object.  Now things get interesting.  This JsonObject must eventually be deleted to prevent memory leaks.  We can call `JSON::deleteObject` and it will be gone.  Great.   However, if we create multiple objects and attach them together in a tree, then deleting the root of the tree will delete **all** the objects in the tree in a recursive fashion.  So for example:

```
JsonObject a = JSON::createObject();
JsonObject b = JSON::createObject();
b.setString("hello", "world");
a.setObject("myChild", b);
...
JSON::deleteObject(a);
```
would *also* delete  the JsonObject created for `b` because it was made part of the tree data by adding it to `a`.  Deleting the tree rooted at `a` deletes all child nodes.

stickbreaker, 2018-01-14:
@nkolban Thank you for the education. I would recommend that some explicit comments be added to the .h at both `createObject()` and `parseObject()` that `JSON::deleteObject()` must be call to release the objects.  The standard scoping rules do not apply. 

Chuck.

closed 2018-01-22 by FayeY