espressif/esp-idf #1500: OTA example strange behavior

chegewara, 2018-01-14:
Few weeks ago i changed a little ota example to resolve IP address from domain name and all worked fine. Now it still working just i have some strange behavior. After reading whole bin file app stuck for about 40 seconds somewhere in here https://github.com/espressif/esp-idf/blob/master/examples/system/ota/main/ota_example_main.c#L269-L272

```
I (22290) ota: Have written image length 1754512
I (62512) ota: Connection closed, all packets received
I (62514) ota: Total Write binary data length : 1754512
D (62514) esp_image: reading image header @ 0x110000
D (62518) esp_image: image header: 0xe9 0x07 0x02 0x02 40081060
V (62524) esp_image: loading segment header 0 at offset 0x110018
```

negativekelvin, 2018-01-14:
Must be stuck at https://github.com/espressif/esp-idf/blob/master/examples/system/ota/main/ota_example_main.c#L253

FayeY, 2018-01-17:
Any update about this issue?

chegewara, 2018-01-17:
Not from my side, i dont know what i can do. Its esp-idf example.

FayeY, 2018-01-17:
Could you provide tcp packet capture for debugging?

chegewara, 2018-01-17:
Not really, i cant configure wireshark in promiscuse mode on my laptop. But this code can be used to recreate it:
https://github.com/chegewara/esp32-ota-with-ble-setup

closed 2018-05-03 by chegewara