espressif/esp-idf #1501: I2S example is wrong at 1st 16bit output

MhageGH, 2018-01-14:
I2S example(i2s_example_main.c) is wrong at 1st 16bit output as the image. It should change output resolution every 5 second. I think there is error concerned with DMA buffer in I2S library according to some own tests.
[image]

tuanpmt, 2018-01-14:
Hello @MhageGH 
You may encounter the same situation here
https://github.com/espressif/esp-idf/issues/300#issuecomment-277605202

> Because i2s tx faster than sinwave/triangle calculation. Please re-calculate to a buffer first, then using i2s_write_bytes

MhageGH, 2018-01-15:
Hello @tuanpmt 
Thank you for your comment. 
However the present i2s_example_main.c already forms as same as the resolved code in #300 (comment). 
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2s/main/i2s_example_main.c
I think it is other problem.

projectgus, 2019-03-14:
Hi @MhageGH ,

Thanks for being patient, sorry that noone got back to you about this issue for such a long time.

Do you happen to still have the I2S test code that produced the waveforms shown in the graphs?

Angus

costaud, 2020-05-26:
I'll close this issue for now, feel free to reopen if you get further information.

closed 2020-05-26 by costaud