espressif/esp-idf #1502: Blufi Username/Password Fields

paulreimer, 2018-01-14:
Hi there,

This topic has been posted quite a lot in the esp32.com forums in some variants, let me summarize the main requests before referencing the existing threads:
- Are there any plans to open-source the Blufi mobile-side code (which implements sending credentials + encryption)? (e.g. source code to recreate `libblufiSDK.a` for iOS, `espblufi-v1.5.1.jar` for android)
- There are unimplemented fields in the blufi spec (`ESP_BLUFI_EVENT_RECV_USERNAME`, `ESP_BLUFI_EVENT_RECV_CA_CERT`, etc) mentioned in the esp-idf, will they be implemented soon?
- How can we extend or add new fields to the Blufi service spec for our own purposes?
- How can we completely re-implement Blufi if needed (e.g. to send other credential types, encrypted securely?) An example app sending with an encryption handshake and a string exchanged, for example.

Requesting docs/code for mobile implementation (iOS, Android Blufi apps):
- https://esp32.com/viewtopic.php?f=13&t=3709
- https://esp32.com/viewtopic.php?f=2&t=3566
- https://esp32.com/viewtopic.php?f=2&t=3407
- https://esp32.com/viewtopic.php?f=13&t=3215
- https://esp32.com/viewtopic.php?f=13&t=978
- https://esp32.com/viewtopic.php?f=2&t=1918
- https://esp32.com/viewtopic.php?f=2&t=631

Extending/Requesting new fields to Blufi service (or re-implementing Blufi completely):
- https://esp32.com/viewtopic.php?f=2&t=3498
- https://esp32.com/viewtopic.php?f=13&t=3333

As far as the forums go, [this one](https://esp32.com/viewtopic.php?f=2&t=3498) seems to be the closest to a working solution (which is on the path of re-implementing Blufi completely), but there is no handshake / key generation listed there, and I believe Blufi does this (and more).

paulreimer, 2018-01-14:
I should note that personally, I need a way to set a username/password and cacert through Blufi (e.g. separate from SSID/password, for example: credentials for a web service). Everything else about Blufi works OK for me, although I would prefer open-source vs. binary blobs.

paulreimer, 2018-01-19:
Any response? Even just "yes, possible", "no, 100% impossible", or "we cannot say yet".

Seems like our choices at this time are to completely re-implement the mobile side of Blufi? Which seems like a lot of work, instead of just adding one username field to the existing Espressif mobile code?

Weijian-Espressif, 2018-02-11:
@paulreimer thanks for your reminder.   Sorry , There is no plan to open encryption part of blufi apk now.

Weijian-Espressif, 2018-03-13:
@all, we will open the source code later.

Weijian-Espressif, 2018-03-19:
@all, blufi ios app has open source code, Android will be later.

Weijian-Espressif, 2018-04-03:
@FayeY please close the issue.

closed 2018-04-03 by FayeY