espressif/esp-idf #1504: ESP-IDF in Windows fails

dwitamghosh, 2018-01-15:
when i run make menuconfig, the following error occures..

> 
make[1]: Entering directory /home/nour/Desktop/iotbits_esp32/esp-idf/tools/kconfig' cc -DCURSES_LOC="<curses.h>" -DLOCALE -c -o mconf.o mconf.c mconf.c:21:20: fatal error: curses.h: No such file or directory #include <curses.h> ^ compilation terminated. make[1]: *** [mconf.o] Error 1 make[1]: Leaving directory/home/nour/Desktop/iotbits_esp32/esp-idf/tools/kconfig

I think i have to use ncurse library, but how can I install it for windows???
Thnx in advance.

xuhongv, 2019-01-22:
Please do action , and have a  try  again： 
`sudo apt-get install git wget make libncurses-dev flex bison gperf python python-serial`

tangqingcai, 2019-03-26:
How about msys32.exe in windows 10 ?

projectgus, 2019-03-26:
@dwitamghosh Sorry for the long delay in getting back to you. Are you still facing this issue with Windows? Normally this means either the Windows MSYS2 installation was incomplete, or you are running the MSYS2 shell via something other than "mingw32.exe". Does the prompt in the MSYS2 shell say "MINGW32" or something else?

@xuhongv apt-get command will only work for Debian/Ubuntu Linux or WSL with Ubuntu subsystem, not for the MSYS2 environment.

@tangqingcai If using ESP-IDF with MSYS2, it's important to run "mingw32.exe" not "msys2.exe".

projectgus, 2019-06-04:
Closing due to a lack of additional information to debug.

closed 2019-06-04 by projectgus