espressif/esp-idf #1508: random ESP not starting

SirRibeiro, 2018-01-15:
Hello,
Forcing the esp_restart sometimes a get the following and ESP crashes, not starting the code

```cpp
I (34021) wifi: state: run -> init (0)
I (34023) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (34024) wifi: pm stop, total sleep time: 0/27742409

I (34036) wifi: flush txq
I (34036) wifi: stop sw txq
I (34036) wifi: lmac stop hw txq
E (34036) wifi: esp_wifi_connect 836 wifi not start
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11804
entry 0x40078cec
W (73) rtc_clk: Bogus XTAL frequency: 0 MHz
I (194) cpu_start: Pro cpu up.
I (194) cpu_start: Single core mode
I (194) heap_init: Initializing. RAM available for dynamic allocation:
I (198) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (204) heap_init: At 3FFBCE98 len 00023168 (140 KiB): DRAM
I (210) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (216) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (223) heap_init: At 4008FF18 len 000100E8 (64 KiB): IRAM
I (229) cpu_start: Pro cpu start user code
```
To start the code once again i just need to power off and on again, and everything works fine until next crash.

Any ideas why is this happening?

Thanks

igrr, 2018-01-15:
Which IDF commit are you using?

SirRibeiro, 2018-01-15:
Is this right?

git log -1

commit 2c95a77cf93781f296883d5dbafcdc18e4389656
Merge: 2c2e7f6 552ba35
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Oct 24 10:02:51 2017 +0800

    Merge branch 'bugfix/ioctl_declaration' into 'master'
    
    add missing ioctl declaration
    
    See merge request !1442

igrr, 2018-01-15:
A fix for lock-up during clock frequency switching was merged just a couple of days after the commit you are using (https://github.com/espressif/esp-idf/commit/6d4ed4ff6c4ca6e3b3bd4af3c7aeabe33570b540 and a few nearby commits). Could you please update IDF to latest master and see if that fixes the issue?

SirRibeiro, 2018-01-15:
Ok, perfect, i will try and then I give the feedback.
Thank you very much for the fast response

SirRibeiro, 2018-01-15:
So I already got some results,
First by doing the remote esp_restart the ESP no longer crashes.
But if I remove and plug the power very fast it crashes once again giving the following output:

```cpp
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11804
entry 0x40078cec
E (30) boot: No bootable app partitions in the partition table
user code done

```
I understand this is some kind of protection on the ESP, but is there a away to force another reboot?
So that the ESP don't say blocked?

Thank you very much, great work!

chegewara, 2018-01-21:
Hi, ive got the same issue. Latest commit, hello-world example used to test. Partition table: 2 partitions (factory and ota). Stuck no matter which partition is used to start app. Logs:
```
Restarting now.
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:8444
load:0x40078000,len:0
load:0x40078000,len:15812
entry 0x4007949c
I (28) boot: ESP-IDF v3.1-dev-241-gf4009b9 2nd stage bootloader
I (28) boot: compile time 05:38:12
I (28) boot: Enabling RNG early entropy source...
D (34) qio_mode: Probing for QIO mode enable...
D (39) qio_mode: Raw SPI flash chip id 0x1640c8
D (44) qio_mode: Manufacturer ID 0xc8 chip ID 0x4016
I (49) qio_mode: Enabling default flash chip QIO
D (54) qio_mode: Initial flash chip status 0x2
D (59) qio_mode: QIO mode already enabled in flash
D (63) qio_mode: Enabling QIO mode...
D (67) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (74) boot: magic e9
D (76) boot: segments 04
D (79) boot: spi_mode 02
D (81) boot: spi_speed 00
D (84) boot: spi_size 02
I (87) boot: SPI Speed      : 40MHz
I (91) boot: SPI Mode       : QIO
I (95) boot: SPI Flash Size : 4MB
D (99) bootloader_flash: mmu set paddr=00000000 count=1
D (104) boot: mapped partition table 0x8000 at 0x3f408000
D (110) flash_parts: partition table verified, 5 entries
I (115) boot: Partition Table:
I (119) boot: ## Label            Usage          Type ST Offset   Length
D (126) boot: load partition table entry 0x3f408000
D (131) boot: type=1 subtype=2
I (134) boot:  0 nvs              WiFi data        01 02 00009000 00004000
D (142) boot: load partition table entry 0x3f408020
D (147) boot: type=1 subtype=0
I (150) boot:  1 otadata          OTA data         01 00 0000d000 00002000
D (157) boot: load partition table entry 0x3f408040
D (162) boot: type=1 subtype=1
I (165) boot:  2 phy_init         RF data          01 01 0000f000 00001000
D (173) boot: load partition table entry 0x3f408060
D (178) boot: type=0 subtype=0
I (181) boot:  3 factory          factory app      00 00 00010000 00130000
D (189) boot: load partition table entry 0x3f408080
D (194) boot: type=0 subtype=10
I (197) boot:  4 ota              OTA app          00 10 00140000 002c0000
I (204) boot: End of partition table
D (219) boot: OTA data offset 0xd000
D (219) bootloader_flash: mmu set paddr=00000000 count=1
D (219) boot: OTA sequence values A 0xffffffff B 0xffffffff
D (223) boot: OTA sequence numbers both empty (all-0xFF)
I (229) boot: Defaulting to factory image
D (233) boot: Trying partition index -1 offs 0x10000 size 0x130000
D (240) esp_image: reading image header @ 0x10000
D (244) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (251) esp_image: image header: 0xe9 0x07 0x02 0x01 40080cac
V (257) esp_image: loading segment header 0 at offset 0x10018
V (263) esp_image: segment data length 0x4d68 data starts 0x10020
V (269) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
I (277) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x04d68 ( 19816) map
D (286) bootloader_flash: mmu set paddr=00010000 count=1
V (294) esp_image: loading segment header 1 at offset 0x14d88
D (297) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
[... 36 lines cut ...]
D (554) esp_image: Calculated hash: e79e4ada
D (559) bootloader_flash: mmu set paddr=00030000 count=1
I (566) boot: Loaded app from partition at offset 0x10000
I (570) boot: Disabling RNG early entropy source...
D (576) boot: Mapping segment 0 as IROM
D (580) boot: Mapping segment 6 as DROM
D (584) boot: calling set_cache_and_start_app
D (588) boot: configure drom and irom and start
V (593) boot: d mmu set paddr=00020000 vaddr=400d0000 size=85172 n=2
V (599) boot: rc=0
V (601) boot: rc=0
V (603) boot: i mmu set paddr=00010000 vaddr=3f400000 size=19816 n=1
V (610) boot: rc=0
V (612) boot: rc=0
D (614) boot: start: 0x40080cac
I (617) cpu_start: Pro cpu up.
I (621) cpu_start: Single core mode
I (625) heap_init: Initializing. RAM available for dynamic allocation:
I (632) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (638) heap_init: At 3FFB1E50 len 0002E1B0 (184 KiB): DRAM
I (645) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (651) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (657) heap_init: At 40087F2C len 000180D4 (96 KiB): IRAM
I (663) cpu_start: Pro cpu start user code
```

igrr, 2019-04-10:
Sorry for the late follow-up @HFRibeiro, some time ago we have started enabling RTC watchdog inside the bootloader (https://github.com/espressif/esp-idf/commit/9c715d7946a9595bad53307cf0a141d4226d0a5a) which might partially help with such issues.

Also the ESP32 hardware design guidelines and module datasheets have been updated to include a bulk capacitor discharge circuit, which can eliminate the cause of the issue.

closed 2019-04-10 by igrr