espressif/esp-idf #1509: eclipse-retup.rst: ambiguous instructions

plasticuser, 2018-01-15:
https://github.com/espressif/esp-idf/blob/master/docs/get-started/eclipse-setup.rst

In project properties, it says "Edit the PATH environment variable. Keep the current value, and append the path to the Xtensa toolchain that will installed as part of IDF setup (something/xtensa-esp32-elf/bin) if this is not already listed on the PATH."

Of five of us here, not one of us knows what that means. Is it added comma separated? Is "something" the $PATH$?

chegewara, 2018-01-15:
http://esp-idf.readthedocs.io/en/latest/get-started/index.html#setup-path-to-esp-idf

krzychb, 2018-01-15:
> Of five of us here, not one of us knows what that means. Is it added comma separated? Is "something" the $PATH$?

* Yes, it is added comma separated
* "something" means full path to "xtensa-esp32-elf/bin"

In my case (Linux) the PATH looks as follows:

```
/bin:/home/krzysztof/bin:/home/krzysztof/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/usr/games:/usr/local/games:/snap/bin:/home/krzysztof/esp/xtensa-esp32-elf/bin
```

[image]

projectgus, 2018-01-23:
Reopening as I think we can reword this section to be clearer.

closed 2018-03-16 by igrr