espressif/esp-idf #1510: socket() FDs too big in IDF >= 3.0

markwj, 2018-01-16:
Modifying the standard hello_world template App, to add:

```
#include <sys/socket.h>
int sock = socket(AF_INET, SOCK_STREAM, 0);
printf("Allocated socket #%d\n",sock);
if (sock >= 0) close(sock);
```

I get this output when compiling against ESP IDF v3.0 / v3.1:

> Hello world!
> This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 16MB external flash
> Allocated socket #4096

When compiling against v2.1, I get:

> Hello world!
> This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 16MB external flash
> Allocated socket #0

This messes up the select() call, using:

```
FD_ZERO(&fds);
FD_SET(sock,&fds);

```

Why?

projectgus, 2018-01-16:
Hi @markwj,

We started setting LWIP_SOCKET_OFFET so we could unify calls to POSIX I/O functions like read() and write(). Previously you had to be careful to either include POSIX headers or LWIP headers, not both, and you couldn't mix file I/O and socket I/O in the same source file. This change is mentioned at the bottom of the [Breaking Changes section of the v3.0-rc1 release notes](https://github.com/espressif/esp-idf/releases/tag/v3.0-rc1).

FD_ZERO and FD_SET should be OK, the LWIP versions should be used and they are offset-aware. There is an alternative implementation in newlib/include/sys/types.h however this version shouldn't be compiled in if ESP_PLATFORM is set (the ESP-IDF build system sets this macro).

Can you please give some more details about the configuration where this isn't working?

markwj, 2018-01-17:
Thanks for the quick reply. I'm really wondering why such an approach (FD offsets) was used. I understand the requirement to determine the component subsystem, but solving the issue by drastically changing the file descriptor system seems dangerous and likely to break stuff (as it did here).

The application I am working on is a large and complex project (binary size currently >1.5MB, and made up of a large number of open source components). Several of those components use the select() call in different ways. Some worked, some didn't. I did manage to isolate the problem to using select() in the simplistic common manner as described in these articles:

  https://www.gnu.org/software/libc/manual/html_node/Waiting-for-I_002fO.html
  https://www.gnu.org/software/libc/manual/html_node/Server-Example.html

`int result = select(FD_SETSIZE, &fds, 0, 0, &timeout);`

The first parameter is the maximum FD number + 1, and in the new 3.x scheme must NOT have had the offset removed, while the 2nd/3rd/4th parameters are the FDSETs, and in the new 3.x scheme MUST have had the offset removed. FD_SETSIZE doesn't include the offset, so the select() call will always return (after timeout) with no activity found. The FDSETs themselves seem fine.

The fix in application code is:

`int result = select((int) max_fd + 1, &fds, 0, 0, &timeout);`

So now the first parameter includes the offset, but 2nd parameter FDs don't.

Given the current implementation, I don't think much can be done. Changing the call at the application level works around it for me. I don't think any sanity checking can be added to select() because it has no way of knowing the size of the FDSETs other than the nfds first parameter.

Perhaps just update the release notes to add a comment about this common use case of 'select(FD_SETSIZE,...'?

Palantir555, 2018-01-18:
I was about to create a new issue about this same thing.

I've been testing the latest version of the SDK with our ESP32 BSP in the ([Xively Client](https://github.com/xively/xively-client-c)), and the change of FD_*s introduced in [454e473](https://github.com/espressif/esp-idf/commit/454e47385c5182fb8b4999ae91c54304180ad4cf) broke the connectivity of our port. Same behaviour described by @markwj 

We're using standard types (`fd_set` for the read/write/error FDs, `int` for the max_fd), just as Mark mentioned, but the behaviour [our code](https://github.com/xively/xively-client-c/blob/master/src/bsp/platform/esp32/xi_bsp_io_net_esp32.c) is still broken:

```
fd_set rfds;
fd_set wfds;
fd_set efds;
[...]
const int max_fd = MAX( max_fd_read, MAX( max_fd_write, max_fd_error ) );
const int result = select( max_fd + 1, &rfds, &wfds, &efds, &tv );
```

@projectgus Am I missing any variables that require a bigger type to account for the offset?

Palantir555, 2018-02-02:
@projectgus Thanks a lot for the support.

We **were** using our own build system to compile our library and BSP, and yours for the application.
As it happens, I was just working on a small refactor of our library, to get our ESP32 BSP compiled as its own IDF component.

I've completed that work, rebuilt everything with the latest SDK, and the issue is fixed!
Defining ESP_PLATFORM by itself didn't fix the issue (I'm sure lwip's FD_SET macros were in use when the connection fails), so I must have missed some other macro, which resulted in silently incompatible build configurations.

I didn't need to change any source code, and everything seems to be working fine now, thanks again!

projectgus, 2018-02-05:
I'm very glad to hear that it came good for you. If you notice anything still amiss from the ESP-IDF side, please let us know.

A bit odd that it didn't fix itself the first time. Only thing I can think of, some build systems don't pick up changes in the Makefile itself - ie adding extra CPPFLAGS - as a cue to recompile the source files. IDF build system does this, but maybe the Xively system doesn't? This is a total blind guess.

Angus

markwj, 2018-03-02:
> Some good news, this approach is going to go away soon (mostly likely IDF v3.1) when we unify select() support between sockets and other kinds of file descriptors. So passing FD_SETSIZE will work again, and all file descriptors will have low numeric values again. The workarounds mentioned here will all continue to work when we make this change.

Sounds good. I'm happy for this issue to be closed, as the documentation in the release note addresses it.

closed 2018-06-08 by FayeY