espressif/esp-idf #1512: LWIP update

zafrirron, 2018-01-16:
Hi
Looking at LWIP version, you are using old version, lots of changes since you import LWIP,
would you consider updatating LWIP?
(not sure how many customizations done...)

hchaudhary1, 2018-01-18:
@zafrirron are you looking for a particular feature?

zafrirron, 2018-01-19:
Looking to port HomeKit protocol support github project that requires newer LWIP according to the project writer.

FayeY, 2018-01-23:
Hi @zafrirron , the ESP32 has already supported Homekit, but it is not open on the github, you can send a request for ESP32 Homekit SDK on the [ESP website](http://www.espressif.com/en/company/contact/technical-inquiries-category).
Thanks.

renssies, 2018-01-27:
@FayeY don't close this too quickly. The SDK you are referring too is for MFI licensees only and Espressif has said to have no plans for a non-commercial SDK. So hobbyists are creating their own SDK for the ESP32.

Since WWDC 2017 there is a Non-commercial HomeKit spec for Hobbyist, Enthusiasts, and prototyping without needing a license. People have already created an SDK for the ESP8266 (https://github.com/maximkulkin/esp-homekit). They want to port this to the ESP32, however, this can not be done due too LWIP being out of date and therefor missing features. 

So until Espressif decides to also open an SDK for non-commercial use, you should still consider updating LWIP

zafrirron, 2018-01-29:
Thanks @renssies,
I did not expect such answer from @FayeY, thought he understands most users here and ESP32 IDF current state, we all in development and prototyping stages, so sending me to MFI option.... was disappointing 👎:-( did not bother to answer.

FayeY, 2018-01-29:
Hi both,
Sorry for the misunderstanding before.
This issue has been reopened and assigned to our engineers internally, we will evaluate this request.
Thanks for your interest in ESP32!

affonsoloyola, 2018-02-11:
Same thing here. Trying to prototype something that needs newer features os LWIP. 
I tried to get the HomeKit SDK with Espressif support but they said it’s only available for MFi licenced companies. Please update LWIP. That version is too old!

goldsimon, 2018-03-01:
While you're at it, you could try to minimize the changes to upstream sources so that updating lwIP would be a simple matter of updating the sources and recompiling. lwIP tries to be compatible and/or simple to upgrade, so this should not be too hard as long as the sources are kept clean.

Also: a) some of the local changes seem bogus and should not be required and b) for things that are real bugs, it would be nice to get them reported back to the lwIP project.

zhangyanjiaoesp, 2018-03-15:
@zafrirron@hchaudhary1@renssies@affonsoloyola@goldsimon
Recently we are updating our LWIP  to the offical v2.0.3, but it will take some time for the system to work stablely

goldsimon, 2018-03-15:
That's great news! May I suggest again to work with the lwIP project (I'm the maintainer of that stack) to get you a clean copy of the stack? That will also make it easier to integrate new releases, as there should be no stability issues otherwise (when upgrading a clean lwIP stack).

Just post your questions or issues to the lwip-devel@nongnu.org mailing list.

zhangyanjiaoesp, 2018-03-15:
@goldsimon yes, you are right ! we are doing what you said exactlly.

jukorp, 2018-03-15:
i'm looking forward to the new lwip 2.0.3 version.
Estimate when ready?

jukorp, 2018-03-15:
Are there other problems than the lwip version? wolfssl?

ourairquality, 2018-05-08:
I would also like to see a sync to the upstream lwip. The esp-open-rtos has been following the upstream lwip and attempting to contribute changes upstream. It would make it easier to port code across these development environments. Is someone planning to work on this, or is it waiting for a contribution?

jeffsenn, 2018-06-26:
Big +1 on this.  I just spent the day finding and fixing a leak (LWIP Bug #51937) that has already been fixed "upstream".

NZSmartie, 2018-07-15:
I've started a fork where i've removed Espressif's copy of LWIP in favour of using @goldsimon's version hosted on nongnu,org. The build does compile with "Copy L2 to L3" enabled under both LWIP and Ethernet component configuration options. 

Modifications have/are being done to components of ESP-IDF to support the new LWIP and I've found quite a few redundancies with developers reimplementing code that already exists in LWIP. Hopefully once it's all done, keeping LWIP up to date will be as easy as `git pull` within `components/lwip/lwip`

Please note that not all configuration options have been tested yet, but i hope to have that done in the next week.

twelve12pm, 2018-08-02:
Big +1 on using clean upstream lwIP!

comcomservices, 2018-08-08:
@NZSmartie, using your fork! Thanks a ton!

zhangyanjiaoesp, 2018-09-07:
Hi all， sorry for replying later. 
The lwIP in ESP-IDF has been updated to the [offical stable version 2.0.3](https://savannah.nongnu.org/projects/lwip/) in the latest master. 
And the lwIP is as a submodule in ESP-IDF now, you can find the esp-lwip[ here](https://github.com/espressif/esp-lwip)

closed 2018-09-07 by FayeY