espressif/esp-idf #1513: spi_master.c faulty MISO delay configuration

Roody75, 2018-01-17:
Regarding the used SPI clock frequency the MISO delay configuration is applied the wrong way. This has the effect that additional '1' bit(s) occur in front of the content of an SPI RX frame and therefore the whole received SPI data frame is right shifted 1 (or 2) bits. Additionally this issues shows up only depending on the selected SPI mode (in our case mode 1).

The following is statet in the ESP32 technical reference (Chapter 5.4.2 SPI-Timing, Page 74):

> When GP-SPI is used as master and the data signals are not received by the SPI controller via GPIO matrix, if GP-SPI output clock frequency is not higher than clkapb/2, register SPI_MISO_DELAY_MODE should be set to 0 when configuring the clock polarity. If GP-SPI output clock frequency is not higher than clkapb/4, register SPI_MISO_DELAY_MODE can be set to the corresponding value in Table 23 when configuring the clock polarity.

The according logic taking care for this is in spi_master.c (line 414ff.):
```
            //Configure polarity
            //SPI iface needs to be configured for a delay in some cases.
            int nodelay=0;
            int extra_dummy=0;
            if (host->no_gpio_matrix) {
                if (effclk >= apbclk/2) {
                    nodelay=1;
                }
            } else {
                if (effclk >= apbclk/2) {
                    nodelay=1;
                    extra_dummy=1;          //Note: This only works on half-duplex connections. spi_bus_add_device checks for this.
                } else if (effclk >= apbclk/4) {
                    nodelay=1;
                }
            }

            if (dev->cfg.mode==0) {
                host->hw->pin.ck_idle_edge=0;
                host->hw->user.ck_out_edge=0;
                host->hw->ctrl2.miso_delay_mode=nodelay?0:2;
            } else if (dev->cfg.mode==1) {
                host->hw->pin.ck_idle_edge=0;
                host->hw->user.ck_out_edge=1;
                host->hw->ctrl2.miso_delay_mode=nodelay?0:1;
            } else if (dev->cfg.mode==2) {
                host->hw->pin.ck_idle_edge=1;
                host->hw->user.ck_out_edge=1;
                host->hw->ctrl2.miso_delay_mode=nodelay?0:1;
            } else if (dev->cfg.mode==3) {
                host->hw->pin.ck_idle_edge=1;
                host->hw->user.ck_out_edge=0;
                host->hw->ctrl2.miso_delay_mode=nodelay?0:2;
            }...
````
As far as I see the variable `nodelay` is used in the opposite way IMHO it rather should look this way:

```
            //Configure polarity
            //SPI iface needs to be configured for a delay in some cases.
            int nodelay=1;
            int extra_dummy=0;
            if (host->no_gpio_matrix) {
                if (effclk >= apbclk/2) {
                    nodelay=0;
                }
            } else {
                if (effclk >= apbclk/2) {
                    nodelay=0;
                    extra_dummy=1;          //Note: This only works on half-duplex connections. spi_bus_add_device checks for this.
                } else if (effclk >= apbclk/4) {
                    nodelay=0;
                }
            }
```

negativekelvin, 2018-01-17:
Linking to https://github.com/espressif/esp-idf/issues/1346 so they can be resolved together

ginkgm, 2018-02-27:
I think the registers are correct. 

The `delay` affects MISO, which makes the reading slower maybe half a cycle to meet the fetching edge of ESP32. It helps to improve the hold time timing (make it larger), but is not necessary (usually we have input delay).  When the frequency gets larger, the period may gets so small that `delay` together with MISO input delay violate the setup requirement. So we cancel the `delay` when frequency is large enough.

The 'dummy' affects the SCK, it means we skip 1 clock without writing and reading. But the slave still see the clock, so the MISO comes at the same time but the clock comes 1 clock later. This helps to read faster to compensate the input delay. So only the frequency is high enough (comparable with the input delay) will we use the `dummy`.

closed 2018-07-12 by FayeY