espressif/esp-idf #1514: ESP32 crashed when enable software control bluetooth and wifi coexistence

noanbui, 2018-01-18:
I have developed an application on espwroom32 that uses Bluetooth and Wifi at the same time. I set wifi at AP mode. When I run with both of them, enable software control bluetooth and wifi coexistence in menuconfig and connect to a Bluetooth device to transfer some data, my application crash after a few minutes. Here are the details:

[00:11:15.306] EVT <= 13 05 01 80 00 01 00
ASSERT_PARAM(8192 0), in rwbt.c at line 360host_recv_pkt_cb called from ISR!
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40084124  PS      : 0x00060234  A0      : 0x8005574c  A1      : 0x3ffc05b0
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffc1028  A7      : 0x3ffd5560  A8      : 0x80084124  A9      : 0x3ffc0590
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0x00000000
A14     : 0x3ffbb7e0  A15     : 0xfffffffc  SAR     : 0x00000019  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4008405c  LEND    : 0x40084063  LCOUNT  : 0x00000000

Backtrace: 0x40084124:0x3ffc05b0 0x40055749:0x3ffc05d0 0x40086b6f:0x3ffc05f0 0x400826a1:0x3
ffc0610 0x400d1fcd:0x00000000

Sdkconfig and log can found here: [https://github.com/noanbui/esp32](url)
Additional, I use btstack library for Bluetooth function. And if I disable software control bluetooth and wifi coexistence, after I connect to a blueooth device, my application can run without crash but wifi disappear.
I continue to find cause of this issue, hope that you all are interested and give me some guide to resolve this soon.
Thanks

negativekelvin, 2018-01-18:
https://github.com/espressif/esp-idf/issues/1322

noanbui, 2018-01-18:
thank you so much @negativekelvin

FayeY, 2018-01-23:
Closing this, as this issue is a duplicate issue.

closed 2018-01-23 by FayeY