espressif/esp-idf #1515: "examples/performance/tcp_perf" and "examples/wifi/iperf" give different results

abcdcadc, 2018-01-18:
while idf/examples/performance/tcp_perf  gives  speed around 1.8Mbit/sec (I run tcp_perfon esp and on PC side, i send continuously nodelay tcp package), examples/wifi/iperf gives much higher (about 10Mbit/sec) (I did as the manual, server built on PC side by iperf2.06 and client on esp32)

I tried to modify the configuration( in "wifi" and "lwip" of menuconfig) of tcp_perf following the iperf, but still same
is there any else configuration I must do to improve the throughput of tcp_iperf as iperf?

liuzfesp, 2018-01-31:
@abcdcadc tcp_perf example has different configuration than iperf example, it's a obsolete example, we will remove it in idf v3.1.

abcdcadc, 2018-01-31:
@liuzfesp thanks  for rep,
in version3.1, if I follow configuration of iperf, can I get my expected speed?

liuzfesp, 2018-02-01:
Absolutely, @abcdcadc

liuzfesp, 2018-02-01:
@abcdcadc the throughput optimization code is already merged into IDF master branch, it will be ready in 2 days in github IDF master branch. In other words, 2 days later, pull the latest idf master, you can get the stable iperf example.

liuzfesp, 2018-02-01:
@abcdcadc please pull the latest idf master branch, use iperf example for your test. The throughput optimization commit is: 17f01eb718a20f1fcfba69351907488c4c7e8b9c

abcdcadc, 2018-02-02:
@liuzfesp , i did on the newest version of iperf, but the result still is bad
[image]

liuzfesp, 2018-02-02:
@abcdcadc could you describe the detailed test steps? Test it in the air or shield box?

heyinling, 2018-02-02:
@abcdcadc I would also suggest you check the following items (may not directly related to your issue):

1. use ethernet to connect your PC NIC to AP, make sure the PC to AP side is stable
2. use native ethernet NIC instead of USB NIC. USB NIC might be the bottle neck
3. switch off anti-varies SW and windows firewall

abcdcadc, 2018-02-02:
@liuzfesp the procedure follows the "readme.md" in iperf folder:
1/ connect PC to router (via wifi)
2/ connect esp32 to router by (sta ssid password)
3/ create server on PC (using iperf2.09, the newest version of iperf 3.0 doesnt work by error: reset by peer???) 
4/connect esp32 to server (as readme.md)
esp32 chip is in air
@heyinling 1/ you mean connect directly PC to router using cable RJ45? ( now my PC are connecting via wifi)
3/ definitely no firewall and anti virsus

heyinling, 2018-02-02:
yes, use RJ45 interface. this will make sure PC to router side can reach 100M. 

And need to make ESP32 not too far away form router (use ESP32 to scan router, RSSI larger than -50).

liuzfesp, 2018-02-02:
@abcdcadc,  in the air, the throughput is much lower than in the shieldbox, but your result looks abnormal.  Could you double check you IDF master is newest and have run the "git submodule update" to update the submodules?

abcdcadc, 2018-02-02:
@heyinling , for simplification, i will turn the PC into AP and make esp32 join that AP, is it ok?
@liuzfesp , i just git clone this morning, it seems no updated until now
could you tell me what is shieldbox, is it possible to buy and equip one for my esp32?

heyinling, 2018-02-02:
you may try this first. I suppose with this config you should at least get result improved to Mbps.

But if you use USB wifi NIC, or your WIFI NIC don't support 11g, or HT40, it could still be the bottle neck for the test.

shield box is a box that you put your ESP32 and AP inside it. then it will shield all the RF signals outside. If you don't have one, I would suggest you run test on a relative clean environment (don't have a lot WIFI AP / STAs).

liuzfesp, 2018-02-02:
@abcdcadc if no shield box, you can also test it in carport.

abcdcadc, 2018-02-02:
@liuzfesp that was my mistake because esp32 was far from router, I moved it nearer (20cm) and got as below:
[image]
 quite better and stable but still very lower document 10Mbit/sec
@heyinling I also tried to use RJ45 cable with the above distance and got:
[image]
it even is worse than using wireless!!!

liuzfesp, 2018-02-02:
Correct one of my previous reply message, "if no shield box, you can also test it in carport." should be "if no shield box, you can also test it in a garage with walls".

abcdcadc, 2018-02-02:
@liuzfesp , you mean wifi performance is strongly affected by environment, i will lose more than 10 Mbit/sec if i am at a clutter environment, right?

liuzfesp, 2018-02-02:
@abcdcadc could you replace esp32 with your another PC (let's say PC2), PC2 connects to the router with wireless. And run the test case again to check whether is because the test environment is not good.

liuzfesp, 2018-02-02:
@abcdcadc you are right, the performance is greatly affected by environment

closed 2018-03-05 by FayeY