espressif/esp-idf #1516: read data on peripheral side during scan/response from Central (GAP only data transfere)

JoanMCD, 2018-01-18:
Hello,  for a real LE device I want send sometimes soome data from the Central (i.e. PC/Smartphone) to the peripheral (battery powered ESP32). IMO a typical BLE case.

In the normal case the ESP32 is adverticing and if needed the Central can active scan and response with data (GAP). The most time the ESP is in sleep. A connection (GATT) is not possible. On the ESP side I miss the Event to grab the transmitted data or better an example how to implement this. Any hints ?

negativekelvin, 2018-01-18:
I think you are asking for an event triggered by a scan request that your app can use to decide whether to stay awake and receive an advertising packet or connection request  @Yulong-espressif

JoanMCD, 2018-01-19:
Hi negativekelvin , thx for answer !
I never want to create a connection (never want to use GATT).

If you want transport data (some byte) from:
- Peripheral (Client) to Central (Server) you can do this connectionless by broadcast (advertising) in the ADV_IND package.
- Central to Peripheral, by additional data in the SCAN_REQ package during active scanning.

See the graphic here:
https://devzone.nordicsemi.com/attachment/515dae67de062987ed99c9ac72a663af
or here:
http://dev.ti.com/tirex/content/simplelink_academy_cc2640r2sdk_1_13_03_11/modules/ble_scan_adv_basic/resources/scanning_flowchart.png

JoanMCD, 2018-01-19:
it's part of the standard

locks like:
83 0c 55 87 45 0b 22 ac eb 12 e6 2d bb f5 cc 1c fd   
 Advertising / AA 8e89bed6 / 12 bytes  
   Channel Index: 37  
   Type: SCAN_REQ  
   ScanA: ac:22:0b:45:87:55 (public)  
   AdvA: f5:bb:2d:e6:12:eb (random)  
   Data: 55 87 45 0b 22 ac eb 12 e6 2d bb f5  
   CRC:  cc 1c fd

negativekelvin, 2018-01-19:
This is from 4.2 specification
> 2.3.2.1 SCAN_REQ
The SCAN_REQ PDU has the Payload as shown in Figure 2.8. The TxAdd in
the advertising channel PDU header indicates whether the scanner’s address
in the ScanA field is public (TxAdd = 0) or random (TxAdd = 1). The RxAdd in
the advertising channel PDU header indicates whether the advertiser’s
address in the AdvA field is public (RxAdd = 0) or random (RxAdd = 1).
The Payload field consists of ScanA and AdvA fields. The ScanA field shall
contain the scanner’s public or random device address as indicated by TxAdd.
The AdvA field is the address of the device to which this PDU is addressed.
The AdvA field shall contain the advertiser’s public or random device address
as indicated by RxAdd.
**Note: This packet does not contain any Host Data.**

chegewara, 2018-01-19:
@JoanMCD 
```
Hello, for a real LE device I want send sometimes soome data from the Central (i.e. PC/Smartphone) to the peripheral (battery powered ESP32). IMO a typical BLE case.

In the normal case the ESP32 is adverticing and if needed the Central can active scan and response with data (GAP). The most time the ESP is in sleep. A connection (GATT) is not possible. On the ESP side I miss the Event to grab the transmitted data or better an example how to implement this. Any hints ?
```
Scan response is additional advertising packed that can be only send from peripheral to central (observer). Advertising concept is to have 2 packets with 31 bytes payload each. One advertising packet is broarcasted and can be "read" with passive scan, the other is scan response and can be read with active scan.

Only way to read any data with esp32 and send from your phone by switching esp32 from broadcaster to observer, read data and switch back. But in this case your phone needs to be advertiser. Or just use GATT connection.

FayeY, 2018-01-23:
Hi @JoanMCD , did the above suggestion help you resolve the issue?

negativekelvin, 2018-01-23:
@FayeY I think it is still valid feature request that SCAN_REQ event is passed to application

JoanMCD, 2018-01-23:
Wow, thank you all for you feedback. I have to read all at first. I was involved in a other BLE topic (send and get INDICATIONS) :) the last days.

@negativekelvin - well researched. It's a clear answer. I was sure, that I get (read, found, dream about, ...) that a data transfere in both directions is possible with SCAN_REQ and RESPONCE. I'm wrong, because I'm taking about the ESP and BLE 4.x / not 5.x (maybe I read it there ?!).

By reading the comment from chegewara I had two ideas:
First - place a seperate Broadcasing/Advertiser device :( (a replacement for Smartphone) to only deliver messages to the periferal (observer in this case). It must broadcast in a very high rate to keep the waketime of the Observer small  or...

It's possible to read the content of the SCAN_REQ (ESP32 side) and missuse (fake) the content of the  Payload field (ScanA and AdvA) with own data?

JoanMCD, 2018-01-23:
@FayeY @all Yes, it helps me. With your new information I found, that I only have to send an one-bit information that the ESP32 should keep awake to be connectable for a further data exchange... And this 'bit' is the existence of the SCAN_REQ undepending it's content.

Have someone an example :)

JoanMCD, 2018-01-24:
I didn't find a matching event at the advertiser side if a SCAN_REQ was received ...
Something like ESP_GAP_BLE_SCAN_**REQUEST**_EVT

Yulong-espressif, 2018-01-25:
@JoanMCD It is no SCAN_REQ at the advertiser side to the application, because the SCAN_REQ didn't have any payload and didn't have hci event from controller to the host.

Yulong-espressif, 2018-01-25:
@JoanMCD So I don't know what do you want to do? Why you need the SCAN_REQ callback?

JoanMCD, 2018-01-25:
@Yulong-espressif With the info I got above I changed the concept.

Normal I will have a sequence of: sleep, wakeup, advertise, sleep, wakeup, ...

By receiving a SCAN_REQ with my advertise I want to interrupt this sequence, 
keep awake long enough to be connected, receive data from the central and not until this go sleeping again.

I don't need a payload in the SCAN_REQ, but the callback for the decision.

chegewara, 2018-01-25:
@JoanMCD If you dont care what is in scan request, then you can do like this:
- central is sending scan request, if central receive scan request then 
- central is connecting to peripheral and send payload (write to characteristic), when its done
- disconnect and go to sleep?

JoanMCD, 2018-01-25:
@chegewara not sure if there is a typo in your text (or a misunderstanding here), if you mean this:

1. central is sending scan request, if **periferal** receive scan request then
2. central is connecting to peripheral and send payload (write to characteristic), when its done
3. disconnect and go to sleep?

than, yes.

But between 1 & 2 At the periferal I need a hook to switch from 'the advertising and sleeping only loop' to avdvertising (+ be connectable) , wait for connection, connection and data exchange, disconnect, sleep... 

So:
0.5 the peripheral is advertising (broadcasting)
1. central is sending scan request, if **peripheral** receive scan request 
1.2 In the callback of the SCAN_REQ (at the peripheral) switch to advertising with bit 'connectable'
1.4 the peripheral wait some seconds for connetion
2. central is connecting to peripheral and send payload (write to characteristic), 
    Naming: The Central becomes the Client, the Peripheral the Server (GATT)
3. when its done disconnect and go to sleep.

My problem is the point of the decision at 1.2

Btw. All this is only nessesary because the ESP is not able to create a connetion and keep it during low power modes / sleep. Other chips (i.e. SiliconLabs and TI) have different sleep and power down modes and weakup events. The ESP every time starts up from beginning (reset) and all connection data are lost. To preperare, connect, bind, discover it each wakeup needs to much energy/time.  

Correct me if I'm wrong. 

The real story is very simple: Let's say I have a battery powerd sensor broadcasting a new mearurement every minute to a system. This is possible up to now. But sometimes there is the need for FW update or change parameters (i.e. transmission rate), than I need to connect the sensor (in time) and sent data to it.

chegewara, 2018-01-25:
In this story i dont see the point of switching between only advertising and advertising + connectable. Why dont stick with connectable? This changes nothing in battery usage.

JoanMCD, 2018-01-25:
the SCAN_REQ is handled by the Stack and is send very fast. The different is the time the peripheral waits after it's broadcast for an SCAN_REQ or an Connection

chegewara, 2018-01-25:
Then stay with option SCAN_REQ to broadcast data you want to, but leave peripheral connectable to setup options or whatever its required for. You dont need to turn on/off connectable bit. This will only make your code more complicated.
This approach can be used to to keep peripheral more safe and turn on connectable with switch or NFC trigger.

You have to keep in mind that all i write here is just my opinion and im not expert also i dont know what kind application you are trying to build.

JoanMCD, 2018-01-25:
Yes, that's right.  In the theory of the concept I don't want to set a 'I'm connectable' bit and than I'm not waiting for the connetion and go sleeping. But this is only a detail of the story. 
And - I'm ever happy about your feedback. To talk about a problem helps to solve it. :) 

The highest priory in my project is so save energy and life with one battery charge as long as possible. 
At the moment I try to get this with the ESP32 (here i would also have the option to switch in WiFi). 
An alternative is to use a dedicated low-power BT BLE module and a ATmega328 communicating by 'AT' commands (you get a wake if new data arrived) or a programmable module like BGM111 and leave the Arduino IDE programming world.
A nice feature of the last one is to only broadcast on one channel, also this reduce the running time.

chegewara, 2018-01-26:
Since its not issue i would suggest to move this discussion to forum.

negativekelvin, 2018-01-26:
@chegewara yes it is an issue because ESP_GAP_BLE_SCAN_REQUEST_EVT does not exist

chegewara, 2018-01-26:
@negativekelvin Just read this
https://github.com/espressif/esp-idf/issues/1516#issuecomment-360344531

negativekelvin, 2018-01-26:
@chegewara right there is currently no hci event but I don't see why it couldn't be added like silabs did

Weijian-Espressif, 2018-03-14:
@negativekelvin sorry,  It is not a hci event, our ble code has not the event.

negativekelvin, 2018-03-14:
@Weijian-Espressif you mean the riverawaves code that handles scan request does not generate hci event, correct?  Is that not fixable?

Weijian-Espressif, 2018-03-16:
there is no hci event for ESP_GAP_BLE_SCAN_REQUEST_EVT, so our BLE do not support the EVENT.

negativekelvin, 2018-03-16:
Right, so modify the low level stack code and add one.  TI has it:

```
Macros
--
#define | HCI_EXT_DISABLE_SCAN_REQUEST_REPORT   LL_EXT_DISABLE_SCAN_REQUEST_REPORT
  | Disable.
 
#define | HCI_EXT_ENABLE_SCAN_REQUEST_REPORT   LL_EXT_ENABLE_SCAN_REQUEST_REPORT
  | Enable.
```

http://dev.ti.com/tirex/content/simplelink_cc2640r2_sdk_1_50_00_58/docs/blestack/ble_user_guide/doxygen/ble/html/group___scan__req.html

Weijian-Espressif, 2018-04-09:
sorry, this is a custom protocol, we need to discuss.

negativekelvin, 2018-04-09:
Ok no problem.  If it is disabled by default it should not cause any issue.  If it is optional at least we can experiment.

Weijian-Espressif, 2019-03-18:
@all, Sorry, this is not defined by BLE spec, we have no plans to add the event now.

closed 2019-03-18 by Alvin1Zhang

negativekelvin, 2019-03-18:
> BLUETOOTH SPECIFICATION Version 5.0 | Vol 2, Part E page 1227
> Host Controller Interface Functional Specification
> 7.7.65.19 LE Scan Request Received Event
> Description:
> The LE Scan Request Received event indicates that a SCAN_REQ PDU or an
> AUX_SCAN_REQ PDU has been received by the advertiser. The request
> contains a device address from a scanner that is allowed by the advertising
> filter policy. The advertising set is identified by Advertising_Handle.
> This event shall only be generated if advertising was enabled using the LE Set
> Extended Advertising Enable command.
> The Scanner_Address_Type and Scanner_Address indicates the type of the
> address and the address of the scanner device.

chegewara, 2019-03-18:
> This event shall only be generated if advertising was enabled using the LE Set
Extended Advertising Enable command.

Im guessing esp32 is not supporting this.

negativekelvin, 2019-03-18:
Yeah it looks like that is the only way to enable it other than having custom hci command like TI, but it should work with legacy advertising too so it doesn't really need full extended advertising support.  Sad.

> The Scan_Request_Notification_Enable parameter indicates whether the
> Controller shall send notifications upon the receipt of a scan request PDU that
> is in response to an advertisement from the specified advertising set that
> contains its device address and is from a scanner that is allowed by the
> advertising filter policy.