espressif/esp-idf #1517: GPIO36 and 39 ADC noise and cross coupling

jmorsap, 2018-01-18:
I am having some funny behavior when using GPIO36 and 39 as normal adc inputs. I use the following setup code

```
	adc1_config_width(ADC_WIDTH_12Bit);
	gpio_set_direction(MY_INPUT, GPIO_MODE_INPUT);
	adc1_config_channel_atten(MY_INPUT,ADC_ATTEN_11db);
        gpio_set_direction(MY_INPUT2, GPIO_MODE_INPUT);
	adc1_config_channel_atten(MY_INPUT2,ADC_ATTEN_11db);
```

and just loop either of the the following sections

`
```
vTaskDelay(delayms/portTICK_PERIOD_MS); //adc takes almost no time (a few uS)

printf("%d,%d \r\n",mycase ,adc1_get_voltage(MY_INPUT));`

`
vTaskDelay(delayms/portTICK_PERIOD_MS); //adc takes almost no time (a few uS)

printf("%d,%d \r\n",mycase ,adc1_get_voltage(MY_INPUT2));`
```

I get a pretty noisy signal, and when I check gpio39 with the scope, I can see that the line is getting pulled low. I've confirmed that this only occurs when the adc1_get_voltage is used and is not caused by the printf command. The problem is caused on both pins by using the ADC on either pin. 

[image]
 a closer look
[image]

Also, grounding GPIO36 appears to affect the noise on GPIO39. 

Any thoughts on what is causing this and any solutions or workarounds? Right now the plan is to redesign our whole board, but we would of course prefer not to do that.

negativekelvin, 2018-01-18:
There is something with the LNA and SAR state machine but these are basically undocumented black boxes and the LNA got removed from the datasheet recently so who knows what is happening with it.

costaud, 2018-01-22:
We have found the cause of this issue, will fix this soon.

ESP32DE, 2018-01-22:
@costaud 
you are alive, thought you did not exist anymore
good - same here . wait for the fix

costaud, 2018-01-22:
@ESP32DE Oh yeah, nice to meet you again too. Seems like I just get back from a wormhole : )

negativekelvin, 2018-02-24:
Bug list has been updated:
> 3.11. When the power switch which controls the temperature sensor, SARADC1 sensor, SARADC2 sensor, AMP sensor, or HALL sensor, is turned on, the inputs of GPIO36 and GPIO39 will be pulled down for about 80 ns.

closed 2018-03-12 by FayeY