espressif/esp-idf #1521: Problems with ble encryption on Devices with bluetooth Version < 4.2

ManODi, 2018-01-19:
Hello,

I changed the gatt_security_server example on the ESP_WROOM_32 adding a static public mac address with:  esp_base_mac_addr_set(macAddress); 
and set:  esp_ble_gap_config_local_privacy(false);
I also changed the mac type to public in advertising.

Using a device with bluetooth Version >= 4.2 it works fine. I have to do bonding once and then i can reconnect.
But using an older device with a lower version it seems to fail at esp_ble_set_encryption(param->connect.remote_bda, ESP_BLE_SEC_ENCRYPT);

When I try to reconnect the message is:
E (1080965) BT: l2cble_start_conn_update, the last connection update command still pending.

Did anyone else encounter this Problem? Or do I have to change some settings for compatibility?

Tanks in advance

negativekelvin, 2018-01-19:
https://github.com/espressif/esp-idf/issues/1471

chegewara, 2018-01-19:
https://github.com/espressif/esp-idf/issues/1364

FayeY, 2018-01-23:
Closing this, as this issue is a duplicate issue. Please follow the issues mentioned above instead.

closed 2018-01-23 by FayeY