espressif/esp-idf #1522: a2dp sink - cleanup is buggy

andrewvoznytsa, 2018-01-19:
Hi guys,

I'm trying to run simple sample based on your a2dp sink sample (ESP IDF 3.0rc1, source code of my app attached). The difference with your app is very minor - my app starts a2dp for 50 seconds, then tries to shutdown a2dp/cleanup resources, waits for 10 seconds and continues loop.

It works good **IF** I disconnect phone from a2dp sink **BEFORE** shutdown happens. But if I let it stream expecting that a2dp sink cleanups reasources properly then next attempt to initialize sink fails with

**E (60562) BT: warning : media task already running**

(full log attached).

I tried to fix myself and traced code to btc_a2dp_stop_media_task() until btc_media_ctrl_post(SIG_MEDIA_TASK_CLEAN_UP) - seems like SIG_MEDIA_TASK_CLEAN_UP is never handled by BTC media task (tried to vTaskDelay(10 seconds) after btc_media_ctrl_post(SIG_MEDIA_TASK_CLEAN_UP)  - did not help at all - seems like task is stuck with something else).

So it looks a bit more complex than trivial - could you please help with that issue?

[a2dp_sink_app.zip](https://github.com/espressif/esp-idf/files/1647179/a2dp_sink_app.zip)
[esp32_a2dp_log.txt](https://github.com/espressif/esp-idf/files/1647180/esp32_a2dp_log.txt)

Weijian-Espressif, 2018-02-11:
@blueMoodBHD  please help to see the problem.

blueMoodBHD, 2018-03-16:
@andrewvoznytsa The bug have fixed in latest master, refer to commit 488268d58b5ffad16b64ef40ee96917e6725b575 
Please test, thanks!

closed 2018-03-22 by FayeY