espressif/esp-idf #1524: SPI latency between transactions. Is there a faster way

versamodule, 2018-01-19:
The SPI for me works just fine, but the transaction to transaction time is terrible using esp-idf directly. 
I ran 2 tests (one using esp-idf directly and the other using Arduino code).  Both tests send 2 transactions back to back.  With  esp-idf directly it was 112us until the next transaction, with Arduino code, it was a mere 1.2us. 

using esp-idf directly: Calling this routine 2x

`
    uint8_t temp[2];
    esp_err_t ret;
    spi_transaction_t t;

    temp[0]=0xaa;
    
    memset(&t, 0, sizeof(t));       //Zero out the transaction
    t.length=8;                 //Len is in bytes, transaction length is in bits.
    t.tx_buffer=&temp;               //Data

     ret=spi_device_transmit(lcd_spi, &t);  //Transmit!
     assert(ret==ESP_OK);            //Should have had no issues.
`
as can be seen here. 

[image]

Yet doing the same thing with Arduino code, its magnitudes faster. only taking 1.2us. 

`
    digitalWrite(chipSelectPin, LOW);
    SPI.transfer(0x55);
    digitalWrite(chipSelectPin, HIGH);
`
as can be seen here. 

[image]

The problem is I do not want to use Arduino code, is there another solution that would dramatically shorten transaction time?   How is Arduino able to pull this off.

negativekelvin, 2018-01-19:
spi_device_queue_trans or https://github.com/loboris/ESP32_NEW_SPI_MASTER_EXAMPLE

versamodule, 2018-01-19:
Thanks, I'll give it a try tonight.

FayeY, 2018-01-24:
Hi @versamodule , did the above suggestion help you resolve the issue?

versamodule, 2018-01-24:
I ran into a deadline issue, and have not had the time to try it yet.

Sincerely,

Jimp@Versamodule.com

FayeY, 2018-02-24:
Closing for now. Please feel free to reopen it if this issue persists.

closed 2018-02-24 by FayeY