espressif/esp-idf #1526: C macro comparison against typedef constant ...

nkolban, 2018-01-20:
I'm having some issues compiling with esp_log.h and am stumped by what I am finding.

If we look here:

https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h#L164

We find the following definition:

```
#if (LOG_LOCAL_LEVEL >= ESP_LOG_INFO)
```

I interpret this to be executing the following logic if the value of the macro `LOG_LOCAL_LEVEL` is greater or equal to the value of the macro `ESP_LOG_INFO`.  That sounds fair.   However, the macro `ESP_LOG_INFO` is not defined.  Instead, we have a typedef at:

https://github.com/espressif/esp-idf/blob/master/components/log/include/esp_log.h#L35

That looks as follows:

```
typedef enum {
    ESP_LOG_NONE,       /*!< No log output */
    ESP_LOG_ERROR,      /*!< Critical errors, software module can not recover on its own */
    ESP_LOG_WARN,       /*!< Error conditions from which recovery measures have been taken */
    ESP_LOG_INFO,       /*!< Information messages which describe normal flow of events */
    ESP_LOG_DEBUG,      /*!< Extra information which is not necessary for normal use (values, pointers, sizes, etc). */
    ESP_LOG_VERBOSE     /*!< Bigger chunks of debugging information, or frequent messages which can potentially flood the output. */
} esp_log_level_t;
```

It is my naive belief that the C pre-processor doesn't know anything about typedefs or variables.  How could this possible work?

igrr, 2018-01-20:
Thanks for raising this @nkolban, we have a fix for this issue in the merge queue at the moment. Should appear on GitHub soon.

igrr, 2018-01-28:
My bad, the fix in the merge queue was for a different issue. New fix for *this* issue coming shortly.

closed 2018-01-30 by igrr