espressif/esp-idf #1527: Do you have a plan to add "nat" function

donny681, 2018-01-20:
I want to use esp32 to make a  wifi repeater.I'm trying to add napt in lwip library.But there is something wrong.Here is my demo.[https://github.com/donny681/esp32_repeater](https://github.com/donny681/esp32_repeater)
I know that esp8266 have  a project about wifi repeater.[https://github.com/martin-ger/esp_wifi_repeater](https://github.com/martin-ger/esp_wifi_repeater).I think that it also should work in esp32.
Do you have a plan to add nat in lwip?

liuzfesp, 2018-02-01:
Hi @donny681 based on current feature plan, we don't plan to add nat in idf v3.1. We have put this feature into our future feature list, may consider implement it in idf v4.0.

donny681, 2018-02-06:
@liuzfesp Great,When do you publish  idf v4.0? Because I have a project which needs it.Can you publish  a test version.I think that just changing the lwip.

liuzfesp, 2018-02-06:
@donny681 currently we are developing for v3.1,  for v4.0, I can't promise you a exact time, it should be 2 or 3 months later. For test version, I hope I can got some time to do it after back from Spring Festival.

belinea123, 2018-09-13:
We would need NAT for our project, too. Since IDF v3.1 is released now, it would be great if you could integrate the feature in your planning for v4.0.

crc1021, 2019-01-15:
We also are in need of NAT, is there an updated status for the planning?

ericpelland, 2019-01-19:
Adding a +1 for wanting a NAT feature.

alessandrobilato, 2019-01-22:
Add another +1 for wanting a NAT feature.

FrankFundel, 2019-03-15:
Add another +1 for wanting a NAT feature.

kirillemko, 2019-03-19:
Waiting for NAT feature too

martin-ger, 2019-04-06:
Have a look at this: https://github.com/jonask1337/esp-lwip

FrankFundel, 2019-04-08:
> Have a look at this: https://github.com/jonask1337/esp-lwip

What is the difference between that one and yours (https://github.com/martin-ger/lwip_nat_arduino) ?

martin-ger, 2019-04-08:
It is based on the same code base, but it has be integrated into the latest esp-idf version of the lwip-lib.

FrankFundel, 2019-04-08:
really? Is there a documentation oder anything? I can‘t find a hint that it has been added.

martin-ger, 2019-04-09:
It is not officially added. This is a private effort to port the NAT feature to the ESP32.

kotakomputer, 2019-05-28:
+1 ...  any good news about NAT support in current ESP-IDF version?

beyonlo, 2019-09-27:
Hey all.
+1 about NAT feature.

Any plans/news about NAT support on the ESP-IDF 4 version?

david-cermak, 2020-03-05:
Hi all, it seems like the NAT functionality is a very desirable and useful feature and have to consider adding it into `esp-lwip`.

@martin-ger Thanks for your effort to support the community! Would you mind if your commits were taken in to the Espressif lwip repo?

martin-ger, 2020-03-05:
Would love to see that.

closed 2020-04-15 by igrr