espressif/esp-idf #1529: Bluetooth gatt_security_server example

chegewara, 2018-01-21:
Right after reboot, no bonded devices. Logs (`E (8826) BT: l2cble_start_conn_update, the last connection update command still pending.`):
```
I (693) heap_init: Initializing. RAM available for dynamic allocation:
I (700) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (706) heap_init: At 3FFCAAC0 len 00015540 (85 KiB): DRAM
I (712) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (718) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (725) heap_init: At 40093820 len 0000C7E0 (49 KiB): IRAM
I (731) cpu_start: Pro cpu start user code
I (77) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (126) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (396) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (396) SEC_GATTS_DEMO: app_main init bluetooth
I (436) SEC_GATTS_DEMO: The number handle = 8
I (436) SEC_GATTS_DEMO: advertising start success
I (7996) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (8586) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (8586) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (8826) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
E (8826) BT: l2cble_start_conn_update, the last connection update command still pending.
I (8906) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (8906) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (8906) SEC_GATTS_DEMO: address type = 0
I (8906) SEC_GATTS_DEMO: pair status = success
I (8916) SEC_GATTS_DEMO: Bonded devices number : 1

I (8916) SEC_GATTS_DEMO: Bonded devices list : 1

I (8926) SEC_GATTS_DEMO: 78 1f db c0 c7 f3
```

PS After disconnecting re-connect time is very long or there is GATT INTERNAL ERROR (0x81) followed by GATT CONN TIMEOUT error (0x8), just like in other issue reported.

Weijian-Espressif, 2018-02-11:
@chegewara can you uodate to the latest ESP-IDF and try ? we fix a bug for bonded device.

chegewara, 2018-02-16:
@Weijian-Espressif Nope, still not working. At least not with my samsung SGS4, android 5.0.1. Its ble v4.0 device. Ive had to modify a little bit gatt_secure_server example to not use private address to:
`esp_ble_gap_config_local_privacy(false)`

This is output log after reseting esp32 and connecting with phone:
```
I (511) SEC_GATTS_DEMO: app_main init bluetooth
D (521) nvs: nvs_open_from_partition bt_config.conf 1
D (521) nvs: nvs_get_str_or_blob bt_cfg_key0
D (521) nvs: nvs_get_str_or_blob bt_cfg_key0
D (531) nvs: nvs_close 2
D (531) nvs: nvs_open_from_partition bt_config.conf 1
D (541) nvs: nvs_set_blob bt_cfg_key0 496
D (541) nvs: nvs_close 3
D (561) nvs: nvs_open_from_partition bt_config.conf 1
D (561) nvs: nvs_set_blob bt_cfg_key0 530
D (601) nvs: nvs_close 4
V (601) SEC_GATTS_DEMO: event = 0

V (611) SEC_GATTS_DEMO: event = 16

I (611) SEC_GATTS_DEMO: The number handle = 8
V (611) SEC_GATTS_DEMO: GAP_EVT, event 22

V (611) SEC_GATTS_DEMO: event = c

V (611) SEC_GATTS_DEMO: GAP_EVT, event 0

V (621) SEC_GATTS_DEMO: GAP_EVT, event 1

V (631) SEC_GATTS_DEMO: GAP_EVT, event 6

I (631) SEC_GATTS_DEMO: advertising start success
V (36061) SEC_GATTS_DEMO: event = e

I (36061) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
E (36401) BT: L2CA_EnableUpdateBleConnParams - unknown BD_ADDR 781fdbc0c7f3
E (36401) BT: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 781fdbc0c7f3 not connected
E (36401) BT: Device not found

E (36411) BT: BTA got unregistered event id 31

E (36411) BT: BTA got unregistered event id 31

V (36421) SEC_GATTS_DEMO: event = f

I (36421) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT
D (36431) nvs: nvs_open_from_partition bt_config.conf 1
D (36431) nvs: nvs_set_blob bt_cfg_key0 284
D (36441) nvs: nvs_close 5
V (36441) SEC_GATTS_DEMO: GAP_EVT, event 8

I (36441) SEC_GATTS_DEMO: remote BD_ADDR: 781fdbc0c7f3
I (36451) SEC_GATTS_DEMO: address type = 0
I (36451) SEC_GATTS_DEMO: pair status = fail
I (36461) SEC_GATTS_DEMO: Bonded devices number : 0

I (36471) SEC_GATTS_DEMO: Bonded devices list : 0

V (36471) SEC_GATTS_DEMO: GAP_EVT, event 6

I (36481) SEC_GATTS_DEMO: advertising start success

```

chegewara, 2018-02-16:
Ok, ive been diging and looks like my tests procedure sucks. I have to find new phone with bluetooth 4.2.
I think this issue can be closed just for now.

projectgus, 2018-02-16:
Closing for now. @chegewara please reopen if you think IDF is doing the wrong thing here.

closed 2018-02-16 by projectgus

chegewara, 2018-03-26:
@projectgus @Weijian-Espressif Looks like issues with security are solved (finally), at least with this example. Even my old samsung S4 has no more issues with this. But:
there is one more issue hard to track and i really dont know where it come from; during connection there is two times connection parameters update request with about 400ms one after another, and with samsung S5 (marshmallow) and samsung S9+(oreo) i dont have issue, smartphone is fast enough to handle both requests, but with samsung S4 (lollipop) im getting this error:
`E (31124) BT: l2cble_start_conn_update, the last connection update command still pending.`

This sometimes causing issue with connection, but only on oldest one.

EspressifApp, 2018-03-27:
@chegewara When connecting for the first time, bludriod will reduce the connection parameters when searching for the service, and then restore the parameters when done. Pairing will also adjust the connection parameters. you can test your phone with gatt_service_table demo.

chegewara, 2018-03-27:
@EspressifApp The issue with this message and older smartphone is when esp32 and phone is already bonded/paired and im trying to connect again after disconnecting. 
`E (31124) BT: l2cble_start_conn_update, the last connection update command still pending.`
Im not saying its bug, its a chance it is android side issue, because for sure i have issue when im trying to disconnect from esp32 and esp32 is not receiving disconnect message. But its only with old samsung S4 and ive found it is known issue. So i can live with that and im not complaining about it, just wanted to mention.

illysky, 2018-03-30:
This is happening with LightBlue on iOS also. I can connect once, but then I can connected after this because of "l2cble_start_conn_update, the last connection update command still pending"

Lube, 2018-09-05:
Can confirm, it works on my Moto G4 (Android 7), but fails after trying to reconnect with my Samsung J7 (Android 5.1)