espressif/esp-idf #1533: make flash经常报错，自己多试几次build project，make flash等操作就又成功执行make flash。

liuyang77886, 2018-01-22:
我make flash经常报错，自己多试几次build project，make flash等操作就又成功执行make flash。
make[1]: Entering directory '/home/aithinker/project/hello_world/build/xtensa-debug-module'
make[1]: Nothing to be done for 'build'.
make[1]: Leaving directory '/home/aithinker/project/hello_world/build/xtensa-debug-module'
。。。。。。中间省略
python /home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "2MB"  -o /home/aithinker/project/hello_world/build/hello-world.bin /home/aithinker/project/hello_world/build/hello-world.elf
esptool.py v2.1
Flashing binaries to serial port /dev/ttyS2 (app at offset 0x10000)...
python /home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "/dev/ttyS2" --baud 115200 --before "default_reset" --after "hard_reset" write_flash -z --flash_mode "dio" --flash_freq "40m" --flash_size detect 0x1000 /home/aithinker/project/hello_world/build/bootloader/bootloader.bin 0x10000 /home/aithinker/project/hello_world/build/hello-world.bin 0x8000 /home/aithinker/project/hello_world/build/partitions_singleapp.bin
esptool.py v2.1
Connecting....
Traceback (most recent call last):
  File "/home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 2524, in <module>
    _main()
  。。。。。。中间省略
  File "/home/aithinker/project/esp32/esp-idf/components/esptool_py/esptool/esptool.py", line 1559, in slip_reader
    waiting = port.inWaiting()
  File "/usr/lib/python2.7/site-packages/serial/serialutil.py", line 572, in inWaiting
    return self.in_waiting
  File "/usr/lib/python2.7/site-packages/serial/serialposix.py", line 456, in in_waiting
    s = fcntl.ioctl(self.fd, TIOCINQ, TIOCM_zero_str)
IOError: [Errno 22] Invalid argument
make: *** [/home/aithinker/project/esp32/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1

malaimoo, 2018-01-23:
这个看起来像是你电脑 usb 串口的问题 ，亦或是 esptool 用到的 serial package 的 bug。

FayeY, 2018-01-26:
Hi @liuyang77886 , did the above suggestion help you resolve the issue? Is this issue still unsolved with the latest IDF?

closed 2018-02-23 by FayeY