espressif/esp-idf #1534: console example crash

chegewara, 2018-01-22:
While trying to join AP console app crash:
```
V (198658) event: exit default callback
D (201008) event: SYSTEM_EVENT_STA_CONNECTED, ssid:, ssid_len:11, bssid:, channel:1, authmode:3
V (201008) event: enter ***ERROR*** A stack overflow in task eventTask has been detected.
abort() was called at PC 0x400889ec on core 0
```

igrr, 2018-01-27:
Fixed in https://github.com/espressif/esp-idf/commit/24b07a3f4534a48a0556bf5257558e756334bbfd.

igrr, 2018-01-27:
My bad, that's a different issue.

liuzfesp, 2018-02-01:
Hi @chegewara the crash says the stack is overflow. As a workaround, please enlarge the stack size of eventTask when configure the log level into verbose level.  You can modify it in esp_task.h.  We will fix it in v3.1

closed 2018-02-11 by igrr