espressif/esp-idf #1538: Brownout trips when VDD3P3_RTC is fed 1.8v

hchaudhary1, 2018-01-24:
I have developed a 1.8v ESP32-D2WD board. It is working well in all aspects, except I have disable the brownout. I am unable to configure the brownout to something below 1.8v.

The entire digital domain: VDD3P3_RTC, VDD3P3_CPU, and VDD_SDIO are powered from 1.8v 😈 
Looking into the menuconfig and the Technical Ref Manual 2.8, the brownout has a 3-bit (eight settings). However, it is centered around 2.5v. It is not clear on how to center this around 1.8v...

How can I trigger the brownout at ~1.7v?

projectgus, 2018-01-24:
Hi @hchaudhary1 ,

This question got asked on the forum recently (are you the same person?) The short answer is, unfortunately you can't configure the brownout detector to run any higher than 1.8V:
https://esp32.com/viewtopic.php?f=13&t=4266&sid=8657805189f73b869a682beffc68d641

Angus

hchaudhary1, 2018-01-24:
yes, same person.. just posting on github so it can get tracked

projectgus, 2018-01-24:
This is a hardware limitation, unfortunately we can't fix it in software.

I've updated the forum post with the latest information from the hardware team.

closed 2018-01-24 by projectgus

hchaudhary1, 2018-02-12:
hi @projectgus I saw that a new Tech Ref Manual has been released, however, you guys did NOT update the 1.8v descriptions. Any update on this?
Also, PICO chip's documents has same mistakes..

thanks :D

projectgus, 2018-02-13:
I believe the documentation team plans to update the datasheet & TRM together. However this probably won't happen until after Chinese New Year.

Thanks for the heads-up about the PICO datasheet, as well.

hchaudhary1, 2018-04-18:
@projectgus hi. Can this issue be re-opened? Better to close this when the documentation is updated... Looks like many months since I reported this issue, and documentation still is not updated. 

Reference Manual & Hardware Design Guidelines need to updated

Please refer here for more details:
https://esp32.com/viewtopic.php?f=13&t=4266#p19148

thanks :D

projectgus, 2018-04-19:
Hi @hchaudhary1 ,

Sorry for the delay in releasing the documentation. I assure you that it is still going to be released, as it happens I saw a final draft of the new version earlier this week.

Thanks for the pointers to the other documents which mention the voltage range, I've mentioned these to the documentation team as well.

The reason the issue was closed here is that it's not related to ESP-IDF. There's nothing for us to fix in the ESP-IDF code or repo. I realise there is no issue tracker for documentation issues, the best place to report these is the esp32.com forum.

hchaudhary1, 2018-04-19:
Thanks for the update