espressif/esp-idf #1539: Changing Serial flasher settings rebuilds entire project

hex007, 2018-01-24:
Just changing Serial flasher settings should not cause entire project to be rebuilt.

Version: d89542eb2b41d867882b822ed3e1c7cb6c97f05c

projectgus, 2018-01-24:
Agreed.

There is a workaround for this, which is you can pass ESPPORT & ESPBAUD as environment variables or on the command line (ie `make ESPPORT=COM4`), and this overrides whatever is set in the config (and doesn't trigger any rebuilding).

We are working on some changes to the configuration system which will decouple these settings from the main project configuration. Will update this issue when they're available.

hex007, 2019-02-17:
@projectgus Any update on this ?

projectgus, 2019-02-17:
@hex007 The workaround of setting environment variables remains the best option if you're using the GNU Make based build system.

The [CMake based build system is currently in preview release](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html) but will become the default build system in IDF V4.0. In this sytem the serial port and baud rate settings are removed from sdkconfig entirely so they no longer appear in menuconfig and therefore can't trigger full rebuilds. If using this build system, you can set the port & baud on the command line ie `idf.py -p COM4 -b 921600 build flash monitor` or you can still set them via environment variables. See the linked "Getting Started" guide above if you're interested in this.

This issue will probably stay open as a "known issue" until CMake becomes the default build system, and then we will close it.

projectgus, 2019-08-01:
CMake based build system is now the default in IDF (v4.0-dev):
https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html

... and does not exhibit this issue.

For GNU Make users, setting ESPPORT & ESPBAUD environment variables remains the best workaround.

closed 2019-08-01 by projectgus