espressif/esp-idf #1541: Bluetooth classic SPP profile: Provide pin code for a connection

my-abousamra, 2018-01-24:
Hi,

I'm trying to connect to a bluetooth classic device using SPP profile but this device needs 6 digits pin code to accept connection so how can I provide this pin code from my ESP32 application, Plus how can I set a pin code for my device in order to make other devices provide it whenever want to connect to mine. I've browsed the esp code but in vain. I don't know if it is supported yet or not and if not when will it be released?

Thanks

Weijian-Espressif, 2018-02-11:
@blueMoodBHD Please help to see this problem

blueMoodBHD, 2018-02-12:
@my-abousamra I'm sorry to tell you that we don't support pin code new, and we will have a discussion of whether to support  that and the specific time. 
Thanks for your understanding.

chegewara, 2018-02-12:
@blueMoodBHD 
Maybe i misunderstanding something here, but bluetooth LE is supporting pin authentication and classic bluetooth does not? How so.

blueMoodBHD, 2018-02-12:
I mean that BT does not provide APIs related to this feature now, we set IO capacity as noinputnooutput , so encryption will not use the pin code. And we will provide the APIs leater.

my-abousamra, 2018-02-12:
@blueMoodBHD 
I don't find a reason why not to support pin code for BT classic. Anyway, you may have your reasons. I want to connect to BT classic devices that need a pin code and now I can't connect to them. If you wan't support this feature then ESP32 BT classic is useless for me and I should try another one cause your module doesn't support that small feature!

blueMoodBHD, 2018-02-23:
@my-abousamra Sorry for the late reply, I just finished the holiday. Now, we set IO capacity as noinputnooutput, so you can connect to other devices without pin code. And, we will support pin code as soon as possible.

my-abousamra, 2018-03-20:
@blueMoodBHD any updates, please?

blueMoodBHD, 2018-03-21:
@my-abousamra We will try to finish this in v3.1, thanks.

my-abousamra, 2018-04-19:
@blueMoodBHD  when will it be released?

avinashs24, 2018-06-06:
I have a requirement too for this functionality. Any idea on when it will be implemented?

blueMoodBHD, 2018-06-07:
I have almost completed this feature, but since v3.1's feature release has been completed, it cannot be merged into ESP-IDF. After v3.1 is officially released, this feature will be integrated into master.

blueMoods, 2018-07-16:
Hi all,
This feature has merged into Master, now. Refer to commit: 8571170abe3b2b7e5344e9603dadf2e6f74916e9

blueMoodBHD, 2018-07-16:
Hi all,
This feature has merged into Master, now. Refer to commit: 8571170

my-abousamra, 2018-07-18:
Just works 
Thank you

closed 2018-07-18 by my-abousamra

Cividati, 2018-09-22:
how can I import the code of the password into my code? I don't have a lot of experience with this. Thx for attention.

my-abousamra, 2018-09-23:
@Cividati  check bluetooth examples in esp-idf

chegewara, 2018-11-26:
Hi @blueMoodBHD, i know that feature has been tested and confirmed its working when esp32 is connecting to peer device with fixed pin. My question is, is it possible to setup esp32 with to be accessed secure simple pairing as well? If yes, it would be nice to have example or some direction where it should be implemented in code. Thanks.

blueMoodBHD, 2018-11-27:
Hi @chegewara , Yes, we support Secure Simple Pairing, and we have examples `example/bluetooth/bt_spp_acceptor`. In `app_main`, we set IO capacity first:
```
     /* Set default parameters for Secure Simple Pairing */
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));
```
    Then, if the remote device support Secure Simple Pairing, too. We will use it when pairing. `esp_bt_gap_cb` will callback with `ESP_BT_GAP_CFM_REQ_EVT`,`ESP_BT_GAP_KEY_NOTIF_EVT` or `ESP_BT_GAP_KEY_REQ_EVT`.

chegewara, 2018-11-30:
Hi @blueMoodBHD what i mean is Simple Secure Pairing with fixed pin. I changed this code:
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bt_spp_vfs_acceptor/main/example_spp_vfs_acceptor_demo.c#L208-L218

to match fixed pin:
```
    /* Set default parameters for Secure Simple Pairing */
    esp_bt_sp_param_t param_type = ESP_BT_SP_IOCAP_MODE;
    esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_OUT;
    esp_bt_gap_set_security_param(param_type, &iocap, sizeof(uint8_t));

    /*
     * Set default parameters for Legacy Pairing
     * Use fixed pin code
     */
    esp_bt_pin_type_t pin_type = ESP_BT_PIN_TYPE_FIXED;
    esp_bt_pin_code_t pin_code;
    pin_code[0] = '1';
    pin_code[1] = '2';
    pin_code[2] = '3';
    pin_code[3] = '4';
    esp_bt_gap_set_pin(pin_type, 4, pin_code);
```
And i have 2 results:
- with ` esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_IO;` there is connection with pin comparison,
- with `esp_bt_io_cap_t iocap = ESP_BT_IO_CAP_OUT;` there is no secured connection (no pin request or pin comparison).

blueMoodBHD, 2018-12-03:
Hi @chegewara ,  Fixed pin is used for  Legacy Pairing, and there is no fixed pin in Simple Secure Pairing

Archie1978, 2019-12-12:
Hello  @chegewara , I retesting  remake the example bt_spp_initiator, a2dp_sink, bt_spp_vfs_initiator, a2dp_sink and bt_spp_acceptor. 

   There is nothing work in examples, Is there something to change in the conf ?
And BE work with pin code ?

chegewara, 2019-12-12:
At the time ive been testing it worked in BLE. To use fixed pin with classic bluetooth you have to change settings in menuconfig.

Archie1978, 2019-12-12:
Ok, I had changed by menuconfig 

> Component config → Bluetooth → Bluetooth controller → Bluetooth controller mode (BR/EDR/BLE/DUALMODE)→BLE only
> Component config → Bluetooth → Bluetooth Host→NimBLE - BLE only

``
CC build/main/main.o
In file included from /home/archie/workspace/esp-idf/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/main.c:28:0:
/home/archie/workspace/esp-idf/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_av.h:13:26: fatal error: esp_a2dp_api.h: No such file or directory
compilation terminated.
/home/archie/workspace/esp-idf/make/component_wrapper.mk:291: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
/home/archie/workspace/esp-idf//make/project.mk:610: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2
``
Other problem "Secure Simple Pairing" is only on "Classic Bluetooth" in the menu.
[ ] Bluedroid memory debug (NEW)
[*] Classic Bluetooth
[ ]     A2DP (NEW)
[ ]     SPP (NEW)
[ ]     Hands Free/Handset Profile (NEW)
[*] Secure Simple Pairing (NEW)
[*] Bluetooth Low Energy (NEW)
[*]     Include GATT server module(GATTS) (NEW)
            GATTS Service Change Mode (GATTS automatically send service change indication)  --->
[*]     Include GATT client module(GATTC) (NEW)
[ ]         Save gattc cache data to nvs flash (NEW)
[*]     Include BLE security module(SMP) (NEW)
[ ]         Slave enable connection parameters update during pairing (NEW)
[ ] Disable BT debug logs (minimize bin size) (NEW)

Can you explain , the problem ?

chegewara, 2019-12-12:
Like i said, for BLE bluedroid stack it works without any changes, not NimBLE.
For classic bluetooth to use fixed pin you have to disable SSP as per this comment:
https://github.com/espressif/esp-idf/issues/1541#issuecomment-443575348

Archie1978, 2019-12-15:
Great, the compilation work:
(Top) → Component config → Bluetooth → Bluedroid Options → Secure Simple Pairing = false

Note: I tested the code on phone, windows and linux. Only linux did not work :/