espressif/esp-idf #1542: Which SPI PON to use when  it's by default

hpwit, 2018-01-25:
in order to go really fast on SPI
I want to use what it's written here
"You can use this structure to specify the GPIO pins of the bus. Normally, the driver will use the GPIO matrix to route the signals. An exception is made when all signals either can be routed through the IO_MUX or are -1. In that case, the IO_MUX is used, allowing for >40MHz speeds."

When I do that I can put speed up to 80MHZ but I cannot find the pins to use for clock and data !!:)
any idea ?

negativekelvin, 2018-01-25:
Check function 2 column of io mux table in datasheet

chegewara, 2018-01-25:
https://raw.githubusercontent.com/gojimmypi/ESP32/master/images/myESP32%20DevKitC%20pinout.png

hpwit, 2018-01-25:
Hello
I am using arduino to program my esp32, and I think there is on issue there when I try to do that.
By looking at the code of the spi_common.c in this library it would work but i can't find it in the Arduino_eps32 is that normal ? I would like to check if the implementation is the same 
thk you

negativekelvin, 2018-01-25:
Arduino uses it's own spi driver and it does not support io mux https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c

hpwit, 2018-01-25:
i can find https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/driver/driver
the driver spi_master
and i was able and use that  code via arduino and it worked (as long as I precise the pins number)

esp_err_t ret;
    spi_device_handle_t spi;
    spi_bus_config_t buscfg={
        .miso_io_num=PIN_NUM_MISO,
        .mosi_io_num=PIN_NUM_MOSI,
        .sclk_io_num=PIN_NUM_CLK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1
    };
    spi_device_interface_config_t devcfg={
        .clock_speed_hz=10*1000*1000,               //Clock out at 10 MHz
        .mode=0,                                //SPI mode 0
        .spics_io_num=PIN_NUM_CS,               //CS pin
        .queue_size=7,                          //We want to be able to queue 7 transactions at a time
        .pre_cb=lcd_spi_pre_transfer_callback,  //Specify pre-transfer callback to handle D/C line
    };
    //Initialize the SPI bus
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 1);
    assert(ret==ESP_OK);
    //Attach the LCD to the SPI bus
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &spi);
assert(ret==ESP_OK);

how come ?

negativekelvin, 2018-01-25:
Yes you can use it but it won't be tied into Arduino spi calls used by other libraries.

hpwit, 2018-01-26:
Hello
Thank you for your answer.
I know it won't be tied to other SPI librairies. I have an issue using the
code as it is. it seems that it does not recognize that i am using default
SPI ports => when i put a frequency > 32Mhz it crashes.
When i do not specify any pins I can go faster but no signal on any pin "I
have checked with an oscilloscope (just the pin CLK0 does something a bit
weird)"
rgds
Yves

ginkgm, 2018-03-27:
link to #1008

ginkgm, 2018-03-27:
Hi @hpwit ,
Please see #1008 first, here are some more clues about this issue.
1.  ESP32 spi master can only work under divisors of 80MHz, i.e. 80, 40, 26.6, 20, 16, 13.3, ... Mhz
2. Native pins can be find in our [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf), as well as the IO_MUX chapter of the [Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf). When the one of the pins are set to non-native pins, the driver automatically route signals through the GPIO matrix. We are going to add pin information into the [programming guide](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/spi_master.html).
3. There're different restrictions for writing and reading, and the writing limit is always higher than reading. The freq limit for writing is 80MHz with native pins and 40MHz for GPIO matrix pins.
4. For reading, the limits come from timing issue: data comes too late compare to the clock, violating the setup requirements when the clock is fast. When the native pins is not used (so the GPIO matrix is used), the data comes even later. 

    For this we have a workaround, inserting dummy cycles right before the reading phase. However this is only for half-duplex mode, the full-duplex mode cannot use it. This is why the frequency checking exist [there](https://github.com/espressif/esp-idf/blob/9ea38560ecb2b2c2c854ad6862db21bd4319f6e1/components/driver/spi_master.c#L262).

    In summary, the reading frequency limit is related to whether native pins are used, whether dummy bits can be used (whether in half-duplex mode), and the time required for the slave to validate the MISO after the launch edge.

    Recently we released a patch(9d69d1106754926e4ee6590dcb71b09e4abf558d) for users to operate devices under 40MHz in full-duplex mode while write-only.
5. Important notice: the intervals between transactions are long (25us typ.). When data are sent by small transactions, the intervals cost a lot. Try to squash data into one transaction if possible to avoid intervals.

closed 2018-04-03 by igrr

hpwit, 2018-04-10:
Hello @ginkgm 
thank you for your answer.
I guess the 25us between transactions is what throw off my calculation
a went to send 32bits by 32 bits I have a loop for 40000 reps of the write command
but i needs to consider that I have 40 000 000 * 25us of offset
Yves