espressif/esp-idf #1543: logger stuck on 76923 baud rate

morganrallen, 2018-01-25:
Not quite sure where to start with providing feedback here... `make menuconfig` has baudrates set to `115200`, confirmed in `sdkconfig` yet `76923` is the only speed it seems to work.
I'm using `miniterm.py` from `python-serial`

```
17:18 $ miniterm.py /dev/ttyUSB0 76923 --raw
--- Miniterm on /dev/ttyUSB0  76923,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
*�P�V��ET�Q�UU�T��-�QRQT�JU�WU�J���u������W*��R�uRuT�T��U���-QBU��j����T�T�Z�Q�WխQ(�uT�Z�QJ-]�Z�,QR�R�Z�QڬR�Z�Q����I���T�U�UZU%�(Ք��(��UVK%�(Ք��(�J�jZ�%�(ՔTQeQR��TTT�T�+���u����T�\E�YQ�UV+��Ej��P�EԵ��TI (20) boot: ESP-IDF v3.1-dev-270-g2a10ffce-dirty 2nd stage bootloader
I (20) boot: compile time 17:11:33
I (20) boot: Enabling RNG early entropy source...
```

```
17:20 $ miniterm.py /dev/ttyUSB0 115200 --raw
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7400
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:14840
entry 0x400792d8
��o����lo�
          #�l#�p�n�$���#�����|$$#��ܒ#�lB�
```

So it appears to switch when the logger takes over. I've been ignoring it for a while.. but `gdb` won't connect on that rate.

igrr, 2018-01-25:
Which development board are you using? It might be that the board is equipped with a 26MHz crystal whereas your (default) configuration is set to 40MHz. Please check this section of the getting started guide if that is the case: http://esp-idf.readthedocs.io/en/latest/get-started/index.html#monitor

morganrallen, 2018-01-25:
Oh nailed it! For the record it's a Heltec/TTGO LoRa OLED board.

closed 2018-01-25 by morganrallen