espressif/esp-idf #1544: SD card in SPI mode fails to initialize

jcherry4, 2018-01-25:
Hi I am trying to use sd_card_example and I get the following output on the terminal:

I (414) example: Initializing SD card
I (414) example: Using SPI peripheral
D (424) sdspi_host: sdspi_host_init_slot: SPI2 miso=19 mosi=23 sck=18 cs=5 cd=-1 wp=-1, dma_ch=1
D (434) spi: SPI1 use gpio matrix.
V (434) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (444) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (444) intr_alloc: Connected src 30 to int 13 (cpu 0)
D (454) spi_master: SPI1: New device added to CS0, effective clock: 400kHz
I (464) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (474) sdmmc_cmd: sdmmc_card_init
V (474) sdmmc_cmd: sending cmd slot=1 op=0 arg=0 flags=20 data=0x0 blklen=0 datalen=0 timeout=1000
V (484) sdspi_host: sdspi_host_start_command: slot=1, CMD0, arg=0x00000000 flags=0x4, data=0x0, data_size=0 crc=0x4a
V (494) sdspi_transaction: r1 = 0x01 hw_cmd.r[0]=0xffffffff
V (494) sdmmc_cmd: cmd response 00000001 00000000 00000000 00000000 err=0x0 state=0
V (524) sdmmc_cmd: sending cmd slot=1 op=0 arg=0 flags=20 data=0x0 blklen=0 datalen=0 timeout=1000
V (524) sdspi_host: sdspi_host_start_command: slot=1, CMD0, arg=0x00000000 flags=0x4, data=0x0, data_size=0 crc=0x4a
V (524) sdspi_transaction: r1 = 0x01 hw_cmd.r[0]=0xffffffff
V (534) sdmmc_cmd: cmd response 00000001 00000000 00000000 00000000 err=0x0 state=0
V (564) sdmmc_cmd: sending cmd slot=1 op=8 arg=1aa flags=1c30 data=0x0 blklen=0 datalen=0 timeout=1000
V (564) sdspi_host: sdspi_host_start_command: slot=1, CMD8, arg=0x000001aa flags=0x20, data=0x0, data_size=0 crc=0x43
V (564) sdspi_transaction: r1 = 0x05 hw_cmd.r[0]=0xffffffff
V (574) sdmmc_cmd: cmd response ffffffff 00000000 00000000 00000000 err=0x0 state=15
D (584) sdmmc_cmd: sdmmc_send_cmd_send_if_cond: received=0xff expected=0xaa
E (584) sdmmc_cmd: sdmmc_card_init: send_if_cond (1) returned 0x108
D (594) vfs_fat_sdmmc: sdmmc_card_init failed 0x(108)
V (604) intr_alloc: esp_intr_free: Disabling int, killing handler
E (604) example: Failed to initialize the card (264). Make sure SD card lines have pull-up resistors in place.

 I have updated esp_idf as of 1/25/18 and have PU resistors on all lines.
I am using SPI mode
I have tried the following pin assignments:

 PIN_NUM_MISO 2
 PIN_NUM_MOSI 15
 PIN_NUM_CLK  14
 PIN_NUM_CS   13

 PIN_NUM_MISO 19
 PIN_NUM_MOSI 23
 PIN_NUM_CLK  18
 PIN_NUM_CS   5

Also I have verified operation using Arduino IDE and and SD_test code.

Any help with this would be much appreciated.

Thanks,

John

jcherry4, 2018-02-05:
I believe I solved the problem "well sort of" 
It turns out the current esp_idf code base 1/25/18 does not support standard density SD cards <2GB in SPI mode.
In tracing through the code  sdmmc_card_init sends a code 8 to the SD card, if the card is <2G it will not send a response and the following error is issued:
D (584) sdmmc_cmd: sdmmc_send_cmd_send_if_cond: received=0xff expected=0xaa

I installed a 16GB card and verified it worked.
So I am wondering if there are any plans to support  standard density SD cards in SPI mode?

Thanks ,
John

igrr, 2018-02-06:
Hi John, sorry for the late response. I was able to reproduce the issue with one of the SD cards, fix is coming shortly.

igrr, 2018-02-06:
My bad, the error I was seeing ended up being unrelated.

The attached patch might solve the problem for you, could you please try it?

[0001-sdspi-handle-error-flags-for-R3-R7-responses.patch.zip](https://github.com/espressif/esp-idf/files/1697581/0001-sdspi-handle-error-flags-for-R3-R7-responses.patch.zip)

jcherry4, 2018-02-17:
Hi Igrr,

Sorry to took so long to get back to you.
I don't understand how a patch gets integrated into a source file.
Is there a script I run to do it automatically?

Thanks John

JosuGZ, 2019-01-17:
@jcherry4 You should be able to apply the patch by using `git apply <filename>`.

https://git-scm.com/docs/git-apply

igrr, 2020-02-13:
The patch has been merged some time ago in bbbdec24d93d1edab2b7b3e6a7c1b6b4667c319c. Please feel free to re-open if the issue still exists.

closed 2020-02-13 by igrr