espressif/esp-idf #1545: mdns_server_t not defined in mdns.h

jestebanes, 2018-01-25:
Hi, maybe it was my mistake, but when I try to use mdns.c component the compiler says

unknown type mdns_server_t

This type is declared in mdns/private_include/mdns_private.h
But if I include this header, then the compiler start to find others undefined types
mdns_ip_protocol_t
mdns_result_t
mdns_txt_item_t

I think something is missing or is a work in progress

Best regards

me-no-dev, 2018-01-25:
actually the whole mdns server is rewritten and the API has changed :) you can check the example on the usage now.

negativekelvin, 2018-01-25:
@me-no-dev how do you bind servers/ports/services to different interfaces and query on specific interfaces?

me-no-dev, 2018-01-25:
@negativekelvin you do not. It binds on all available interfaces (requires a function to be called in the system event handler for now). When searching, you do get the interface that the result came on in the result structure, so you can filter the interface you want. If you want it to act only on a particular interface, you can start mdns before you start the network and then pass on only the events for the interface you want.

negativekelvin, 2018-01-25:
@me-no-dev not a fan of that approach.  It doesn't really fit with multiple interface use.  For example it is likely a service will be only available on a single interface or that you don't want to send any query packets on an interface or a service with the same name and type on one interface will have a different port number on another interface. Etc...  And this should be able to be configured manually.

me-no-dev, 2018-01-25:
@negativekelvin the service now is RFC compliant. It's how it work on all computers and is Apple certified. What you are saying, I have not seen to be configurable anywhere. Can you please give some examples? Any real use case?

negativekelvin, 2018-01-25:
If I am using more than one of ESP_IF_WIFI_STA, ESP_IF_WIFI_AP, ESP_IF_ETH at a time and I create a webserver and bind it to ESP_IF_WIFI_AP then naturally I only want to advertise it on ESP_IF_WIFI_AP so I would expect `mdns_service_add("ESP32-WebServer", "_http", "_tcp", 80, serviceTxtData, 3)` to take an argument where I can specify one or more interfaces.  And the same with init, setting a hostname and instance name per interface, querying, etc...

me-no-dev, 2018-01-26:
Can your computer do that? While I can agree that there could be services running only on particular interface, hostname and instance name are not something that is set based on interface. They are global. That's why the first part of hostname is "host" and not "interface". I could look into specifying interface for service but that is either one or all interfaces, not more than one but not all, because you can attach a service either to a particular IP or to all IPs (sockets do not bind to interfaces, they bind to IP or to ANY IP). As far as searching goes, all implementations search on all interfaces and tell you on which interface the answer came, so it's up to you to filter the results if you care only for one of them.
Show me how this can be done with avahi or bonjour and I will look into it right away ;)
Here is the help from Apple's mdns on computers:
```
dns-sd -E                              (Enumerate recommended registration domains)
dns-sd -F                                  (Enumerate recommended browsing domains)
dns-sd -R <Name> <Type> <Domain> <Port> [<TXT>...]             (Register a service)
dns-sd -B        <Type> <Domain>                     (Browse for service instances)
dns-sd -L <Name> <Type> <Domain>                       (Resolve a service instance)
dns-sd -Q <name> <rrtype> <rrclass>             (Generic query for any record type)
dns-sd -Z        <Type> <Domain>               (Output results in Zone File format)
dns-sd -G     v4/v6/v4v6 <name>              (Get address information for hostname)
dns-sd -H                                   (Print usage for complete command list)
dns-sd -V                (Get version of currently running daemon / system service)
```

negativekelvin, 2018-01-26:
I don't think whether a typical device can do something or what apple does is a good reason for implementation decisions.  The question should be is that something you would want to do with esp32.  And the answer is yes, there are cases where I would want esp32 to act like two completely separate devices concurrently on AP and STA or AP and ETH  If it was a computer I would run two mdns instances each bound to a single interface.  Binding a service to one interface or all is fine.  I don't see any good reason why query can't be limited to one or all as well since it just iterates through interfaces anyway.  
```
static void _mdns_search_send(mdns_search_once_t * search)
{
    uint8_t i, j;
    for (i=0; i<TCPIP_ADAPTER_IF_MAX; i++) {
        for (j=0; j<MDNS_IP_PROTOCOL_MAX; j++) {
            _mdns_search_send_pcb(search, (tcpip_adapter_if_t)i, (mdns_ip_protocol_t)j);
        }
    }
}
```
I don't have a problem with all results being processed together.

me-no-dev, 2018-01-26:
Feel free to implement and PR the changes you are proposing. Make sure that the service passes Bonjour Conformance Tests, cuz regardless what you think about Apple's decisions, it's their service and I rewrote mDNS to be specifically compatible with them (HomeKit), therefore breaking functionality is unacceptable. I honestly do not have the time to go back into mDNS now.

me-no-dev, 2018-01-26:
>  If it was a computer I would run two mdns instances each bound to a single interface.

Good luck with that

negativekelvin, 2018-01-26:
Yes the official motto seems to be trending away from "ESP32: Anything is possible!" and toward "ESP32: Good luck with that!" lately

me-no-dev, 2018-01-26:
I think you are overreacting a bit ;) "Good luck with that" was about computer things and not ESP32. Don't mistake "Anything is possible!" with "We will make the ESP32 cook you a dinner if so you desire". As you see, what you want is totally possible, if you put a little work into it too ;)

negativekelvin, 2018-01-26:
Yes I know but there are other issues where idf implementations are unnecessarily limiting and it would be a great benefit if espressif devs would have the mindset to recognize and avoid that.  In this case you are right since the code in question is open source I as an uncompensated idf contributor could study and fix it and make a PR and it can sit around unmerged because it has breaking changes that conflict with the homekit branch and no one has time to test it. ;)

negativekelvin, 2018-01-26:
Sorry to @jestebanes for hijacking your issue

jestebanes, 2018-01-27:
It's ok, I'm learning a lot

david-cermak, 2019-03-15:
Hi @jestebanes 

Was this issue caused by an API change in mdns, is is still valid? 
Can we close the issue?

jestebanes, 2019-03-15:
There was a long time ago, I think that issue was solved at the next library compilation.

closed 2019-03-15 by jestebanes