espressif/esp-idf #1548: NTP Synchronization Issue

forthlightning, 2018-01-26:
I know this may be a long shot but I am seeing some weird behavior from my time(NULL) function. Hoping to get some clarity on where the issue might be! Any help would be appreciated!

I have several devices logging data in the field. They take a sensor reading every minute, timestamp the message, and then publish that message via MQTT. 

I can see by the ingestion time in the DB that we are receiving the data each minute as expected. But today the timestamp seemed to diverge from the ingestion time into the database. Seems all the data is clustering toward the beginning of the hour. For example the data for 3:00 am is labeled 2:58 and data for 3:10 am is labeled 3:00:15. 

I use `long int timestamp = time(NULL);` to assign the time... Not sure if this is user error or what!

After I remotely restarted the device the timestamps went back to normal and have been behaving as expected ever since. I have only seen this behavior one time before and it was several months ago.

```
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000483)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000481)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000473)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000472)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000469)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000463)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000460)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000456)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000452)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000449)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000445)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000441)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1517000438)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996890)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996889)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996888)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996887)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996885)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996884)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996883)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996882)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996880)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996879)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996878)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996877)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.6, 
    "Timestamp" : NumberInt(1516996876)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996874)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.5, 
    "Timestamp" : NumberInt(1516996873)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.6, 
    "Timestamp" : NumberInt(1516996872)
}
{ 
    "Tag" : "DataLabel0", 
    "Value" : 57.4, 
    "Timestamp" : NumberInt(1516996871)
}
```

igrr, 2018-01-27:
@forthlightning can you infer the time between last device restart and the start of odd device behavior, using the values in the database?
You mentioned 'several months', have the devices been running without reset during this time?

Also, would you mind attaching sdkconfig used in your application, and specifying IDF version used?

forthlightning, 2018-01-29:
So I have observed this behavior twice, once was about three months ago, at that time the device had been running for about 2 months. After observing this I implemented 12-hour device restarts in an effort to combat this issue. I had not seen this behavior again for the next three months, up until the morning of opening the issue. On that day the most recent restart was 02:00 PST, and 05:45  PST was the first reading with a strange timestamp.

I'm using the ESP-IDF commit 6cc809961033c9007a4250c8401c2c7e41c99d7b, quite a bit behind I know, but I have looked through commits and didn't see anything that looks relevant to what I'm seeing.

Here is the sdkconfig:

```
/*
 *
 * Automatically generated file; DO NOT EDIT.
 * Espressif IoT Development Framework Configuration
 *
 */
#define CONFIG_GATTC_ENABLE 1
#define CONFIG_ESP32_PHY_MAX_TX_POWER 20
#define CONFIG_TRACEMEM_RESERVE_DRAM 0x0
#define CONFIG_FREERTOS_MAX_TASK_NAME_LEN 16
#define CONFIG_BLE_SMP_ENABLE 1
#define CONFIG_TCP_RECVMBOX_SIZE 6
#define CONFIG_LWIP_ETHARP_TRUST_IP_MAC 1
#define CONFIG_TCP_WND_DEFAULT 5744
#define CONFIG_SPIFFS_USE_MAGIC_LENGTH 1
#define CONFIG_ESPTOOLPY_FLASHSIZE_4MB 1
#define CONFIG_IPC_TASK_STACK_SIZE 1024
#define CONFIG_ESPTOOLPY_FLASHFREQ "40m"
#define CONFIG_MBEDTLS_KEY_EXCHANGE_RSA 1
#define CONFIG_UDP_RECVMBOX_SIZE 6
#define CONFIG_MBEDTLS_AES_C 1
#define CONFIG_MBEDTLS_ECP_DP_SECP521R1_ENABLED 1
#define CONFIG_MBEDTLS_GCM_C 1
#define CONFIG_ESPTOOLPY_FLASHSIZE "4MB"
#define CONFIG_HEAP_POISONING_DISABLED 1
#define CONFIG_SPIFFS_CACHE_WR 1
#define CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER 1
#define CONFIG_SPIFFS_CACHE 1
#define CONFIG_INT_WDT 1
#define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL 1
#define CONFIG_MBEDTLS_SSL_PROTO_TLS1 1
#define CONFIG_MBEDTLS_ECDSA_C 1
#define CONFIG_ESPTOOLPY_FLASHFREQ_40M 1
#define CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS 3
#define CONFIG_MBEDTLS_ECDH_C 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ELLIPTIC_CURVE 1
#define CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM 5
#define CONFIG_MBEDTLS_SSL_ALPN 1
#define CONFIG_MBEDTLS_PEM_WRITE_C 1
#define CONFIG_LOG_DEFAULT_LEVEL_INFO 1
#define CONFIG_BT_RESERVE_DRAM 0x10000
#define CONFIG_IP_LOST_TIMER_INTERVAL 120
#define CONFIG_ESP32_PANIC_PRINT_REBOOT 1
#define CONFIG_MBEDTLS_ECP_DP_BP384R1_ENABLED 1
#define CONFIG_MBEDTLS_ECP_DP_SECP256K1_ENABLED 1
#define CONFIG_CONSOLE_UART_BAUDRATE 115200
#define CONFIG_LWIP_MAX_SOCKETS 4
#define CONFIG_LWIP_NETIF_LOOPBACK 1
#define CONFIG_EMAC_TASK_PRIORITY 20
#define CONFIG_TIMER_TASK_STACK_DEPTH 2048
#define CONFIG_TCP_MSS 1436
#define CONFIG_MBEDTLS_ECP_DP_CURVE25519_ENABLED 1
#define CONFIG_FATFS_CODEPAGE 1
#define CONFIG_ULP_COPROC_RESERVE_MEM 0
#define CONFIG_ESPTOOLPY_BAUD 921600
#define CONFIG_FLASHMODE_DIO 1
#define CONFIG_ESPTOOLPY_AFTER_RESET 1
#define CONFIG_OPTIMIZATION_ASSERTIONS_ENABLED 1
#define CONFIG_TOOLPREFIX "xtensa-esp32-elf-"
#define CONFIG_MBEDTLS_ECP_C 1
#define CONFIG_FREERTOS_IDLE_TASK_STACKSIZE 1024
#define CONFIG_MBEDTLS_RC4_DISABLED 1
#define CONFIG_SPIFFS_SIZE 1048576
[... 119 lines cut ...]
#define CONFIG_WL_SECTOR_SIZE_4096 1
#define CONFIG_OPTIMIZATION_LEVEL_DEBUG 1
#define CONFIG_MBEDTLS_ECP_DP_SECP192R1_ENABLED 1
#define CONFIG_FREERTOS_UNICORE 1
#define CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED 1
#define CONFIG_MBEDTLS_KEY_EXCHANGE_ECDH_ECDSA 1
#define CONFIG_SYSTEM_EVENT_QUEUE_SIZE 32
#define CONFIG_BT_ACL_CONNECTIONS 4
#define CONFIG_FATFS_MAX_LFN 255
#define CONFIG_ESP32_WIFI_TX_BUFFER_TYPE 1
#define CONFIG_ESPTOOLPY_BAUD_921600B 1
#define CONFIG_LWIP_LOOPBACK_MAX_PBUFS 8
#define CONFIG_APP_OFFSET 0x10000
#define CONFIG_MEMMAP_SMP 1
#define CONFIG_SPI_FLASH_ROM_DRIVER_PATCH 1
#define CONFIG_MBEDTLS_ECP_DP_SECP192K1_ENABLED 1
#define CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT 2048
#define CONFIG_LWIP_SO_RCVBUF 1
#define CONFIG_MONITOR_BAUD_OTHER_VAL 115200
#define CONFIG_NEWLIB_STDOUT_LINE_ENDING_CRLF 1
#define CONFIG_ESPTOOLPY_PORT "/dev/tty.SLAB_USBtoUART"
#define CONFIG_TASK_WDT_PANIC 1
#define CONFIG_SPIFFS_LOG_PAGE_SIZE 256
#define CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR 1
```

negativekelvin, 2018-01-29:
```
 "Timestamp" : NumberInt(1517000441)
 "Timestamp" : NumberInt(1517000438)
 "Timestamp" : NumberInt(1516996890) 
 "Timestamp" : NumberInt(1516996889)
 "Timestamp" : NumberInt(1516996888)
```
That is weird that the esp32 time varies by 1 sec up to 48 sec while real time varies by 1 min

forthlightning, 2018-01-29:
I use a freeRTOS task which calls the read function, and then sleeps for a minute. The actual delay loop takes about 63 seconds to complete because several readings are taken during each minute period. The rough form looks like this:

```
void device_t::read_task( void *p )
{
	while( true )
	{
                device_1.do_reading();
		vTaskDelay( 60 * 1000 / portTICK_PERIOD_MS );
	}
}
```

igrr, 2019-04-10:
Possibly related to #2513.

KonstantinKondrashov, 2020-06-11:
Hi @forthlightning!
I am sorry for the long reply. Do you have any updates about this issue? Does it still happen sometimes or it was fixed?

Is it possible that the time on the device is lagged and then after updating time by SNTP the time jumps to +1hour?

Thanks.

closed 2021-01-11 by Alvin1Zhang