espressif/esp-idf #1550: Exiting esp_wake_deep_sleep correctly

rojer, 2018-01-27:
If i simply return from `esp_wake_deep_sleep`, the boot process does not continue immediately but instead a WDT reset happens and boot process restarts.
```
[Jan 27 17:43:30.631] rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jan 27 17:43:30.631] 012
[Jan 27 17:43:30.631] +
[Jan 27 17:43:30.631] .
[Jan 27 17:43:31.654] ets Jun  8 2016 00:22:57
[Jan 27 17:43:31.654] 
[Jan 27 17:43:31.654] rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jan 27 17:43:31.654] configsip: 0, SPIWP:0xee
...
```
here we see `esp_wake_deep_sleep` running - it outputs 012, then +, and finally a dot just as it's about to return. but then the boot process halts, and only a WDT reset a second later restarts it.
a second is an eternity, and i'd like my main firmware to boot quicker in this case.
so, how can i do it? how to properly leave it `esp_wake_deep_sleep` so that boot happens sooner?

P.S.: i tried causing a SYS_RESET via TG0_WDT myself, by configuring it for a very short timeout. oddly enough, the reset happens, but a second one is still necessary to get the boot going:
```
[Jan 27 17:49:29.132] rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jan 27 17:49:29.132] 012
[Jan 27 17:49:29.132] +
[Jan 27 17:49:29.132]  ets Jun  8 2016 00:22:57
[Jan 27 17:49:29.132] 
[Jan 27 17:49:29.132] rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jan 27 17:49:30.159] ets Jun  8 2016 00:22:57
[Jan 27 17:49:30.159] 
[Jan 27 17:49:30.159] rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[Jan 27 17:49:30.159] configsip: 0, SPIWP:0xee
```
note the 1 sec pause between the first SYS_RESET (caused by my code) and the second one, which happens on its own.

negativekelvin, 2018-01-27:
> If supplying a custom wake stub, the first thing it does should be to call esp_default_wake_deep_sleep()
https://esp-idf.readthedocs.io/en/latest/api-guides/deep-sleep-stub.html

rojer, 2018-01-27:
ok, looks like i'm blind or something. that did it, thanks!

closed 2018-01-27 by rojer