espressif/esp-idf #1552: xtensa-esp32-elf-gcc: Issues with placement of lookup tables

rojer, 2018-01-27:
(this belongs to https://github.com/espressif/crosstool-NG but that repo doesn't have a public issue tracker)

Consider the following function:
```c
enum thing_type {
  THIS_THING = 0,
  THAT_THING = 1,
  OTHER_THING = 2,
};

RTC_IRAM_ATTR
int get_thing_size(enum thing_type tt) {
  switch (tt) {
    case THIS_THING: return 2;
    case THAT_THING: return 22;
    case OTHER_THING: return 42;
  }
  return -1;
}
```

it gets compiled down to assembler code that uses a small lookup table to return the appropriate value.
notice that i asked for the function to be placed in `RTC_IRAM`, however, the lookup table ends up in `.rodata` with everything else and gets put on flash. obviously, this function no longer works when invoked from the rtc wakeup stub, where the region is not mapped. this was a real head-scratcher of a bug!

```
(gdb) disas get_thing_size
Dump of assembler code for function get_thing_size:
   0x400c0024 <+0>:     entry   a1, 32
   0x400c0027 <+3>:     movi.n  a8, -1
   0x400c0029 <+5>:     bgeui   a2, 3, 0x400c0038 <get_thing_size+20>
   0x400c002c <+8>:     l32r    a8, 0x400c0020
   0x400c002f <+11>:    add     a2, a8, a2
   0x400c0032 <+14>:    l8ui    a8, a2, 0
   0x400c0035 <+17>:    sext    a8, a8, 7
   0x400c0038 <+20>:    mov.n   a2, a8
   0x400c003a <+22>:    retw.n
End of assembler dump.
(gdb) x/1x 0x400c0020
0x400c0020:     0x3f414c57
(gdb) x/3b 0x3f414c57
0x3f414c57 <CSWTCH$55>: 0x02    0x16    0x2a
```
from the map file:
```
 .rodata.CSWTCH$55
                0x000000003f414c57        0x3 /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(bts_data.o)
```

now, deep sleep wake-up stubs are not used as often, but i imagine this has iplications for IRAM_ATTR functions as well.
paging dr. @igrr and dr. @jcmvbkbc :)

negativekelvin, 2018-01-27:
https://esp-idf.readthedocs.io/en/latest/api-guides/deep-sleep-stub.html#loading-data-into-rtc-memory

rojer, 2018-01-27:
@negativekelvin yeah, i read that part, but i don't have control over generation of these constants, they are produced by compiler.

rojer, 2018-01-27:
for the moment i moved code to a file with `rtc_wake_stub_` prefix, which does help, but as i said, this issue also has implications for IRAM-pinned code.

jcmvbkbc, 2018-01-28:
My first obvious suggestion would be using -fno-jump-tables

igrr, 2018-01-28:
This issue (applied to IRAM_ATTR) has come up internally once or twice, but we didn't find a systematic solution for it. Ideally, we'd need some flag like -fjump-tables-in-text-section, which would place the jump tables into the same section as the function itself (even if the section is changed from the default using an attribute). Such flag doesn't exist, instead we have a JUMP_TABLES_IN_TEXT_SECTION macro which for Xtensa is 1 if PIC is enabled. We have a task on the internal issue tracker to check the impact of setting JUMP_TABLES_IN_TEXT_SECTION to 1 unconditionally. I'll try to get it done at some point in February.

rojer, 2018-01-28:
@igrr so, how about `-fno-jump-tables` for now, as Max suggested?

igrr, 2018-01-28:
@rojer feel free to add that to CFLAGS of your component in the meantime; we'd like to do some tests comparing the current situation with -fno-jump-tables and JUMP_TABLES_IN_TEXT_SECTION before switching this globally.

lookinano, 2019-10-11:
As I just spent 2 complete days to tackle an IRAM issue related to this problem, I would recommend to add a small hint in the docs on how to work around the issue. If you are not  familiar on how the compiler handles switch...case, as I wasn't, it takes some time to track down problems like this.
Unfortunately I just found this issue when I already solved it because then I knew about lookup-tables. So another option might be to rename this issue to something like. _ISR flash related crashes because of switch...case in IRAM_ATTR function_. That would have helped during my research about the problem.