espressif/esp-idf #1553: .rtc_data and .rtc_rodata should be placed into the RTC_FAST segment

rojer, 2018-01-27:
Currently both are placed into the RTC_SLOW segment, which not only eats at the space available for ULP code, but also makes accessing data and constants very slow, unnecessarily so.

negativekelvin, 2018-01-28:
Because it has different addresses on data and instruction buses would you have to split it up?

rojer, 2018-01-28:
right, that's was also my guess as to why the data was placed into rtc_slow in the first place.
well, ideally - no. ideally we'd have some solution to tell the compiler to place them both into one physical segment, but not stomp on each other. my linker script-fu is not strong enough to tell if it's possible.
but even if it's not, perhaps just divide RTC_FAST ram into code and data sections, making the sizes adjustable.
another idea is to employ a two step linking process: first figure out the space necessary for code, then adjust the size of the code part of the RTC_FAST section for final linking.

igrr, 2018-01-28:
@rojer on the LD script side, something along these could work:
https://gist.github.com/igrr/53069880dbe278a42801178da69aaf46

I haven't checked whether this runs correctly though — there could be issues with CRC of RTC fast memory not being calculated correctly. Existing code might also depend on an assumption that the ULP can access variables with RTC_DATA_ATTR attribute.

rojer, 2018-01-29:
@igrr hey, nice trick! and it works, as far as i can tell - rtc_data gets placed right after rtc_text, even though their address space is different:
```
.rtc.text       0x00000000400c0000      0x256
                0x00000000400c0000                . = ALIGN (0x4)
 *(.rtc.literal .rtc.text)
 .rtc.text      0x00000000400c0000       0x34 /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/esp32/libesp32.a(sleep_modes.o)
                0x00000000400c000c                esp_default_wake_deep_sleep
 *rtc_wake_stub*.o(.literal .text .literal.* .text.*)
 .text.bts_data_point_size
                0x00000000400c0034       0x1e /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(rtc_wake_stub_bts_data_ram.o)
                0x00000000400c0038                bts_data_point_size
 *fill*         0x00000000400c0052        0x2.
 .text.bts_data_ram_push
                0x00000000400c0054       0x92 /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(rtc_wake_stub_bts_data_ram.o)
                                         0x96 (size before relaxing)
                0x00000000400c0064                bts_data_ram_push
 *fill*         0x00000000400c00e6        0x2.
 .text.esp_wake_deep_sleep
                0x00000000400c00e8       0xd2 /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(rtc_wake_stub_handler.o)
                                         0xfa (size before relaxing)
                0x00000000400c0108                esp_wake_deep_sleep
 *fill*         0x00000000400c01ba        0x2.
 .text.bts_get_rtc_timestamp_us
                0x00000000400c01bc       0x9a /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(rtc_wake_stub_utils.o)
                0x00000000400c01d8                bts_get_rtc_timestamp_us
 *fill*         0x00000000400c0256        0x0.
 *fill*         0x00000000400c0256        0x0.
 *fill*         0x00000000400c0256        0x0.
                0x00000000400c0256                _rtc_text_end = ABSOLUTE (.)

.rtc.dummy      0x000000003ff80000      0x256
                0x0000000000000256                . = SIZEOF (.rtc.text)
 *fill*         0x000000003ff80000      0x256.
                0x000000003ff80256                _rtc_dummy_end = ABSOLUTE (.)

.rtc.data       0x000000003ff80258        0x8
                0x000000003ff80258                _rtc_data_start = ABSOLUTE (.)
 *(.rtc.data)
 *(.rtc.rodata)
 *rtc_wake_stub*.o(.data .rodata .data.* .rodata.* .bss .bss.*)
 .rodata.CSWTCH$1
                0x000000003ff80258        0x3 /home/rojer/go/src/cesanta.com/mos_apps/bts/build/objs/mosapp/libmosapp.a(rtc_wake_stub_bts_data_ram.o)
 *fill*         0x000000003ff8025b        0x1.
...
```

rojer, 2018-01-29:
@igrr oh yeah, and there are no issues with CRC, my wake-up stub runs fine and can manipulate its data as well. as i have not got as far as working with ULP yet, i have not checked that aspect.

igrr, 2018-01-29:
Ok, good to know. Hope this workaround works for you for now. I will check if we can integrate it nicely without conflits for ULP (perhaps by adding another section attribute).