espressif/esp-idf #1554: Support for RELRO ELF binaries?

paulreimer, 2018-01-28:
This might be a toolchain question or hopefully an easy answer.

Generating ELF binaries seems to not respect the arguments `-Wl,-z,relro` (and the stronger `-Wl,-z,relro,-z,now`) during the linking phase?

From this description: https://tk-blog.blogspot.ca/2009/02/relro-not-so-well-known-memory.html
I was expecting to see
> • the ELF sections are reordered so that the ELF internal data sections (.got, .dtors, etc.) precede the program's data sections (.data and .bss)
> • bonus: the entire GOT is also (re)mapped as read-only

Here is an example command, and the resulting ELF sections:
```
# xtensa-esp32-elf-c++ -Wl,-pie -Wl,-z,relro,-z,now -nostdlib -shared mod1.o -o mod1.elf
# xtensa-esp32-elf-readelf --sections mod1.elf
There are 13 section headers, starting at offset 0x4b0:

Section Headers:
  [Nr] Name              Type            Addr     Off    Size   ES Flg Lk Inf Al
  [ 0]                   NULL            00000000 000000 000000 00      0   0  0
  [ 1] .dynsym           DYNSYM          00000144 000114 000040 10   A  2   1  4
  [ 2] .dynstr           STRTAB          00000184 000154 000019 00   A  0   0  1
  [ 3] .rela.dyn         RELA            000001a0 000170 00003c 0c   A  1   0  4
  [ 4] .text             PROGBITS        000001dc 0001ac 000064 00  AX  0   0  4
  [ 5] .rodata           PROGBITS        00000240 000210 000028 01 AMS  0   0  4
  [ 6] .got.loc          PROGBITS        00000268 000238 000018 00   A  0   0  4
  [ 7] .dynamic          DYNAMIC         00000280 000250 0000a0 08  WA  2   0  4
  [ 8] .got              PROGBITS        00000320 0002f0 000004 00  WA  0   0  4
  [ 9] .bss              NOBITS          00000324 0002f4 000004 00  WA  0   0  4
  [10] .shstrtab         STRTAB          00000000 0002f4 00005f 00      0   0  1
  [11] .symtab           SYMTAB          00000000 000354 000120 10     12  10  4
  [12] .strtab           STRTAB          00000000 000474 000039 00      0   0  1
Key to Flags:
  W (write), A (alloc), X (execute), M (merge), S (strings)
  I (info), L (link order), G (group), T (TLS), E (exclude), x (unknown)
  O (extra OS processing required) o (OS specific), p (processor specific)
```

In fact, the binary is bitwise identical when specifying `-Wl,-z,norelro,-z,lazy`. Which should be the *opposite* of the above command and should have some effect. So maybe that argument just isn't getting parsed?

jcmvbkbc, 2018-01-28:
> Generating ELF binaries seems to not respect the arguments -Wl,-z,relro (and the stronger -Wl,-z,relro,-z,now) during the linking phase?

`-z relro` is currently not supported by the xtensa ld. It's not just a matter of parsing the option, BFD handler also needs to know about it and handle it correctly. `-z now` is supported.
I'm curious what's the use case for the relro on esp32?

paulreimer, 2018-01-28:
OK fair enough. I like the idea (with partial relro) of re-ordering the sections to put `.got` before `.text`, to protect against buffer overflow attacks. Seems like an easy win.

As to the use case, feel free to check my journey at the link below, but the short version is that I would like to:

- make a dynamic loader running on ESP32
- to receive ELF binaries from the network
- reloc them
- MMU map them into pages as read-only
- exec them with unprivileged PID

https://esp32.com/viewtopic.php?f=13&t=4293

I thought about implementing lazy loading/PLT, but I actually think that RELRO is a better fit anyway. To be clear, my loader would be enforcing the "RO" part anyway, but I would like the original binaries to be organized and marked for RELRO, ideally.

paulreimer, 2018-01-28:
I don't even want a PLT, and I wish I could disable it (GOT-only). But I think `-fno-plt` is not supported in the gcc version in the ESP32 toolchain.

jcmvbkbc, 2018-01-28:
> put .got before .text

The `.got` section is not used on xtensa, it contains a pointer to the dynamic section and that's it. literals are usually merged with the corresponding text sections. The `.got.loc` section describes regions with literals that will need to be fixed up.

> I don't even want a PLT, and I wish I could disable it

Not passing -fpic to the compiler should do that.

paulreimer, 2018-01-28:
Ah, yes I had seen the xtensa specifics, wasn't sure what `.got.loc` was for. But I guess I meant just the concept of having the jump table (whatever section it is in) before the text section, making it just a bit harder to rewrite it via buffer overflow.

Hmm if I don't specify `-fpic` then I don't get a GOT either, though? Do you mean specifically omit `-fpic` for the compiler steps (e.g. still link with `-fpic`)? I got close with using `-pie` (GOT-without-PLT) but that didn't seem right, I think I want `-fpic -shared`.

jcmvbkbc, 2018-01-28:
> I guess I meant just the concept of having the jump table (whatever section it is in) before the text section

There's no jump table, each call to external symbol loads target address from a literal and calls to that address. In the presence of -fpic the literal will initially point to the PLT entry which would call into the dynamic linker, which in turn should fix up the initial literal to point to the final location.
You could compile w/o -mtext-section-literals and place .literal sections separately from the corresponding .text sections... but that solution only works for small programs. Normally literals will be mixed with code.

jcmvbkbc, 2018-01-28:
> it would be possible to keep the literals in .text (e.g. using -mtext-section-literals), and not even use a GOT or -fpic, but just do the relocs in-place in .text, before going R/O and executing

Right.

> What are "literals" in this case?

Values loaded into registers that don't fit into movi (+/- relaxations involving addi/addmi) immediate are turned into literals loaded by l32r instruction. E.g. reference to a variable v in .data or .bss: `movi aN, &v ; l32i aM, aN, 0` which becomes `.literal .Lpv, &v ; l32r aN, .Lpv ; l32i aM, aN, 0`, or a call to a function f: `movi aN, f ; callx8 aN` which becomes `.literal .Lpf, f ; l32r aN, .Lpf ; callx8 aN` produce literals. With -mtext-section-literals they are automatically placed before the beginning of the function that has such code, without it they go to the .literal section. The linker script then usually has rules like the following: `.text : { *(.literal .text .literal.* .text.*) }` that combine literals and text sections from linked objects into a text section in the linked module.

> My goal is to have the .text section immutable (from network -> RAM -> flash -> RAM), and a separate section which contains the relocated addresses.

You can do something like that in the linker script instead: `.literal : { *(.literal .literal.*) } .text : { *(.text .text.*) }` to produce separate .literal and .text sections. In that case there should be no runtime relocations within .text.

jcmvbkbc, 2018-03-07:
FWIW xtensa linker now [supports -z relro](https://github.com/jcmvbkbc/binutils-gdb-xtensa/commit/d6ab64818b9d212691c200c292dc0f18d193598c), i.e. it puts .data.rel.ro in a separate page-aligned segment and provides GNU_RELRO segment.

paulreimer, 2018-03-07:
Well that's awesome news!

paulreimer, 2018-03-07:
I did end up implementing using shared objects so I will be able to take advantage of it (although my loader kinda already compensated for it.) Still wish I had `-fno-plt` but it works well if I do all the GOT indirection beforehand. Thank you for adding this useful feature!

igrr, 2018-03-20:
@paulreimer can this issue be closed, or is there still some question to answer?

Alvin1Zhang, 2019-10-12:
@paulreimer Thanks for reporting. Can this issue be closed, or is there still some question to answer? Thanks.

closed 2021-03-08 by Alvin1Zhang